prusa3d/Prusa-Firmware-Buddy #4527: [BUG] Experimental printing with different nozzle sizes - repeatable reboot and potentially misreported error

FyrbyAdditive, 2025-03-12:
### Printer model

XL Multi-Tool

### Firmware version

6.2.2

### Upgrades and modifications

Yes, but not affecting this case

### Printing from...

Have tried via PrusaConnect and USB

### Describe the bug

I believe I have encountered a repeatable firmware bug which has manifested itself via the experimental printing with different nozzle size feature. After probing the bed as part of starting this specific print (but not others) the printer restarts and tells me there is an "Extruder Preheat Error".

When this initially happened I tested the thermistor and the resistance was a little outside the specified range, and is being replaced under warranty. However it also happens with a new thermistor that is working fine (it is not the heater, that is fine).

Also "normal" prints are working fine since and this error is not repeatable with any other print. This may cause Prusa Research to have to replace more parts under warranty, and maybe/maybe it isn't related to this experimental feature.

I have included the project and the bgcode. This was sliced with 2.9.0.

### How to reproduce

Load PC into 1, 3, 4, 5, load a 0.25 into 1, 3, 4 and 0.4 to 5. Run the bgcode. It reboots when 1 is purging before the print after probing and the error occurs.

### Expected behavior

It prints without crashing.

### Files

[For Prusa.zip](https://github.com/user-attachments/files/19207512/For.Prusa.zip)

FyrbyAdditive, 2025-03-12:
I may have included the wrong bgcode file in the initial submission the attached one is correct.

[Prusa XL - X Carriage Cover - Fyrby - Prusa XL_0.4n_0.15mm_PC,PC,PC,PC_XLIS_41m\[4\].bgcode.zip](https://github.com/user-attachments/files/19208903/Prusa.XL.-.X.Carriage.Cover.-.Fyrby.-.Prusa.XL_0.4n_0.15mm_PC.PC.PC.PC_XLIS_41m.4.bgcode.zip)

FyrbyAdditive, 2025-03-12:
Possibly related but different circumstances https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4528

CZDanol, 2025-03-19:
Thanks for your report. Internal ticket: BFW-6950

FyrbyAdditive, 2025-11-11:
Hi. Sorry I am bad at missing these notifications. The 3MF is in the zip I originally submitted, however reminder to disregard the bgcode in that zip and use the one in the follow up message. Thank you.