prusa3d/Prusa-Firmware-Buddy #4529: [BUG] Prusa iOS App time remaining and forecasted end time are incorrect

neuron-whisperer, 2025-03-12:
### Printer model

MK4S

### Firmware version

6.2.2

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

Some metrics on the Prusa iOS App are currently incorrect. I am running two MK4S printers, both upgraded last night to 6.2.2, and both are exhibiting this problem. I have not noticed this issue before now.

Comparing these metrics from PrusaLink  (which all seems right) with same-time metrics from the Prusa App:

[image]

[image]

"7h 13m to go" and "End 15:44" are obviously incorrect. Percent Complete, Started, and Printing Time are correct and consistent with those in PrusaLink. PrusaLink is also correct about the remaining time and estimated time, and the print ended right on time.

Same problem with the other printer:

[image]

[image]

"7h 5m to go" and "End 10:27" are obviously incorrect, while the other metrics are correct and consistent with those in PrusaLink. PrusaLink is also correct about the remaining time and estimated time, and the print looks like it will end right on time.

### How to reproduce

Start print. Observe metrics in Prusa iOS App and compare with those in PrusaLink.

### Expected behavior

Metrics should be correct, internally consistent, and consistent between apps.

LarGriff, 2025-03-17:
Daylight savings time just went into effect.  My printer started showing the wrong time until I set settings>summertime

neuron-whisperer, 2025-03-17:
That's not the reason.

First, in the examples above, the time isn't just off by an hour - it's off by like 6.5 hours ("31 minutes remaining" vs. "7h 13m remaining").

Second, I updated my time settings - this photo is from a moment ago...

[image]

...and it's still happening on both printers - the following photos were taken within about two minutes of each other...

[image]

[image]

[image]

closed 2025-05-27 by github-actions[bot]