prusa3d/Prusa-Firmware-Buddy #4530: [BUG]

luukakkerman, 2025-03-13:
### Printer model

XL Single tool

### Firmware version

6.2.2

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

We have 2 XL Singe tool printers printing fine simultaneously while standing on the same table. (Yes, we plan to move them to their own table to improve print quality and issues like these). Starting a new job one while the other is running another print without problems.

But since a couple days, one struggles to start a new print job by not being able to home the Z axis. Rebooting does not help. I learned from Prusa forum this can be due to vibrations next to debris or other mechanical issues. When the other printer is not printing, the Z homing does succeed and the print starts and prints like normal which seems to indicate it's the vibration from the other printer causing the issue.

Since we have them running for almost a year side to side and never seen this before, I like to let you know since it might be a result of upgrading the firmware a couple days ago to 6.2.2?

Tomcus, 2025-03-13:
Hi, the loadcell (that is used for the Z homing) is exteremely sensitive to a lot of external influeneces, so it is possible that due to the vibration something got loosen a bit in the the toolhead and that causes the loadcell to report a big spike in data. The Z touches are deted by big changes in loadcell data (the check is relative). So if the loadcell data are really noisy the Z touch can hide in them and not be detected. I would suggest you contact official support and try to troubleshoot it with them.

IIRC there were no changes with loadcell on 6.2, but I might be wrong (you can always try downgrading and checking it for yourself).

luukakkerman, 2025-03-13:
> Hi, the loadcell (that is used for the Z homing) is exteremely sensitive to a lot of external influeneces, so it is possible that due to the vibration something got loosen a bit in the the toolhead [...]

Thanks for your quick reply and suggestion to downgrade. You're probably right there weren't any changes related to loadcell because downgrading from 6.2.2 to 6.2.0 didn't made any difference. 
Pausing the other printer, or reducing its shaking to the table by holding it firmly with both hands, makes the other able to Z-home. Also when the other printer prints slow steady perimeters instead of shaky travel moves, the bed moves nicely up until the shaking begins again.

I'll upgrade to 6.2.2 again and this Bug Issue report can be closed. Will invest in a sturdier table for these two beasts than.

CZDanol, 2025-03-13:
I would maybe also recommend checking if some of the screws hasn't come loose.

closed 2025-04-02 by luukakkerman