prusa3d/Prusa-Firmware-Buddy #4532: [BUG] Y Axis Layer Shift

novisso, 2025-03-13:
### Printer model

MK4S

### Firmware version

6.2.2

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

We had this issue a while ago, we upgraded to the 6.2.0-RC1 firmware at the request of the support team.  This seemed to have fixed the problem, at least we were able to print hundreds of copies of our model without the y axis layer shift.  I just upgraded my machines to 6.2.2, and in the fourth batch of 6 of our model the Y axis layer shift has appeared again.

[image]

### How to reproduce

This is tough, because its intermittent, prior to 6.2 it was happening every other print.  This layer shift didn't happen until the 3rd batch on one of our MK4s printers.  I checked the belt tensions, linear rods and bearings, checked for obstructions, etc. 

### Expected behavior

The expectation is that, given proper printer maintenance the Y Axis would not shift.

### Files

This is a proprietary model.  I am happy to provide it directly to Prusa research.

murk-sy, 2025-03-14:
You should send the files to info@prusa3d.com with the title: Prusa-Firmware-Buddy: [BUG] Y Axis Layer Shift #4532

And add a link to this issue in the email, just to make sure it's as clear as possible what you're sending.

Mikyner, 2025-03-20:
Today I had a massive Y-axis layer shift in the 2.2mm height on MK4S MMU3 FW 6.2.2
With this gcode

[cos_0.4n_0.2mm_PLA,PLA,PLA_MK4SMMU3_1h50m (1).zip](https://github.com/user-attachments/files/19365549/cos_0.4n_0.2mm_PLA.PLA.PLA_MK4SMMU3_1h50m.1.zip)

novisso, 2025-03-23:
I have had it happen on two different machines now.  Not frequently, only about 1 in 10 prints.   I have gone over the hardware on both machines, checking belt tension, re-calibrating, checking bearing alignments, loose screws, re-checking belt tensions, looking for obstructions etc.  The beds on both printers move back and forth with ease, the belts are properly tuned, I tried the pliers on the motor shaft check and the belt is not slippy.  And after a failure it will print several batches perfectly.  I sent the model files to the email suggested.

murk-sy, 2025-03-23:
@novisso As a workaround and to ensure it's not just by chance or other causes, could you maybe restart (maybe just one of the printers) every 2-5 prints? Ideally a full switched power-off for 15 seconds and then turning it back on.

That way, if you once get a layer shift immediately after restart, and once not at all, it's less likely to be firmware related. However, if then you get no layer shifts at all, it's highly likely it is a firmware problem.

novisso, 2025-03-23:
I have been power cycling the machines (as suggested above) before each
print since I saw the first layer shift.  So that means the second layer
shift occurred after a power-cycle reset (on a different machine)

Joseph Gledhill

caprazzo, 2025-04-23:
This is similar to the issue I'm having and reported in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4585#issuecomment-2823705293

Prusa-Support, 2025-05-17:
Unfortunately, there hasn't been much follow-up to this case, so it is technically still open, but based on the initial analysis of the provided files, it may be related to rather specific print conditions we can't approve (quite customized P Slicer profiles).

So far, this issue seems to be most closely related to the print settings and left with incomplete troubleshooting.
With that, this issue should probably be closed after all.

Michele Moramarco
Prusa Research

Snowhoschi, 2025-05-29:
See this comment:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4637

I will contact the support tomorrow. If this isn't a tecnical (hardware issue) than it will be a SW issue. 
Both is possible, due to no limit switch for the x- / y-axes.
"The belt was jumping" in my case (jumping = my assessment)
But this is not good.

Prusa-Support, 2025-06-06:
I may see a correlation with #4573 if new cases are coincidentally based on _Avoid Crossing Perimeters_ feature (unconfirmed concern).
This specific issue (#4532) though looks potentially closely related to the use of unapproved print profiles.
Issue #4637 instead looks even more generic and so far, it may hint at hardware problems to be addressed with the help of our Customer Support.

I'm closing this issue due to the fact that the problem doesn't seem to be reproducible with the approved stock profiles already set in Prusa Slicer.

Michele Moramarco
Prusa Research