prusa3d/Prusa-Firmware-Buddy #4536: [BUG] Printer falls of the network every 2 minutes

Rudd-O, 2025-03-15:
### Printer model

MK4

### Firmware version

6.2.0

### Upgrades and modifications

None

### Printing from...

Local Ethernet

### Describe the bug

Ever since I upgraded to firmware 6.x, the printer falls off the network for roughly 30 seconds every 2 minutes.  This started happening with Wi-Fi so I decided to cable the printer via Ethernet... no success either, the problem still takes place.

Here is a clear chart of the problem courtesy of HA:

[image]

This has also been verified via ping using MTR; it's not just lost connectivity on the part of the HA machine:

[image]

It doesn't matter if Wi-Fi or Ethernet are used, the problem happens just the same.

I wonder if disabling Wi-Fi would cause the problem to stop, but [since there is no feature to disable Wi-Fi so far...](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4322).

Rudd-O, 2025-03-15:
The timing of the printer going offline is so precise, it makes me think the printer is restarting its network on a timer.

Rudd-O, 2025-03-15:
```
ping -c300 prusa-mk4.dragonfear
PING prusa-mk4.dragonfear (10.250.1.32) 56(84) bytes of data.
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=1 ttl=255 time=0.303 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=2 ttl=255 time=1.00 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=3 ttl=255 time=0.851 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=4 ttl=255 time=0.867 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=5 ttl=255 time=1.00 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=6 ttl=255 time=0.535 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=7 ttl=255 time=0.550 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=8 ttl=255 time=0.462 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=9 ttl=255 time=0.405 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=10 ttl=255 time=0.367 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=11 ttl=255 time=0.257 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=12 ttl=255 time=1.12 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=13 ttl=255 time=0.860 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=14 ttl=255 time=0.954 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=15 ttl=255 time=0.840 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=16 ttl=255 time=0.791 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=17 ttl=255 time=0.759 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=18 ttl=255 time=0.599 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=19 ttl=255 time=0.473 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=20 ttl=255 time=0.491 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=21 ttl=255 time=0.431 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=22 ttl=255 time=0.278 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=23 ttl=255 time=0.173 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=24 ttl=255 time=1.14 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=25 ttl=255 time=0.702 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=26 ttl=255 time=1.02 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=27 ttl=255 time=0.730 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=28 ttl=255 time=0.726 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=29 ttl=255 time=0.700 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=30 ttl=255 time=0.532 ms
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=31 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=32 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=33 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=34 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=35 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=36 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=37 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=38 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=39 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=40 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=41 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=44 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=45 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=46 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=47 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=48 Destination Host Unreachable
ping: sendmsg: No route to host
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=50 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=51 Destination Host Unreachable
From ens6.zip-cis.castle.dragonfear (10.250.1.1) icmp_seq=52 Destination Host Unreachable
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=53 ttl=255 time=409 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=54 ttl=255 time=0.784 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=55 ttl=255 time=0.440 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=56 ttl=255 time=0.292 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=57 ttl=255 time=0.311 ms
64 bytes from prusa-mk4.dragonfear (10.250.1.32): icmp_seq=58 ttl=255 time=0.189 ms
```

ScottWell1, 2025-03-16:
I am unable to replicate this, and (so far) I have not seen any similar reports here or on the forum. So, it doesn't seem to be endemic to the firmware.  

I ran a ping test again this morning for over 20 minutes and had 100% success (snapshot attached, below).

I am using the 6.2.2 firmware, while your post indicates you are using 6.2.0.  **You might want to try 6.2.2**, though I doubt that will make a difference (the change log doesn't indicate any networking changes, and I never experienced issues when on 6.2.0).

**You might also want to try a Hard Reset**, as described at the link below, to rule out corrupted settings in XFlash.
[https://help.prusa3d.com/article/factory-reset-core-one-mk4-s-mk3-9-s-mk3-5-s-mini-xl_421206](url)

Given the consistent timing you have recorded, I _theorized_ it might be related to PrusaConnect (though I don't know if you even have it configured).  My theory was that if the PrusaConnect firmware client had problems communicating with the PrusaConnect server, perhaps the client could block local ping response or reset the network connection. However, I simulated that with firewall blocking, and it made no difference -- still 100% ping response from the printer.

Given the lack of other reports, I suspect it is something unique to your printer (like corrupted settings, hence the recommendation to try a hard reset), **or** something in your network environment -- but I have no clue what it could be.  

Good luck, hopefully someone else will offer more ideas.

[image]

murk-sy, 2025-03-16:
I've had weird issues with connection too, could you try this procedure and see if maybe something changes?

Though this was mostly related to Connect issues, it did cause network connectivity to drop out fully.

From https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4049#issuecomment-2188090237

> When it doesn't seem to acquire an IP, doing the following seems to resolve it:
>
>    Set wifi to static and back to dhcp
>    Set eth to static and back to dhcp
>    Turn off and unplug the network cable
>    Turn back on after 10 seconds
>    Replug network cable after fully loaded (important!)
>    Works again (might need a few seconds to connect)
>
>Doing the following seems to re-break it:
>
>    Set wifi to static
>    Set eth to static and back to dhcp (eth at this point still works)
>    Turn off for 10 seconds
>    Broken again

Rudd-O, 2025-03-17:
I will try this trick when I get back home.  I have no problem in principle with running static IP, so we'll see how it goes.

Thanks.

Be sure I will report back with results.

Rudd-O, 2025-03-17:
The trick of switching various network settings as you noted did not work.

I am now trying a prusa_printer_settings.ini that looks like this:

```
[network]
hostname=prusa-mk4
dns4=...

[eth::ipv4]
type=static
... standard stuff as per docs...

[wifi]
ssid=

[wifi::ipv4]
type=static
```

Which seems to have been taken up by the printer fine.

Obviously I'm abandoning DHCP for this.

I'm not doing the right thing for this diagnosis.  I am supposed to change one thing at a time.  Here I'm changing two -- annulling the SSID and changing Ethernet to static.  Let's hope it works.

ScottWell1, 2025-03-17:
I wonder if when an SSID configured, maybe it keeps trying to connect periodically and eventually resets both wifi and ethernet?  

I have never had a problem, but then I have never configured an SSID (have used ethernet from day one). I have always used DHCP as well, with a DHCP reservation on the router so the same IP is always assigned.

bkerler, 2025-03-28:
Regarding the ethernet interface, this is known and already has a PR #4342 for it, caused by a wrong implementation of the full duplex detection leading to drop frames even if not fully processed. If this also happens on the wifi interface, the reason most likely is caused by either a hardware problem or router issue.

closed 2025-06-04 by github-actions[bot]