prusa3d/Prusa-Firmware-Buddy #4538: [BUG] .local hostname unavailable after starting print

meeq, 2025-03-15:
### Printer model

CoreOne

### Firmware version

6.3.0+10073

### Upgrades and modifications

None

### Printing from...

PrusaLink via PrusaSlicer

### Describe the bug

I have PrusaSlicer set up to print to my CoreOne over a `cadillac.local` hostname:

[image]

However, once I start a print, PrusaSlicer can no longer communicate with the printer:

[image]

If I go into the printer's LCD menu and update the hostname (even just hitting the save button and keeping the value the same), it starts working again.

### How to reproduce

1. Set up a physical CoreOne printer in PrusaSlicer 2.9.1 using PrusaLink and a `.local` hostname
2. Test to confirm that we can connect to the printer
3. Send a print job to the printer through PrusaSlicer. (It doesn't matter whether the job finishes or not)
4. Go back to the "Edit Physical Printer" modal and press Test

### Expected behavior

PrusaSlicer should still be able to connect to the printer.

CZDanol, 2025-03-16:
Seems related to #4303

CZDanol, 2025-03-16:
Just to be clear - that .local you have not set up on your router DNS, right? You're using mDNS

meeq, 2025-03-16:
Correct; I am using mDNS

meeq, 2025-03-21:
Just to follow-up: as a workaround, I switched PrusaSlicer to connect via IP address instead of mDNS hostname and it has been working fabulously.

dzuelke, 2025-04-01:
Yup this is the same issue as #4303.

How quickly after starting the print does the problem occur for you, @meeq? I haven't been able to pin down the exact time after which the mDNS replies stop once a print has started yet.

@meeq, can you confirm that if you launch the following in a shell just before you start the print (and keep it running):

```ShellSession
$ while dig -p 5353 prusamini.local @224.0.0.251; do sleep 60; done; echo "No more mDNS response after: $(date)"
```

... that it then stays "alive"? (you have to double-`Ctrl+C` out of the `while` loop to exit it)

(for the uninitiated, this `dig` call sends a DNS query to port 5353 to DNS server 224.0.0.251, which is a multicast address that mDNS capable devices listen on and respond to if the `$whatever.local` name in the query is theirs)

closed 2025-06-09 by github-actions[bot]