prusa3d/Prusa-Firmware-Buddy #4539: [FEATURE REQUEST]  IPv6 For PrusaLink

altf4arnold, 2025-03-15:
### Printer model

Mini+

### Describe the feature

Would it be possible to implement IPv6 on the network interface?

bkerler, 2025-03-28:
IPv6 stack can be enabled (I implemented it once), however it has quite some disadvantages:
1. Not the full ipv6 stack is supported by the current lwip stack
2. It does take quite some space in the flash (which is a big issue on Mini)
3. The network bug (fullduplex) has to be fixed first (Still in PR state)

altf4arnold, 2025-03-28:
Thank you very much. I indeed saw the code on the project. Wasn't sure in which state it was though. For the space on the mini, I see that most firmwares are packaged with 2 languages on them. I would gladly sacrifice one language in each version if it means added space for more features. I don't know how realistic this could be?

bkerler, 2025-03-29:
Imho, this would be a tradeoff between having ipv6 vs. having other printing features. I might discuss with the devs but i'm not having big hope on this one for mini. IPv6 doesn't make a lot of sense if you use it on the LAN, except you intend to use it directly on the internet. Is there a specific reason you require IPv6 ?

altf4arnold, 2025-03-29:
I want to deactivate IPv4 on my LAN's and the Prusa mini is currently one of the reasons I'm stuck keeping it up. I know it's not that common, but that's why I was asking.

bkerler, 2025-04-02:
I see and fully understand. I can try to port my code to ipv6 as the software did change a lot recently, but it will take some time, especially as there are still bugs in the LAN stack for which I wrote a PR that waits to be merged first before I can start with porting ipv6 again :)

altf4arnold, 2025-06-02:
Keeping it open

altf4arnold, 2025-08-04:
Still keeping it open

gerdy001, 2025-08-12:
Keeping it open

yodaldevoid, 2026-10-05:
still still important

altf4arnold, 2026-10-05:
Sadly, given the updates on the mini, it feels like Prusa is just letting it die. I would prefer to have an actual EOL notice instead of just leaving things rot on their side if that's what they are planning to do. (Also, no v6 in 2026... come on.)