prusa3d/Prusa-Firmware-Buddy #4542: [FEATURE REQUEST] G-code to get loaded filament and nozzle details by serial

L10nhunter, 2025-03-17:
### Printer model

Prusa MK4(S)

### Describe the feature

I'm writing my own version of prusalink as a fun little side project, and i realized that I don't have any way to check if the filament in the g-code is the same as the filament in the printer. then i realized there was also no way to check ensure nozzle size or filament diameter either. the thing is, the printer already has the functionality inside itself, because if i try to run a print that has a mismatched filament material, diameter, or nozzle size, the printer screen will warn me about it. why can't there also be a way to check those over serial?

### Expected functionality

if an MK4(S) printer is attached to a computer by an open serial port, the computer should be able to write to the serial port with a g or m code, and the printer would reply over the same serial port with that information. optional improvements would be to add arguments to only return one object in the list, instead of all of them.

bkerler, 2025-04-13:
For nozzle diameter, there is the gcode "M862.1 Q".

bkerler, 2025-04-13:
For printing the filament type, parameter P has been added in PR #4600

CZDanol, 2025-04-16:
Internal ticket: BFW-7076

closed 2025-06-23 by github-actions[bot]