prusa3d/Prusa-Firmware-Buddy #4545: [FEATURE REQUEST] When stuck filament detection error showed up, there's no abort option, only unload.

5jvm0u4, 2025-03-20:
### Printer model

MK4S

### Describe the feature

When stuck filament detection error showed up, there's no abort option, only unload.
I'm testing maximum flowrate for filaments, it did not get stuck, thus I do not need to unload it, if I hit unload it'll endlessly trigger the exact error, cause it's already at its maximum, I just want to abort the whole print and please allow me to do it.

### Expected functionality

as mentioned

LarGriff, 2025-03-29:
I’ve asked for this MANY times and keep getting shot down.  Prusa isn’t listening.

5jvm0u4, 2025-03-29:
Nice to see you again, I think they ARE listening, they just think we are the minority and actively choose not to do it this way, I think the only way they change, is either to attract more attention, or some other major thing happens.

relayer111, 2025-04-02:
Prusa is losing quality in its pursuit of bamboo.

bkerler, 2025-08-19:
If the filament is stuck, it will enter a loop for the unload process and there are two options:
1. Respond with "Yes", then it will try to set the State "Filament not in filament sensor" (filament_not_in_fs state)
2. Respond with "No", then it will try to run the Unload process (disabling the e-stepper and running manual_unload state)
see here:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/marlin_stubs/pause/pause.cpp#L913-L924

bkerler, 2025-08-19:
The main function is here: 
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/marlin_stubs/pause/pause.cpp#L859:L863

(Just documenting the main spots to see where the needed changes have to happen)

bkerler, 2025-08-19:
I just wrote a PR for it ... however, until I'm able to trigger this situation in the simulator/dev board, not sure if this PR does work as intended. Will have a chat with the devs on this one.

tensau, 2025-09-20:
Anything new on this issue?