prusa3d/Prusa-Firmware-Buddy #4547: [ENHANCEMENT] Improve Object Selection in Cancel Object screen

bryn51, 2025-03-20:
### Printer model

MK4S, CoreOne, Mini, XL

### Describe the enhancement

When the bed contains multiple instances of the same object the "cancel object" screen does not distinguish between them. In addition, the Cancel Object screen lists only the first 16 objects/instances.

The proposed enhancement would allow a better user experience and more efficient operation of Cancel Object when (say) one instance of many disadheres from the bed.

### Expected functionality

Instead of displaying a list of objects/instances, instead display a graphic screen depicting the whole bed. Colorise objects : (a) the current object being printed (b) objects that will print in their turn (c) objects that are cancelled.  Put an index number next to all of the objects.

Allow user entry of the index number of an object they wish to cancel by way of a control at top or bottom of the screen.  The entry of the number is made by way of the rotational dial on the xLCD screen.

Alternatively, allow user selection of an object to be cancelled by tapping its position on the screen.  Since this is subject to vagueness its not the preferred way.

bryn51, 2025-03-20:
Dang, Just noticed someone else entered a somewhat similar request. How do I cancel this one.?

CZDanol, 2025-03-21:
I'll do it :) Duplicate of #4507. Internal ticket BFW-6860

closed 2025-03-21 by CZDanol