prusa3d/Prusa-Firmware-Buddy #4549: [BUG] PrusaLink file list reloads every couple seconds when behind nginx reverse proxy

nobodyinperson, 2025-03-21:
### Printer model

MINI+

### Firmware version

6.2.2

### Upgrades and modifications

None

### Printing from...

PrusaLink behind nginx reverse proxy

### Describe the bug

When proxying PrusaLink with nginx (relevant bits below), the file list keeps reloading every couple seconds, making it really hard to start prints.

### How to reproduce

- access a PrusaLink-enabled Prusa MINI (or other printer?) behind an nginx reverse proxy with SSL, no subpath
- open the Storage tab to see the file list
- file list reloads every couple seconds, making it virtually impossible to work with and start a print

devtools debugger with `pause on exceptions` says:

```
324/nt/< - main.3fd7d8a85463914cd803.js:1:104628
TypeError: s.data is undefined
```

I have no idea of javascript unfortunately so that's where it ends for me. I tried pretty-formatting that minified file but that didn't really help.

One can open devtools (F12), and then enable "pause on exceptions" in the debugger tab to stop it and see the error, but then starting prints also doesn't really work. 🤷

### Expected behavior

File list doesn't constantly reload every couple seconds and one can actually start prints from it.

### Files

```conf
# excerpt of nginx.conf
# put on different port, didn't get subpath to work properly
server {
    listen 0.0.0.0:1234 ssl ;
    server_name example.com ;
    http2 on;
    ssl_certificate /var/lib/acme/example.com/fullchain.pem;
    ssl_certificate_key /var/lib/acme/example.com/key.pem;
    location / {
        proxy_pass              http://THE-PRINTER-IP;
        proxy_set_header        Host $host;
        proxy_set_header        X-Real-IP $remote_addr;
        proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header        X-Forwarded-Proto $scheme;
        proxy_set_header        X-Forwarded-Host $host;
        proxy_set_header        X-Forwarded-Server $host;
    }
}
```

nvtkaszpir, 2025-03-30:
som suggestions:
- don't use http2
- in the web browser go to the debug tools, network, get some responses (about 5s) and then stop recording and use Export HAR

nobodyinperson, 2025-08-04:
Still relevant...