prusa3d/Prusa-Firmware-Buddy #4552: [BUG] Multi-Color Cancel Object where all objects left were canceled results in just pritning the purge tower

rcmaniac25, 2025-03-23:
### Printer model

MK4S

### Firmware version

6.2.3-RC

### Upgrades and modifications

MMU3

### Printing from...

USB (monitoring/cancel object via Prusa Connect)

### Describe the bug

I had a multi-color print, where there were multiple different height objects.

I noticed that the purge tower continues to get created, and I get that, because it would mess up colors for other objects.

In my case, the largest objects got canceled and only the small objects were left printing. But once the small object was finished, the printer continued printing the purge tower since the tall object had been canceled. This took up time, filament, and if something happened (like the purge tower coming off) then it produces a mess.

I wonder if simply knowing max height of each object and if the print goes through every object and height is above, it simply ends the print. That way it's not dependent on the purge tower still completing.

### How to reproduce

1. Have a multi-color print with multiple, different height objects
2. Cancel the tall object
3. Let the print continue

### Expected behavior

Once the small object finishes, the print finishes, regardless of the state of the prime tower.

rcmaniac25, 2025-05-26:
This continues to exist as of 6.2.4.

rcmaniac25, 2025-08-02:
There has not been a new update to the firmware that addresses anything related to this issue, so this is still an issue.

3d-gussner, 2025-08-07:
@rcmaniac25 I will have a look, sorry for the very late answer.

closed 2026-05-25 by github-actions[bot]