prusa3d/Prusa-Firmware-Buddy #4553: [ENHANCEMENT] Up to 1.8mm nozzles selectable in firmware

FyrbyAdditive, 2025-03-23:
### Printer model

XL

### Describe the enhancement

Currently it is possible to select up to 1.2mm nozzles on the printer. It is possible to use larger nozzles by overriding this in the slicer, and then going to the printer and manually interacting with it to accept the settings and then confirm.

1.4 and 1.8mm nozzles for the V6 adaptor are available and usable with a little effort (as pictured) - a great enhancement would be to allow up to 1.8mm nozzles in firmware to alleviate us of the need to go and manually poke the printer when we would like to use them.

Hopefully this is "low hanging fruit" and just editing a variable somewhere!

### Expected functionality

[image]
[image]

FyrbyAdditive, 2025-03-31:
A workaround for anyone else reading this that wants to see this reflected on the printer seems to be to set it using Prusa Connect, and the XL seems to display the nozzle size correctly then.

CZDanol, 2025-03-31:
Internal ticket: BFW-7007

schrodlm, 2025-06-24:
Closing this issue, thanks again @bkerler!