prusa3d/Prusa-Firmware-Buddy #4554: [FEATURE REQUEST] Allow remapping filament at the filament head mapping screen

bbum, 2025-03-24:
### Printer model

XL 5 head

### Describe the feature

After loading a print, there is a screen that shows the extruders being used in the loaded file on the left and the extruder status on the right.   This screen allows you to load/unload filament so you can fix any mismatches.

It would be far more convenient if it allowed you to say "everywhere you use extruder 1 in the file, actually use extruder 4".

(Yes, hard problem, I'm sure. :) )

CZDanol, 2025-03-24:
Hello. If I understand you correctly, this is what the toolmapping screen actually is for. You select the "G-code filament 1" on the left side, click on the knob, and then decide what toolhead should be used for that slot.

See https://help.prusa3d.com/article/tools-mapping-xl_668277

bbum, 2025-03-24:
Thank you @CZDanol.  I believe you are correct.

I'll test it out this eve and likely close this as "Guido's Time Machine wins again" (a Python thing).

bbum, 2025-03-25:
> Hello. If I understand you correctly, this is what the toolmapping screen actually is for. You select the "G-code filament 1" on the left side, click on the knob, and then decide what toolhead should be used for that slot.
> 
> See https://help.prusa3d.com/article/tools-mapping-xl_668277

Just verified this.

Nope;  the remapping allows remapping of the same material but does not allow remapping to a different material.

In particular, I have 5 colors of PETG loaded and have a 5 color PLA print that I would like to print in PETG.   It would seem like this should be a temperature change only for passable results.

fyby, 2025-03-25:
Thats not so simple. You say, you want basicaly reslice whole print from gcode. Why? Its not only temperatures. Every material has its own max flow for example. So you need to apply different speeds. Also every material has different cooling needs etc.... I understand you. I also had presliced petg print and I wanted to print it with PLA. But when you look into slicer to filament attributes you will see its much more complex.

closed 2025-06-02 by github-actions[bot]