prusa3d/Prusa-Firmware-Buddy #4555: [BUG] Nozzle Cleaning Failed whatever the filament type

Kachidoki2807, 2025-03-24:
### Printer model

CORE One

### Firmware version

6.3.0

### Upgrades and modifications

None

### Printing from...

Prusa Connect and therefore USB drive

### Describe the bug

Whatever the type of filament used, or even without filament, I got a "Nozzle Cleaning Failed" message at almost every print. It might be a mechanical issue due to over-constrain on the loadcell caused by the new pneufit mount plus the cable bundle relief. I found that un-mounting the pneufit wasn't enough, but releasing the two screws of the mount/cover on the back of the Nextruder solved the issue.

### How to reproduce

On affected printer :

1. Do an auto-home, then lower the Z axis to 50mm.
2. Open the "Sensor Info" menu and look at the loadcell value.
3. Touching "normally" the nozzle produce a variation of ~300 units
4. A slight touch on the cable bundle can produce a lot more, up to a thousand unit of variation and it can even not return to ~0.

### Expected behavior

The printer should print without assistance. Maybe unrelated but it reminds me some similar issue at the beginning of the XL that I found the loadcell value also move a lot when touching the cable bundle, but it work properly on my unit. Maybe a firmware fix can be found here too.

Kachidoki2807, 2025-03-31:
I made more measurements, using the following drawing:
[image]

Coordinates are expressed in millimeters (ie: A(x;y)), the loadcell (called L) value does not have unit.
- A(240;10) Is the auto home position Z=2 after homing and is never changed.
- B(10;10)
- C(10;210)
- D(240;210)

My procedure and measurement records:
- After homing the printhead is located at A, L=~0 (tare). 
- from A to B, L=350 then goes down to 300 after a minute
- from B to C, L=-200 then goes up to -160 after a minute
- from C to D, L=-360 then goes up to -310 after a minute
- from D back to A, L=-60 then goes up to -50 after a minute

Another fact, after re-homing, L=~0. Setting the nozzle temperature to 150°C increases L to 360±20. But what I didn't expected is that setting back the nozzle to off, L is still increasing and goes up to 650 when ambiant temperature is reached back. I could explain that by the pressure of the vapor that could made the filament to expanse a bit, and keep the pressure when cooling down.

PrintFandango, 2025-04-20:
in the latest firmware from this week, I also have this issue in Core One.

CZDanol, 2025-04-22:
Hello, loadcell readouts on different places are not that relevant, because the loadcell is retared just before every probe.

Could you please send us a log from the nozzle cleaning procedure?

Turn on logging to a file in Settings → System
Run the nozzle cleaning (possibly as part of the print)
When it fails, abort the print
Turn off the logging
Share the log.txt file

Thanks :)

CZDanol, 2025-04-22:
Just noticed that there is already the #4523 issue for this.

closed 2025-04-22 by CZDanol

PrintFandango, 2025-04-22:
I have added the log to the other issue. thanks