prusa3d/Prusa-Firmware-Buddy #4561: [FEATURE REQUEST] Prusa XL w/ enclosure should have Core ONE-like chamber temp control

jhnphm, 2025-03-26:
### Printer model

XL 5TH

### Describe the feature

The Core ONE can adjust fan speed to maintain a set chamber temp level. The Prusa XL should dynamically adjust fan speed like the Core ONE in order to maintain a set chamber temperature. In addition, it should be able to toggle unused tiles on and off to assist in maintaining the chamber temperature.

peter12311, 2025-03-26:
I absolutely support this proposal. :).

Ganj44, 2025-03-28:
Yes please ! Let's make the enclosure a bit less of an after thought, hardware is there.

MikeLvu, 2025-03-30:
I support this as well.

CZDanol, 2025-03-31:
I am not sure if we have precise enough temperature readings on the XL to pull this off. IIRC there is not a dedicated chamber thermistor. I will ask the product nonetheless. Internal ticket: BFW-7008

peter12311, 2025-03-31:
Maybe we can optionally add a dedicated chamber temperature sensor to get precise readings...

toniklahan, 2025-04-12:
i support this even add an option to use the fan2 port and use the built in sensor to control the temperature

toniklahan, 2025-04-12:
i support this

CZDanol, 2025-06-12:
Hotend thermistor? You mean the one that will report 215 °C when printing PLA?

JeremiahGillis, 2025-07-23:
> I am not sure if we have precise enough temperature readings on the XL to pull this off. IIRC there is not a dedicated chamber thermistor. I will ask the product nonetheless. Internal ticket: BFW-7008

This would be better than having not having anything to control chamber temp. Maybe we could have more accurate measurements if a GPIO board is ever released.

This feature would be very helpful for certain filament types when using an enclosure.

closed 2025-09-30 by github-actions[bot]

JeremiahGillis, 2025-09-30:
Please keep this open. This is important for those of us with enclosures.

3d-gussner, 2025-10-16:
I looked at the schematics and assembly instructions and there are some unused connectors
- Ambient thermistor
- Fan 2

So we would guess that with a good placed additional thermistor and the existing enclosure fan some chamber temp control could be in theory possible.

### BUT

The CoreOne's internal space is optimized (as low as possible volume) and the fans are able to control the chamber.

The XL has a way way bigger internal volume and question is if the existing fans would be able to control the chamber temperature. It would be possible to add a strong fan to Fan2 connector BUT there is also a limitation of flow that can pass the HEPA filter.

All these factors makes it difficult to find a chamber control solution.

For example: Adding a big fan probably compromises the HEPA filter and even may cause an unwanted draft while printing with "tend-to-warp" filaments and ruin the print.

>In addition, it should be able to toggle unused tiles on and off to assist in maintaining the chamber temperature.

The tiles that are close to the print area are set to 70% of the target bed temp and far away tiles to 50% to keep the temperature in the print area stable. Disabling tiles probably causes more issues that having these like it is now.