prusa3d/Prusa-Firmware-Buddy #4571: [BUG] Pre-Print can’t home Z, it just hunts, and also fails nozzle clean

CpuNut, 2025-03-31:
### Printer model

Core One

### Firmware version

6.3.0

### Upgrades and modifications

I fixed it in the source code 🙂

### Printing from...

USB Drive

### Describe the bug

I have the Core One, 6.3.0 and have nozzle clean fails and even worse at the start of a print Z hunting and the plate never touching the nozzle so doesn't even get to nozzle clean.  I’ve even seen it go into nozzle clean when the plate is a cm away!

From looking at the load cell values I see a correlation to the nozzle heater off or full on where the readings are more steady.  When the nozzle heater starts to pulse width modulate there is a corresponding increase in load cell noise.  Looking at the schematic, the +24 for the heater and the +5V which drives the loadcell share the same ground so my above observation is a good correlation.

I downloaded the source code and in two places (although only one seems to affect readings) when getting the low level A/D read to raw_value, I just added a line after: “raw_value = raw_value /3;” for less but still super adequate dynamic range.

Load to the printer and bam everything works perfectly!  I tried various values and /3 or /4 work the best.

I know this may not be the best fix, it should probably be in the circular buffer multiple reading evaluation code, but it works if anyone else is so inclined.  Really on products where I wrote similar code I did a quick slope calculation across readings, which tends to dampen noise, and look for the sudden change in slope.  I guess pretty much an integration.

Ultimately this needs to be fixed in hardware (why not run the load cell off of regulated 3.3V and even use a separate analog ground), but in the mean time having an EEPROM stored load cell sensitivity adjustment would be the best way to fix and also would allow users to adjust to a specific printer and nozzle arrangement.

Craig

### How to reproduce

As described above.

### Expected behavior

Normal MK4 like load cell behavior finding the plate and passing nozzle clean.

CpuNut, 2025-03-31:
Yes I also looked at the schematic this AM, and they are coupled together at some point.  Anyway anyone can check my observation that the load cell value noise increases when the nozzle heater is close to or at the set temperature when the PWM regulation starts to kick in.  

I'm on my iPad and will copy/paste where I made the simple “raw_sample /3" change in the code later today from the PC.  I'm also going to look at how it uses the sequential loadcell read values in the circular buffer as that might be the better place to make the change.

Regardless my printer went from a mostly paperweight to fully functional with this simple one line change which lowers the noise floor (not really the S/N ratio) on the readings.  I know Prusa was going to address the issue but I got tired of waiting and now my printer works while they put in the permanent fix which seems it might be a load cell sensitivity adjustment menu item.

Kachidoki2807, 2025-03-31:
@CpuNut, your issue is really interesting. Maybe it is complementary to mine (#4555). I can see big static variations on the loadcell value. At a given position the value can be widely different depending on factors such as the constrain brought by the bundle of cables. When releasing this constrain by unscrewing here and there, the printer starts to behave like the old good MK4.

I guess my case is more of mechanical issue and yours is hardware, but we probably have both.

I did a similar test to find a correlation between the loadcell value and the temperature and I got surprising results. But I couldn't easily detect a difference on the dynamic variation (aka noise) between heater on or off. It roughly oscillate of ±20 around a value.

CpuNut, 2025-04-01:
I consider myself a beginner on the Prusa code and am taking baby steps but after waiting for three weeks to get my paperweight working I jumped in and this fixed it.  For sure there may be some hardware vulnerability with my printer which I will also visually inspect.  

That said here is the simple change to get my CoreOne working consistently.  Only one of the two changes seems to have an immediate effect but I put the change in both places for consistency.

This fixed the nozzle clean failures and more so the crazy Z hunting and never settling down to even get to the nozzle cleaning stage.

The change I made to: ..\src\common\hx717mux.cpp commented at "NCB":

----------------------------------------
    int32_t raw_value;

    int32_t read_delay = ticks_diff(ticks_us(), ready_ts);
    if (read_delay > static_cast<int32_t>(1.f / HX717::sample_rate / 32.f * 1e6f)) {
        // too much delay between trigger and read, skip the sample and wait for next cycle
        raw_value = HX717::undefined_value;
    } else {
        [[maybe_unused]] bool was_initialized = hx717.IsInitialized();

        // attempt to read the value
        raw_value = hx717.ReadValue(next_channel, ready_ts);

// NCB reduce sensitivity
raw_value = raw_value  / 3;
//NCB

        // we already account for delayed reads above: we should never get an undefined value unless

------------------------------------------
The change I made to: ..src\puppy\dwarf\loadcell.cpp:

// HX717 sample function. Samples loadcell channel only.
void loadcell_irq() {
    if (!loadcell_is_enabled) {
        return;
    }

    uint32_t timestamp = ticks_us();

    [[maybe_unused]] bool was_initialized = hx717.IsInitialized();

    int32_t raw_value = hx717.ReadValue(hx717.CHANNEL_A_GAIN_128, timestamp);

// NCB Reduce sensitivity
    raw_value = raw_value /3;
//NCB

    // we should never get an undefined value unless the read itself took too long, meaning the

CpuNut, 2025-04-02:
Thanks for helping me understand what the 2nd location I found was for, I appreciate that 🙂

A second data point for me is they sent me a new loadcell which I swapped in and there was no difference.  It was a good guess on Prusa's and my part.

I experimented with the 3.6.0 source a bit more yesterday.  First I comment out my change and sure enough the crazy never find the plate Z hunting started again and in the rare circumstance that it does find the plate forget about passing nozzle clean.

Then I played with the "Loadcell Scale" value in what seems to be the engineering compile (items show up in green), that's located in the Settings-Hardware menu when not printing (it’s not in the Tune menu Hardware tree if a print is started).  It defaulted to a value of 19 and while changing it I see it's bound between 5 and 25 if I remember correctly.  

I tried a value of 5 and once again the printer works great with my change commented out.  A value of 10 also worked.  A value of 15 started to have the pre-print Z hunt issues but probably would have started the print.  The default value of 19 is a total no-go on my hardware so I set it back to five and all works great, again with my change commented out.  With the least sensitive setting of five the plate still gently touches the nozzle so no issue there.

So it looks like they are toying with adding the ability to adjust the loadcell sensitivity and I sure hope it's going to be in 3.6.1 for everyone's sake!

I heard that the 3.6.1 release is imminent, maybe today.  I'm sure we'll all stay tuned!

Yes I at least want to check the cable connections and make sure the inter PCB cables are all properly seated and I have toyed with adding some extra capacitance on the Love Board's 5V powering the loadcell.  There is 10 uF there already.  I really don't want to play with the hardware much (or at all) but cutting a trace and adding a bit of resistance before the 10uF would probably clean it up more as a simple R-C filter.  I think the ultimate hardware fix would be feeding the four wire Loadcell with regulated 3.3V.

Anyway it looks like if they keep the "Loadcell Scale" menu item in the public release this issue or subsets thereof will go away for everyone who is experiencing pre-print Z hunting and the nozzle clean problems that some have reported.

For those of us who can compile the code, we will probably still have access to "Loadcell Scale" even if it isn’t in the public release.

Fingers crossed!

CpuNut, 2025-04-03:
Well 3.6.1 fixed the Loadcell issue but too bad the Loadcell Scale extra menu item isn't in there.  Observing it in operation, and looking at the live Sensor Info values, it seems they dialed it down even more than I did with my divide by 3.  It will hopefully be easy to compile, with no surprises and the super user menus will come back.

Voxel3DPrinting, 2025-04-13:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4596

I got the same issue, diagnosed it down to noise as well. present in 6.3.1

Any chance you could provide me with the BBF to see if it resolves my probing errors too? (at my own risk ofcourse)

CZDanol, 2025-04-14:
Internal ticket: BFW-7075

CpuNut, 2025-04-14:
Hi sorry to not get back sooner.  I found that 3.6.1 fixed the issue for me.  I was able to compile the source code and can see from the otherwise hidden engineering menu that they choose the same value of loadcell desense that I did.

So to be clear you're having the same pre-print Z hunting and/or nozzle clean trouble with 3.6.1?

CpuNut, 2025-04-14:
There is an Avoid Crossing Perimeter issue that causes collisions and ultimately an eventual X/Y shift but that can be avoided by leaving that item at its d3fault un-checked in Slicer.  But that is totally different from this issue, just want to be sure.

closed 2025-04-25 by CZDanol