prusa3d/Prusa-Firmware-Buddy #4572: [BUG] summer time

yahbluez, 2025-03-31:
### Printer model

core one

### Firmware version

newest

### Upgrades and modifications

not relevant

### Printing from...

not relevant

### Describe the bug

The printer did not correct the displayed time after the european summer time switch.
Even if a new print job is send the printer did not switch the time.

Ways to solve that bug:

Let the printer use a network time server or 
take the time from communication with  PrusaSlicer
and use the time the slicer sees on his PC.

danopernis, 2025-03-31:
Settings -> Language & Time -> Time Zone Summertime

yahbluez, 2025-03-31:
Perfect! Problem solved.

closed 2025-03-31 by yahbluez