prusa3d/Prusa-Firmware-Buddy #4573: [BUG] Impossible print - layer shifts

martinweber, 2025-04-02:
### Printer model

CoreOne

### Firmware version

6.3.0+10073

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

This print always shows layer shifts. I tried different slicer settings (Prusa presets 0.25 speed, 0.20 structural). Filament is always Prusament PETG (signal white).

It always starts to shift layers at around the same z-height. I checked the belt tension and I hard-reset and calibrated the printer. Other prints with the same filament work fine.

### How to reproduce

Each single print of the file fails with similar results (layer shifts).

[image]

[image]

[Battery-14xAAA-Gridfinity-1x2_v002 (buggy at 0.25).3mf.zip]
(https://github.com/user-attachments/files/19572425/Battery-14xAAA-Gridfinity-1x2_v002.buggy.at.0.25.3mf.zip)

[Battery-14xAAA-Gridfinity-1x2_v002_0.4n_0.25mm_PETG_COREONE_1h5m_buggy.bgcode.zip](https://github.com/user-attachments/files/19572434/Battery-14xAAA-Gridfinity-1x2_v002_0.4n_0.25mm_PETG_COREONE_1h5m_buggy.bgcode.zip)

bkerler, 2025-04-02:
Not gonna say but I'll do: Did you first contact prusa support on that to rule out issues ? 
This does look a lot like an hw issue, not a fw bug. Most likely belt tension or assembly issue.

m-cas, 2025-04-03:
If it repeats always in the same spot, did you try printing from the memory stick directly?
with older releases of the fw on the mk3 and 4 platform, there were many complaining about file that got corruptud in transfer...

CZDanol, 2025-04-03:
Just to be clear - have you tried printing any other model than this one?

martinweber, 2025-04-03:
Thanks for taking the time to reply!

@bkerler I did not contact support but consulted the support site. I will get in touch with them as well. Belt tension was the first thing I checked to eliminate mechanical issues. This was my first suspicion anyways. All prints before that one turned out good. Some with considerable more detail than this. All layers are fine. No issues there.

@m-cas It always appears at the same height. I sliced it with different settings (e.g. layer height 0.20 instead of 0.25 and different speed settings) and it still occurs at the same spot. So there are different g-code files now that show the issue at the same spot. 

The first offset appears at one of the rings in the middle row. The extruder suddenly jumps there (going left then right). Looking at the g-code, the extruder reverses direction right after the wipe when the perimeter finished (layer 35). I don´t know enough to know if that quick reverting of direction could be the issue though (e.g. causing too much force on the belts). 

[image]

I saw that you released a new firmware yesterday. I flashed it today. While doing that, the Prusa USB stick failed. So I replaced it with a SanDisk stick. So I was curious if the stick was silently failing during print and causing the issue. Printing with firmware 6.3.1 and the new stick still shows the error at the same spot. I did transfer the file via PrusaLink though. I now have 7 prints that failed at the same spot. Different USB sticks, firmware 6.3.0 and 6.3.1. Still need to test by transfering the g-code to the USB stick directly.

I printed a similar model (empty 2x1 gridfinity container with same height) immediately after the last failed print and it turned out fine. Preset was 0.20mm SPEED. If there was an mechanical issue, I would expect at least some issues here as well. I also did another print with higher quality settings and the layers are perfectly fine.

[image]

Beikux, 2025-04-03:
I think I found the issue.

This happens because you enabled "Avoid crossing perimeters".
I had layer shifts with this setting and also with disabled Arc fitting. 

For some reason the printer can't keep up when these settings are enabled/disabled and layer shifts happen. Mainly on round objects. 

You can even see it on the screen that it begins to lag when you print a cylinder with arc fitting disabled. 

This must be a firmware bug on the Core One because this was never an issue on the MK4/S.

martinweber, 2025-04-03:
Thanks @Beikux , disabling "avoid crossing perimeters" fixed the print. It is working now.

bkerler, 2025-04-03:
Thanks everyone for ruling this out. The devs are now aware of the issue.

bkerler, 2025-04-03:
@martinweber can you do me a favor ? Please create a gcode file with "avoid crossing perimeters" and one without and upload them here, then I can try to hunt down what gcode might cause that issue

martinweber, 2025-04-03:
@bkerler here are the files:

[crossing-perimeters-gcodes.zip](https://github.com/user-attachments/files/19586684/crossing-perimeters-gcodes.zip)

Beikux, 2025-04-03:
To quickly replicate the issue you can print a cylinder with the speed settings, more perimeters and arc fitting disabled. With reduced minimum layer time to achieve the max print speed. 

This results in layershifts just after the first layer.

bkerler, 2025-04-03:
Thx @Beikux but as I don't own a Core One, it will be hard to test that without having the printer :)

CpuNut, 2025-04-03:
I have had the exact same issue on a very similar print with round holes.  I do have Avoid Crossing Perimeters set to on, as this is useful when printing holes.

I've noticed the three Z axis trapezoidal nuts have more play than I would have expected.  You can test this by pressing up a bit under each black trapezoidal nut and release to check how much play there is.  

I watched and listened to the printer closely and clearly could hear collisions before the X and or Y shift happened.  What I'm trying is in Slicer to disable Ramping Lift and set it to 2.0mm fixed lift.  

In Prusa Slicer this is in Printers, Extruder 1 in the Travel Lift section.  Unclick Use Ramping Lift, and right above it in Lift Height I set it to 2mm.  You may try a different value.  

I'll report back in a couple hours and see if this 3rd try to print a similar object works...

CpuNut, 2025-04-04:
Well scratch that, or collide that, so frustrating.  

Well with the above changes and trying 2mm and 4mm fixed lift, there was no change and collisions start audibly happening near the top of my almost identical design print at about 33mm Z, followed by a mix of a X/Y layer shift(s) with collisions.  I don't think any specific height is the issue.

Printed the same file on my trusty kit built MK4 and now S, it's no problem at all, zero collisions, and zero layer shift.

The Core One is driving me a bit nuts!

Beikux, 2025-04-04:
@CpuNut Try to print the same file but set the gcode resolution to 0.0125 and see if it makes a difference.

CpuNut, 2025-04-06:
For everything, I print right of a dryer after some hours drying, so the filaments are plenty dry.

The belts are properly tensioned, and Z homing works great (finally with 6.3.1).  The Z-screw step motors are firmly mounted

Okay I tried at 0.1mm layers and still had the hard collision clunks (not like bits of infill ticks) and X/Y shift.

This model prints perfectly on the MK4S.

Back at the 0.2mm Structural setting:

I tried Gyroid and Rectilinear infill, no way I'd try grid with everything else going on, and no difference. 

Two different brands of PLA, one old school, and the other recent high speed with no difference.

I tried a lower Flow Rate.  Typical for me is with 100 +/- 3 percent.  I went down to 90% and then 80% kind of under extruding and still had collisions.

Next try is the printer menu Print Speed which feeds into my hypothesis.  I go a crazy slow 35% and perfection!  

Suggestion: If you want to make sure a Tune menu change takes, finalize the change as the test strip is being laid down.  Sometime before this the printer sets the Tune values to defaults so your earlier change will be lost.

What I think is going on and everyone is going to experience if not now, then when the T-nuts wear and develop more backlash…

Speaking in terms of the MK4 and C1 as I'll call it, the MK4 head Z weight is down on the T-nuts, so when a head jump happens, the Z screws immediately lift the nozzle off the print regardless of T-nut backlash before the head jumps sideways because with the move up, the T-nuts are already fully engaged to travel positively up.  

Okay in the C1 this "lift" is indeed a drop before the nozzle jumps.  Now consider T-nuts with some backlash, and the C1 has a bit of backlash just press up under the plate heater.  In this case the Z screws drop the plate and instead unload the T-nuts, and in the bit of time it takes the Z screws to take out the lash, and positively move the plate away from the nozzle is too long before the print head jumps.  The plate with perfect bearings etc., drops at the acceleration of gravity until the Z screws take out the backlash to positively pull the plate lower, and this is too long at 100% speed on some models on some printers (my friend's C1 prints my simple model with holes in top without an issue).

In the case where setting the Print Speed at 35%, it was very obvious visually and by virtue of no clunks, that the Z-screws had time to take up the T-nut backlash and positively move the plate down, not just relying on the acceleration of gravity to take up the backlash, before the head moves.

I think this matters in the land of milliseconds and that is my theory after crazy hours of testing and going glassy eyed staring at and listening to way too many prints.

So why didn't playing around in the Lift (really fall) area of the Slicer: Printers, Extruder1 menu make any difference as I set "Lift" to crazy numbers like 4mm and then 10mm even tried Ramp with a 90 Deg angle at 4mm and I just didn’t see the printer following either setting.  Either of these numbers would have taken up T-nut backlash by a factor of easily 5 to 10x before the head jumps to a new position.

My conclusion is that Prusa needs to make the "Lift" area of the Slicer menu work or we're all going to eventually run into a manifestation of this problem.  I guess another way is to put a fixed delay factor or one derived by speed where the head jump is delayed to give time to move out of the way before a head jump in the firmware.

Feedback? 🙂

CpuNut, 2025-04-07:
Another data big point:  The above were with Avoid Crossing Perimeters checked.  I did it again with that unchecked and at the same 0.2mm Structural with 100% Print Speed and Flow Factor.  

Guess what there were zero collisions and the print came out perfect.

So why doesn't the plate get out of the way fast enough for Avoid Crossing Perimeters but all goes just fine without that checked?

Clearly there is some issue and I always use Avoid when prints have holes like this and there was never a problem and still isn't on the MK4S running the latest official release.

bkerler, 2025-04-10:
This actually sounds like a z-hopping issue related to the PrusaSlicer, same I had on mk4s as well. Will have a closer look at the gcode this weekend.

lrueegg, 2025-04-28:
I have the same issue and was chatting with the support for hours, checking all aspects of the printer. They have not been aware of this issue 🤯.

I'm trying to reprint my model now without 'Avoid crossing perimeters' enabled and hope that this will fix this problem.

lrueegg, 2025-05-03:
Since deactivating 'Avoid crossing perimeter', every print is fine and the printer is working day and night as expected...

So it would be great, if the @Prusa-Support Team would be aware of this bug and could fix it as soon as possible.

RaazP, 2025-05-08:
+1 from me. Really made me scratch my head, but davidr91 in the forums linked me here.
I thought it were the belts skipping teeth, but it was just 'Avoid crossing perimeters'.
BTW: Reducing travel speed to 200 and travel acceleration to 3000 also seems to fix it. In case you want to print with it activated and with the speed profiles.

Prusa-Support, 2025-05-31:
Thanks for reporting.
As long as disabling Avoid Crossing Perimeters [ACP]  proved to be a solution for many, if not all of you, it will surely be worth looking into this feature of Prusa Slicer and how it behaves in relation to Prusa CO. I'm actually a bit surprised not to see any MK4/XL similar report, especially XL which shares the same XYcore motion system.

It almost seems like ACP is making the printer spend a lot of time processing tens of brief G-code lines in rapid succession, without really moving much. Not sure if the nozzle is spending enough time around the same perimeters to be avoided, to ooze and partially stick to the spot where it is over, of the the Buddy board processor struggles to keep up with the many small rapid movements the travel is subdivided into.

In the following pictures you can appreciate how much lines it takes for ACP to move around perimeters compared to the single movement when the feature is disabled, and how fast these tiny many movements are.

[image]

[image]

I wouldn't be surprised to see that the problem is a consequence of slicer, hardware and firmware mixed factors but I'll do my best to escalate.

Michele Moramarco
Prusa Research

Prusa-Support, 2025-07-04:
Several of these prints, shifted in ~45° direction, and not solved by disabling ACP, may actually have something to do with the tension of the belts or problems with one specific motor/pulley.

We are currently working on a new method to evaluate the belt tension more efficiently.
https://www.reddit.com/r/prusa3d/comments/1lp3eo7/psa_vfa_update_2/
I hope it will help to rule out cases potentially related to belt tensioning.

Michele Moramarco
Prusa Research

rongith, 2025-07-11:
Here too I never managed to print a relatively simple 1400g PETG part on the XL. I think it cost me at least 8-10kg of filament. It is not the only part failing to be printed but that 18h print it is a 100% fail.

The Prusa belt tension website is a nice idea but not reliable enough. The main frequency can be masked by many things (fan running in the background, ...) and you can't know it. I prefer an app like Spectroid (Android).
The homing tower test is fine.
I tried many things without success even downgrading from 6.2.1. I'm out of idea.

I never ever had issues to print that part on my Voron 2.4.

ErichStyger, 2025-07-28:
I do not have 'avoid crossing perimeters' checked. But I see that printing issues with round or rounded designs, e.g. https://www.printables.com/model/944916-modern-rectangular-key-tray

Prusa core one FW 6.3.4 and tried as well 6.3.5-alpha, slicer 2.9.2, with default 0.20mm **SPEED** @COREONE HF04:
- it happens in the first layers, around layer 8-15
- printer makes strange 'banging' noise when it happens
- layers get shifted each by about 0.5 or one perimeter or so
- muliple layers get shifted, with a total shift of about 2 mm
- resulting into a small layer shift (x, y) of about 1mm

Image of that print on the bottom side:
[image]

Tried the same model with 0.20mm **STRUCTURAL** @COREONE HF04, but this time even worse:
around 5 m displacement at a higher layer.
Stopped that build, image from the top side:

[image]

This is a factory built core one, and belt tension is OK. I'm not sure what else I can try out, but this makes the core one pretty useless to me :-(.

ErichStyger, 2025-07-29:
I did some more analysis and observations:
- as seen with above image, there are some strings in the lower right corner
- during printing, there are some 'banging' noise events. This has been identified as the nozzle banging against already printed walls/filament. That banging ultimately causes the head to displace, causing the x/y shifts
- looking closer with above re-print, it can be observed (and heard) that it is causing against a line/wall which should not exist (see picture below)
[image]
- looking in more details, it seems that the head took a 'shortcut' for multiple lines, which should be on the round part (arc), but have been printed 'inside', building up a wall the head was banging against

I thought that might be because of an input shaper issue recently reported (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4732). So I disabled the input shaper with GCODE, but that did not help.
I downgraded the firmware to 6.3.2 (I believe this was what was shipped with the machine).
Additionally, because to me the issue seems to be related with some arc optimizations, I turned off 'Arc fitting'.

[image]

Then printed the same item with STRUCTURAL, and voilà: no issues now :-).

I believe the issue is with the arc fitting, so disabling seems to solve the issue on my side.

imax9000, 2025-07-29:
> I thought that might be because of an input shaper issue recently reported ([#4732](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4732)). So I disabled the input shaper with GCODE, but that did not help.

How exactly? I though it could be only disabled in the printer menus 😅

"Avoid crossing perimeters" seems to be problematic mostly because it generates moves at max feed rate (`F21000`) regardless of any speed settings in the profile. If it were to respect the configured travel speed - printing slower would probably work even without disabling it.

musicderp, 2025-08-02:
Can confirm that I also experienced this bug, failed 2 prints with the layer shift in the exact same spot with the Avoid Crossing Perimeters setting, disabled it, no layer shift. It looks like a move was generated that told the printhead to go from one direction to the other at 350mm with no deceleration or stop in between, and that likely caused the layershift.

rfilgas, 2025-08-08:
Ugh, just lost a print to this bug. No avoid crossing perimeters for me.

Prusa-Support, 2025-08-08:
So far, among the "reproducibility steps", there have been several reports about `Avoid Crossing Perimeters` and `Arc Fitting` features from Prusa Slicer, but we have not been able to confirm these concerns so far.

Please note that the Belt Tuner web/app and the instructions have been improved with precise frequency values for each belt [98Hz (upper) / 92Hz (lower)] and an advanced debugging waterfall plot.
We are also experimenting with a tool-free method for more precise tuning, making use of a "stroboscope effect". Stay tuned.
https://www.reddit.com/r/prusa3d/comments/1m995lw/psa_vfa_update_3/

We haven't been able to reproduce the issue, but for now, please attempt once again to tune the belt frequency to the updated values as per the new instructions of the relevant Help article.
https://help.prusa3d.com/article/adjusting-belt-tension-core-one_845048
**Pro-tip.** There have been edge cases where the belt tension was not immediately spread out uniformly across the XY-core structure so _you may want to repeat the belt frequency test after a few print hours or "full print area" manual movements_.

Please check your belts once more, _**especially** in the case where the layer shifting is not perfectly reproducible (different height, different amplitude) and not related to `Avoid Crossing Perimeters` or `Arc Fitting`_.

Michele Moramarco
Prusa Research

Beikux, 2025-08-08:
> So far, among the "reproducibility steps", there have been several reports about `Avoid Crossing Perimeters` and `Arc Fitting` features from Prusa Slicer, but we have not been able to confirm these [...]

Simply create a cylinder and disable arc fitting with the speed profiles.
Set the perimeters to something like 5 and the minimum layer time to 1s to achieve the max speed on the small object. You will get multiple layer shifts after just 5 seconds.

ErichStyger, 2025-08-17:
As an additional data point: I have a factory-built Core One and seeing occasional layer shifts, usually a few millimeters. They seem to be with 'round' extruder paths.
Changing to the 'structural' profile seemed to reduce the issue somewhat, same for reducing the speed. But it still happens, but hard to reproduce.
Then I realized that my printer has been set to non-silent mode. So I turned it to 'silent' and from then I never had layer shifts.
So to me, the issue seems to be related with rapid extruder moves. There are other reports about input shaper possibly causing issues (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4732).

Finally, I decided to order the accelerometer kit to re-calibrate my machine.
The factory values were
- X EI 60 Hz
- Y: EI 50 Hz
After calibration of my machine:
- X ZVD 53 Hz
- Y MZV 59 Hz
It seems to me that the factory programmed values are really not a good match for my printer? It seems to me that Prusa is programming all factory produced printers with the same values, without really calibrating it? Or maybe transport or 'setting the mechanics with some hours of printing' really changed the values that much?

Results so far: disabled the 'silent' mode, printing with the 'speed' slicer settings, and so far no layer shifts with faster printing :-).

So my recommendation is: if you see such layer shifts: try using the silent mode with 'structural' to have less rapid moves.
Buy or borrow the accelerometer kit and calibrate your machine, as the factory values might not be a good fit.

I hope that might help others out here.

3d-gussner, 2025-09-02:
There is a new Prusa blog https://blog.prusa3d.com/new-in-prusaslicer-consistent-surface-finish-and-nerfing-vfas_120400/ please read it and try the PrusaSlicer [2.9.3-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.3-beta1) and the new profiles and report back if the issues are still there.
Please also read the updated [Adjusting belt tension article](https://help.prusa3d.com/article/adjusting-belt-tension-core-one_845048)

Prusa-Support, 2025-09-13:
> I'm currently using the 6.4.0 alpha build.

Thanks for your feedback. Have you also tested the new belt tensioning wizard?

We would appreciate collecting more feedback based on...
- Prusa Slicer 2.9.3 (final), especially based on BALANCED print quality profile (Prusa CO)
- Firmware 6.4.0-alpha (pre-release).
  - New firmware-based belt tuning wizard.

I would also like to confirm that blobs, but also curling up prints (especially), could be relevant for the issue and clarify that those require HW troubleshooting and print preparation checks.

Michele Moramarco
Prusa Research

vertigo235, 2025-09-13:
> Thanks for your feedback. Have you also tested the new belt tensioning wizard?
>
> We would appreciate collecting more feedback based on...
>
> * Prusa Slicer 2.9.3 (final), especially based on BALANCED [...]

Yes of course, I did (cool method by the way!)

I also thought it might be due to the travel being over the part, so I set Z-Hop to 1mm and the problem still persisted.  As noted by others, there is a really *weird* sound that happens when the layer shift occurs, I watched it happen it was not a collision.  

I'm certain there is something going on underneath inside the planner or something. It could be in input shaping, it could be in phase stepping, it could be something else, I don't know.

I do have arc fitting turned off because that was causing other problems with the print.  

I can't share my G-Code because it is a proprietary part that I sell. 

I have 4 other machines running klipper that do not have this problem.

imax9000, 2025-09-13:
> I also thought it might be due to the travel being over the part, so I set Z-Hop to 1mm and the problem still persisted. As noted by others, there is a really _weird_ sound that happens when the layer shift occurs, I watched it happen it was not a collision.

Is the sound similar to what I recorded here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4732#issuecomment-3288455791 ?

vertigo235, 2025-09-14:
> > I also thought it might be due to the travel being over the part, so I set Z-Hop to 1mm and the problem still persisted. As noted by others, there is a really _weird_ sound that happens when the layer shift occurs, I watched it happen it was not a collision.
> 
> Is the sound similar to what I recorded here [#4732 (comment)](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4732#issuecomment-3288455791) ?

Yes I think so.

I can say it's definitely not a collision.

Prusa needs to take this seriously, and stop gaslighting people into thinking that it's belt tension, blob collision, etc.

Just admit it's an issue and get to work on it please.

ErichStyger, 2025-09-14:
I had random layer shifts (PLA, not using TPU yet). I have a square gantry and proper belts tuned on my core on. I tried a lot of different things (USB memory stick, different slicer settings, slower speed, using structural profile, ...) and they all seemed to help in one or the other way, but not consistent. What helped on one model, faced layer shifts on another model. Very frustrating.
I noticed strange/loud/banging noise at the time of layer shifts, so I suspected the nozzle crashing into the print. So did all kind of Z hopping and infill patterns, and again this helped sometimes, but not on different prints. And it failed on different layers too.

Where I am at now is that I suspect that the input shaping somehow goes crazy in some (random?) cases? I had recalibrated with the accelerometer my input shaping and phase stepping, and that seemed to help in some cases. But not in others. Finally, I have decided to *disable* input shaping on my core one (use the menu on the machine and set the frequency to 0 Hz).
So far I have printed to models with that setting (all other settings are the default ones), and to models which always caused issues with layer shifts printed without problems!

Prusa Slicer 2.9.3, Core One HF0.4, 0.2 mm balanced and 0.2 mm structural.
Core One Firmware 6.3.4, with **input shaper disabled** and phase stepping reset to defaults

So if you see issues with layer shifts on your core one, my recommendation is to disable the input shaping on the printer and see if it helps.

bkerler, 2025-09-14:
@vertigo235 I'm not yet sure if it is really a FW issue. I will try the above [crossing-perimeters-gcodes.zip](https://github.com/user-attachments/files/19586684/crossing-perimeters-gcodes.zip) gcode as I've got my core one now and report back.
I am aware of a home related issue which I was able to partly track down on M92 usage which needs further research but I doubt it is related to the arc issue.

bkerler, 2025-09-14:
I was having a look again at the gcode at the spot where the issue is happening and there is something off ...
`M106 S107.1` .... the S value should be a unsigned short of 0..255, not a float value.

bkerler, 2025-09-14:
I started printing and I'm facing the same issue and my printer is perfectly tuned. HOWEVER ... I don't think it is a printer firmware but instead slicing issue. The layer shift clearly seems to happen due to z-hopping in my case. I'll start recording the print with debug msg now so I can figure out the exact spot (around 9.8mm - 10mm for the crossing perimeters file) .

bkerler, 2025-09-14:
So I did compare with crossing perimeters off. Feedrate change is identical, so that can be ruled out.
Regarding the nozzle movements ... if crossing perimeters is on, the nozzle follows the circles at the same height  (9.4) as the circles, if it is off, the nozzle goes directly in a straight line over the circles at z-height 9.956.

bkerler, 2025-09-14:
so imho, I think the problem is that the z-height is changed too fast between line 35334 ```G1 X120.086 Y124.895 Z9.409 F21000``` and line 35335 ```G X127.616 Y128.352 Z9.554```, crashing into the circle.

bkerler, 2025-09-15:
@imax9000 if these can be reproduced, I'd be interested into debugging these.

BootsSiR, 2025-09-15:
While I've had no issues with my Core One, I had problems printing this model yesterday.  I tried it once on v6.3.4 and discovered a layer shift about halfway into the print.  I flashed v6.3.5-alpha remembering that there were layer shift fixes in that firmware and started the print again.  I happened to be in the room when I heard an 'odd' noise from the printer (a light thud?) and it had a huge layer shift.  I cancelled the print and I noticed when the print head moved to it's resting position, the motor stalled as if the print head wasn't where the printer thought it should be.

Here's the 3mf.  I haven't tried to print it a 3rd time yet.

[flexture-Body.zip](https://github.com/user-attachments/files/22348067/flexture-Body.zip)

vertigo235, 2025-09-15:
@imax9000  have you calibrated phase stepping and input shaping on your machine or is it just the factory defaults ?

vertigo235, 2025-09-15:
ON that note, is it possible or does it even make any sense to disable the IS on all travel moves and then re-enable it on moves that have extrusion.

techyg, 2025-09-15:
The print is looking good with Input Shaper manually disabled by setting the shapers to "null" when using the settings menu.

However, based on the Prusa Documentation here:
https://help.prusa3d.com/article/input-shaper-core-one-mk4-s-mk3-9-s-mk3-5-s-xl-mini_451816

It says to disable Input Shaper, you should set the frequencies to 0 and can do this using Gcode. But this seems different than using "null" on the menu, which also seems to disable Input Shaper and was the method I tried.

I am not sure how to get the same behavior of setting the shapers to "null" with M593 or other gcode, nor do I know if setting it to 0 will truly disable it using M593 X0 T0 and M593 Y0 T0, which seems to be what the documentation is suggesting. I will have to experiment and see if it functionally does the same thing.

techyg, 2025-09-16:
> However, I now consistently get a very small (1mm?) layer shift when
> the print reaches the 97% mark, which has repeated 3 times, accompanied by
> the rough sound with a high speed travel.So there is still something not
> quite right.
>
> I had something similar with the SPEED profile and stealth mode disabled.
> Enabling stealth mode with STRUCTURAL worked fine.

Thank you! I enabled stealth mode with structural, and did not get any
layer shifting. That is not what I was expecting. So to summarize.. after
at least 10 prints, the only I am able to get a consistently good print
(without a layer shift) when using "Avoid Crossing Perimeters" is to do the
following:

1. Disable Input Shaper (set values to "null" using menu)
2. Use structural profile
3. Enable silent mode

Not ideal, but for my print not having input shaper seems to make no
difference. Hopefully Prusa can resolve this bug in a future firmware. I
would still like to explore turning off the input shaper in the GCODE so I
can do that before and after each print.

bkerler, 2025-09-17:
Prusa is already actively reviewing this issue, so hopefully we will get more insight soon. But yes, it seems to be related to circle/arc based movements and the high moving speeds, that I'd say can already be confirmed. I printed the problematic gcode with 90° rotation and it still failed at exactly the same spot. So from my side, it's now reproducable and confirmed to be a bug or hw issue.

vertigo235, 2025-09-17:
I will offer some anecdotal and likely coincidental observation, which likely isn't a factor since others have said this makes no difference. 

I finally got an accelerometer and performed the IS and Phase Stepping calibration and thus far I have not experienced the issue.  Granted I'm not doing torture tests, but printing my normal parts with arc fitting off, and avoid walls / perimeters turned on.  This would cause the shift before, but I have not had it yet. Only a few prints though. 

However, prior to doing this, a certain part that I would print would shift every time. 

Other mitigating settings is that I have reduced travel to 150mm/s and my travel accel is 4000, so that might also be a factor.

aaronse, 2025-09-19:
@HaWiWe looks like you're printing V1 Engineering Mostly 3D Printed LowRider 4 CNC parts ( http://docs.v1engineering.com/lowrider ), since the parts are openly shared, are you able to share A) the .gcode files that are causing you problems, B) PrusaSlicer version and settings?

antpata, 2025-09-30:
Just as a data point default YOLO flow calibration print of Orca slicer seems to fail on Core One with out of the box settings for PETG. It seems to fail when printing numbers, with very notable sound like a ratchet slipping. Number areas of the print have lot of small curved moves and whole top surface is full of round shapes and spirals.
0.2 Core One speed profile 0.4 nozzle and both arch fitting and avoid wall crossing on by default. Belts and etc are tuned OK, IS and phase stepping calibrations done before and after.
Core one firmware 6.4.0-RC+11796 or 6.3.4  and OrcaSlicer_Linux_AppImage_Ubuntu2404_V2.3.1-beta
I need to do a test print disabling those features and/or with lower speed. But this is ugly if  Orcha slicer will reproduce this issue so easily and so that it will repeat. I did three different slicings of that YOLO test that all failed before I find this open issue. All other prints until now has been flawless.

Prusa-Support, 2025-11-19:
> This is mostly a matter of _perpetual firmware optimization_, as it has always regularly happened but what seems to help for now may be
> 
> * tuning belts to optimal values;
>   * **(FW 6.4.0 can greatly simplify this process)**
> * ensuring a properly squared printer geometry;
> * disabling `Avoid Crossing Perimeters` or `Arc Fitting` if necessary;
> * comparing results and shapes in other print quality profiles;
>   * (there may be specific scenarios where one among speed/balanced/structural performs better than others)
> * maintaining the printer well as per the [Help documentation](https://help.prusa3d.com/article/regular-printer-maintenance-core-one_829710)...

This issue is still being worked on, and the firmware optimizations currently in progress aim to reduce the potential margin of layer-shifting in edge cases.
There are still several variables to play a role into this issue, that in many cases are not related to the firmware though.

For this reason, I would like to add this the list above.
- Please try to slice again using [PrusaSlicer 2.9.4 - Junction Deviation](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.4-rc1), which could reduce scenarios of partially "stuttering" tool path.
  - https://github.com/prusa3d/PrusaSlicer/issues/11573 and https://github.com/prusa3d/PrusaSlicer/issues/14826 may be examples of extreme scenarios.

Michele Moramarco
Prusa Research

donohuk, 2025-12-11:
> I've encountered this issue as well, in both cases while printing with PETG at low layer heights. I'm using PrusaSlicer 2.9.3 and Core ONE firmware 6.3.4, with accelerometer-related calibrations recalculated on-site.
> 
> A few days ago, I contacted support regarding a crash that occurred during sequential printing — the fan shroud collided with a finished part before starting the next one (That maybe should be another issue as I saw). Right after that, I began experiencing severe layer shifting accompanied by loud bangs. I stopped the print immediately, assuming the shifts were related to the recent crash, and ran the full calibration procedure. After a successful calibration, I attempted another print canceling right from the start the objects I've already printed, but the layer shifting reappeared right from the first layer with the round object.
> 
> I contacted support again, but they were unable to reproduce the issue and suggested I reslice the remaining part of the model. That piece was mostly circular. The print failed once more, with the same loud bangs and an aggressive layer shift (over 2–3 cm). I then resliced the model using the stock “Generic PETG” profile, and the print came out flawlessly. Over the next few days, I completed several additional prints with no issues, mostly using default profiles and only adjusting infill and wall settings.
> 
> A few minutes ago, I attempted to print a completely round cap with thin walls. I enabled Avoid crossing perimeters along with a few other settings, and after several minutes I heard the loudest bang so far — followed by a layer shift of over 3 cm on both X and Y axes. After a quick search online, I found this thread and tried disabling both Avoid crossing perimeters and Arc fitting, leaving all other settings untouched. The print then completed flawlessly.
> 
> I’m wondering whether such violent layer shifts could have caused any damage to the belts or pulleys — I’ve never seen or heard such abrupt movements on a properly functioning machine. I think it would be advisable to first confirm whether this behavior is indeed related to those two settings A.S.A.P. before attempting a fix, so that users can be warned not to enable them until the issue is resolved — to prevent potential hardware damage. The noise was truly monstrous, as if the printer had fallen to the floor.
> 
> I’m also willing to provide the sliced file I just printed if it helps with the investigation. As mentioned, the only difference between the flawless print and the one that failed catastrophically was disabling Avoid crossing perimeters and Arc fitting. Although, to be fair, I’ve only performed one print of each so far. I'm starting right now the second attempt of the previous successful try, and see how it goes.
> 
> EDIT. Second attempt came out flawessly aswell, no aparent issues.
> 
> Cheers!

I think I am having the same issues. Been printing on my new Core One for about 2 weeks. Installed the hardened nozzle that comes in the ultimate assembled core one. Printed a couple items successfully. Then was printing CF PETG and noticed a layer shift about 5mm up on a round section of my part. I tried printing it again but moved the part on the build plate in the slicer. This time had a much smaller layer shift but about same height again on the round section of the item. I decided to print a different part altogether ( badge holder) and as it was just starting to print, I was in the other room, I heard loud banging over and over. Ran into room to find my printer had broken off the extruder tip and was gouging the hell out of my textured build sheet. Going to contact support ASAP but found this thread in my research

[image]

. I wish I could just get on the phone with them to solve this quicker instead of messages/emails back and forth. 

[image]

[image]

[image]

[image]

imax9000, 2025-12-11:
I think at this point it would be helpful if @Prusa-Support posted somewhere visible a summary of possible causes and what steps people can perform to differentiate between them.

I'm currently aware of these:

* Nozzle crashing into the part
    * Could be due to the part curling up, or some sensor issues causing the printer to incorrectly estimate bed position
* Software/hardware issues that cause motors to skip steps
    * My current best hypothesis is that the high CPU load causes the firmware to fail sending signals to stepper drivers at the appropriate time, leading to motors not performing a move correctly

To tell these two apart it's possible to run a problematic print w/o any filament loaded (that is, nozzle will be just moving through the air with nothing to crash into) and listening if the same kind of noise repeats. I was able to reproduce the issue even with belts taken off the motor pulleys, i.e., motors just spinning freely without any load.

Having such a post would both help people to slightly better understand what's happening, and allow them to provide more information in the issue report, making it easier to categorize and investigate.

OperatorDishS0ap, 2026-01-14:
I encountered layer shifting issues as well related to the avoid perimeters setting. My model pictured below is cylindrical with holes within it. 5 perimeters and 30% gyroid infill. My tests have been with a very hard TPU. At first I was using orca slicer (2.3.1) and failed 4 prints due to layer shifting. Altering z hop and speed settings made no difference. Avoid crossing perimeters was enable for all prints. I then proceeded back to prusa slicer (2.9.4) and moved to PLA on the speed profile again with avoid crossing perimeters enabled and the print failed due to layer shifting. I moved to the structural profile, altered the perimeters and infill, and the print finished. I proceeded to use prusa slicer for the TPU and had another 4 failures due to layer shifting. Avoid crossing perimeters was enabled for all. When the shift happens there is a very noticeable clunk that occurs. I watched the fourth print and saw it skip. It did not appear to catch anything on the print. There was no stringing, blobbing, or warping edges. I proceeded at this point to take to the internet and found this thread. I disabled avoid crossing perimeters and viola the print finished perfectly. I printed the model another time to confirm and it was again successful. To further test I went back to orca slicer copied the settings from prusa slicer as best I could, notably disabled avoid crossing perimeters. The print finished successfully twice.
I have only 150hrs on this machine. Belts are tensioned correctly at 98Hz and 92Hz respectively. There is no gantry racking. The cauliflower test print I ran prior showed zero skew to the gantry as well. I have not changed, adjusted, or measured any Input shaper related settings.
I hope this info proves useful. Like others have said my experience leads me to believe there is something wrong with the firmware on the machine. 

[image]

vertigo235, 2026-01-14:
Another workaround is to use stealth mode, I've yet to have it happen in stealth mode.  I wish I knew the parameters in stealthmode that help mitigate this and I'll just create a print profile to use when I have a print that might experience this.  If anyone knows the constraints that are activated with stealth mode...

e0en, 2026-08-21:
I have the same issue. 

- Core One+ upgraded from MK4S
- FW 6.8.1+16182
- 0.4mm Normal Brass Nozzle
- 0.2mm layer height, balanced profile
- ”Avoid crossing perimeters” option turned on

dwbmb, 2026-08-29:
This issue persists with INDX and Gen2 upgrades and 6.9.0 fw.

Seven weeks happy printing with INDX, switched on Avoid crossing perimetres once, leading to immediate failure.

[image]

cyrusjk, 2026-10-09:
I am also experiencing this on a Core One + gen 2 with the 7.0.0 firmware and Prusa slicer 2.9.6.

[4 more replies]