prusa3d/Prusa-Firmware-Buddy #4575: Inconsistent and constant calibration before printing and filament changes [BUG]

simonbugler, 2025-04-02:
### Printer model

Core One

### Firmware version

6.3.0

### Upgrades and modifications

None

### Printing from...

PrusA Connect

### Describe the bug

Printer crashes /bumps print head trying to calibrate for inconsistent number of minutes. Sometimes causing a collision error. 

Resets will sometimes make it calibrate with a double bump. Sometimes it happens again. 

Most notably is how inconsistent it is. 

### How to reproduce

Print something.

thijstriemstra, 2025-04-03:
would love to see a video of that 😅

CZDanol, 2025-04-04:
Hello, please contact our support for troubleshooting.

simonbugler, 2025-04-16:
I seem to have found a 'fix' for this by putting the printer onto the solid floor.

Maybe it should be noted that the Core One needs a VERY stable surface.

closed 2025-04-16 by simonbugler