prusa3d/Prusa-Firmware-Buddy #4576: [BUG] Error#17536 Can Appear on DOCKED Toolhead

Stego72, 2025-04-03:
### Printer model

XL 5T

### Firmware version

6.2.1

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

Extruder just docked can trip error #17536 "Extruder motor not spinning on tool XXX" although not supposed to be spinning at all at that time

[image]
[image]

### How to reproduce

Not possible - occurs at random but only when using more than 1 tool (error happens just after too change on on recently parked tool)

### Expected behavior

No error related to extruder spinning as tool is docked

### Files

No crash dump generated by printer

Stego72, 2025-04-03:
> Hello, we have a fix prepared for this for 6.2.4. Internal ticket: BFW-6869

Thank you SO much - this is a performance and business breaking bug. Is there a version available for test ?

rovili, 2025-04-03:
Why are the people with this issue not informed that it is  now a known issue?!

I went through new cables, new dwarf boards, new XL splitter and new Sandwich board!

Gik-0101, 2025-04-03:
Because it's damaging to companies when something like this is officially admitted.

I've been writing my fingers to the bone about this error since 2024.
So far, the conversion kit has helped me, and I've only had the error once more. I had the dwarf cables replaced, and so far, I haven't had the error again. But I use the XL privately, so I don't have nearly as many print hours and tool changes on the XL as a company.
That's why I'm holding back.
However, like everyone else here, I'm extremely angry and disappointed that this error apparently hasn't been completely fixed yet.
Every extruder error 100% results in wasted filament, wasted energy, and wasted time.

waldo1001, 2025-04-03:
> Because it's damaging to companies when something like this is officially admitted.
>
> I've been writing my fingers to the bone about this error since 2024. So far, the conversion kit has helped me, [...]

I'm in the software business.  I understand, relate, and am just glad that things are moving forward.  Some issues are just very hard to detect (this clearly wasn't something that was easily reproducible).  Admittedly - it sucks.  But we'll get there.  IT's the best machine on the market - and if software can fix this, I'm all for it!

rovili, 2025-04-03:
My printer had 6 hours of print time when the issue first time appeared. That was on the first day after shipment.
It was 100% reproducible. Always only appeared when parked and never happened on a single tool head print.

I was pointing at the FW from the very beginning, since the issue happened for me on all 5 heads. I wasted in the whole process 5kg of filament, because each time prusa wanted proof it happens on all heads.

The fact that I had never extrusion issue on 6.1.3 was totally ignored and I was asked to go back and stay on 6.2.x.

Gik-0101, 2025-04-03:
@waldo1001 
I'm with you 100%.

But keeping quiet about it isn't right either.
I hope Prusa can finally make this printer what it was meant to be all along.
I'm keeping my fingers crossed for them.

Stego72, 2025-04-03:
> My printer had 6 hours of print time when the issue first time appeared. That was on the first day after shipment. It was 100% reproducible. Always only appeared when parked and never happened on a single tool head print.
> 
> I was pointing at the FW from the very beginning, since the issue happened for me on all 5 heads. I wasted in the whole process 5kg of filament, because each time prusa wanted proof it happens on all heads.
> 
> The fact that I had never extrusion issue on 6.1.3 was totally ignored and I was asked to go back and stay on 6.2.x.
> 
> At least I have now all electronics, except Buddy Board and Power Supply as spare parts…
> 
> [@CZDanol](https://github.com/CZDanol) by when can we expect the new FW? Or can we have a Beta/RC? Please!

Think the proper way to escalate into hot-fix land is to (all) write Prusa support - I doubt they know

Stego72, 2025-04-04:
Just got off Prusa Support Chat - Instruction was to downgrade FW to 6.1.3

JulianKetterer, 2025-04-04:
How sure can you be that the error will be gone after the downgrade to 6.1.3?

Stego72, 2025-04-04:
> How sure can you be that the error will be gone after the downgrade to 6.1.3?

As I understand it the RED #17536 Error message was not introduced until 6.2.0 - It is addressing non spinning extruders (cable problems) - side effect is trip of docked toolheads. 

SO: Downgrading is ONLY a viable path if the 17536 is experienced on docked toolheads - anything else and it is the dwarf board connector upgrade etc,

JulianKetterer, 2025-04-04:
When?!

JulianKetterer, 2025-04-04:
But why does the Prusa support still claim the error is not firmware related? I just chatted with them about the issue and they still only adviced me to change the cables, perform a hard reset and install the latest firmware..

Stego72, 2025-04-04:
> But why does the Prusa support still claim the error is not firmware related? I just chatted with them about the issue and they still only adviced me to change the cables, perform a hard reset and install the latest firmware..

Non-Aligned support - the guy I chatted to straight up suggested FW downgrade without me mentioning firmware at all quote: "downgrading worked for another customer i helped recently" - for some reason the FW bug is not in the helpdesk systen - and the error landing page only mentions cable upgrades... Poor governance, management and communication.

Also had to point out that the page he suggested on how to downgrade firmware is wrong - you have to double-press the knob to force FW flash. Page states single press.

murk-sy, 2025-04-10:
Should be resolved with [v6.2.4-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.4-RC1)

draeath, 2025-04-13:
Would be nice to get a signed build with this fix cherry-picked so we don't have to worry about other RC-related bugs.

Kachidoki2807, 2025-04-15:
My feedback on this. I never had extruder issue in the past, but after the recent firmware update I wasn't able to finish a dual tool print of more than 2 or 3 hours. Before to have incriminated the firmware I found the new cable holder recommended by Prusa. Unfortunately it didn't changed anything to my case. Then I found that each time this issue occurred, the toolhead was parked.

This version 6.2.4-RC1 seems to have completely solved my case. I just finished a five tools 15+ hours print twice without a single hiccup.

closed 2025-04-22 by Prusa-Support