prusa3d/Prusa-Firmware-Buddy #4578: [BUG] [COREONE] [6.3.1] Layer shift after last organic support layer

cryptomilk, 2025-04-04:
### Printer model

CORE One

### MMU model

No-MMU

### Firmware versions

6.3.1

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

This happened to me twice in a row, a layer shift after last organic support layer. The first time I thought something happend but twice is probably a bug.

First print (Sliced 2025-03-10, PrusaSlicer 2.9.1)
[image]

Reprint of the failed part (Sliced today with PrusaSlicer 2.9.1):
[image]

I didn't have this problem with 6.3.0.

### How to reproduce

Print the attached GCode.

[00_drybox_base_0.4n_0.2mm_PLA_COREONE_2h14m.bgcode.gz](https://github.com/user-attachments/files/19600204/00_drybox_base_0.4n_0.2mm_PLA_COREONE_2h14m.bgcode.gz)

### Expected behavior

No layer shift.

### Files

[00_drybox_base_0.4n_0.2mm_PLA_COREONE_2h14m.bgcode.gz](https://github.com/user-attachments/files/19600204/00_drybox_base_0.4n_0.2mm_PLA_COREONE_2h14m.bgcode.gz)

3d-gussner, 2025-04-04:
@cryptomilk I moved your issue to the correct repository.

cryptomilk, 2025-04-04:
@3d-gussner Thanks!

cryptomilk, 2025-04-06:
The model is for my filament dryboxes. I've printed them in Prusament Galaxy Black PLA which came with the printer and it printed just fine. I don't have Black PLA anymore so I wanted to print it in Prusament Vanilla PLA. I have 10 filament boxes so it printed fine before.

I've reprinted the model with Firmware 6.3.0. I was there after did the last layer of the support. The nozzle collided with the model.

[image]

I then repainted the support on the model, check in PrusaSlicer, flashed firmware 6.3.1 again and printed again. The nozzle collided with the model again.

[image]

I  can't see anything wrong when checking in PrusaSlicer. Here is the plate.

[00_drybox_base.3mf.gz](https://github.com/user-attachments/files/19619814/00_drybox_base.3mf.gz)

cryptomilk, 2025-04-22:
It took me a few more failing prints to figure out what the issue is exactly. The head always crashed at one of the seam positions. 

[image]

As the seam is always at the edge the model gets a bit higher at this position and higher and higher. At one point it is too high and the nozzle crashes with the model.

[image]

The fix is to move the seam.

[image]

Now the question is, if this is maybe a slicer bug and it should move the seam to a different position for those kind of overhangs. Or should the plate moved a bit further down with those overhangs?

I dunno.

CZDanol, 2025-04-23:
Hello,
good job figuring it out, and sorry that we haven't had time to have a look at it ourselves. 

From what you're writing, it means that this is not a firmware bug, but rather a physical issue. You can try talking this over in the Slicer repo, but I'm not sure that the slicer guys will have an idea how to "fix" this either. 3D printing still sometimes requires knowledge and adjusting the print configuration, and I'd guess this falls into that category.

cryptomilk, 2025-04-23:
Feel free to close it. The "Slicer guys" even don't look at my Pull Requests, I don't think they will look into my bug reports. Also I tried to get in contact with them in other ways but no luck :-(

CZDanol, 2025-04-23:
All right, closing :(

closed 2025-04-23 by CZDanol

CZDanol, 2025-04-23:
Could you give me the link to the issue?

CZDanol, 2025-04-23:
All right, I have pinged them.