prusa3d/Prusa-Firmware-Buddy #4579: [FEATURE REQUEST] Extruder Error 17536 - Offer a retry/recovery option instead of bricking a print with a reboot!

teeny23, 2025-04-04:
### Printer model

XL, XL Multitool

### Describe the feature

Less Wasted Time
Less Wasted Filament

With the ongoing "issues" around Extruder Motor Not Spinning 17536, why are we forced with a hard reboot instead of some sort of recovery/retry option.  I have wasted several KG of filament over this and countless hours.

### Expected functionality

Extruder Motor is not spinning on tool "x"
Reboot Extruder?
Retry?

Anything but a hard reset!

teeny23, 2025-04-05:
> If it is on a docked toolhead official temp-fix is to downgrade to 6.1.3 - it is a confirmed bug that docked toolheads can trip 17536 (fix in 6.2.4 pending)
> 
> If it is on a non-docked toolheadt you should opgrade your cable connectors as per the landing page for the error

Yes sir.  I have indeed upgraded the connectors and covers.   The issue is, at least in my opinion, this shouldn’t cancle the print with no option of recovery/retry. Or to even change filament and sub to a different tool

closed 2025-06-12 by github-actions[bot]

Ro3Deee, 2025-11-08:
this should not be closed...

Ro3Deee, 2025-12-14:
Please reopen this...

teeny23, 2025-12-15:
Issue still exists - FW 6.4.0

even after replacing cables on XL per tech support, toolhead develops LED Flicker over time.  This tells me for sure an extruder error will occur and cancel my print.  THE ONLY solution is to clean the connectors with IPA 99.9% (cable and buddy board) which removes the flicker and error free prints continue.  If you ignore this cleaning, BRICKED PRINT - wasted filament.  The worse part is the the extruder "stops spinning" error occurs on its way to the DOCK!!

Ro3Deee, 2025-12-15:
Poor man's solution for flickering and extruder not spinning on Prusa XL:

[image]
https://youtu.be/xBL0WHxMq0k?si=p9UlQogPvdhqr_qH

Prusa-Support, 2025-12-17:
> Poor man's solution ...

Thanks for your input, but the **practice shown in the video could be potentially hazardous**.

**CAUSES**
The **most common causes** for this error are **a fatigued cable** or **a wobbling connector**.

**DANGERS**
- The practice shown in the video is mostly hazardous because the printer seems to be powered on (_lights are visible behind the printer_) while the connector behind the printer is being plugged. The printer must always be powered off before the user approaches electronics. This may be potentially unsafe for the user, and very often it causes permanent damage to circuits.
-  The use of _contact cleaner_ or _isopropyl alcohol_ (specified in the video's description) is generally safe, but, again, _only if the printer is actually powered off_. The same applies to the metal brush but ideally, please avoid this unless strictly necessary.
- This cleaning practice may be unnecessary except for visible oxidation, dust, powder, debris... Moreover, it may expose more these parts to even a small margin of error, and further damage and wear that could otherwise be avoided.
- All in all, this cleaning practice may probably be similarly effective as simply blowing air (from can, or bulb pump) through the contacts and reseating the connector_.

**SOLUTIONS**

The permanent solution would be replacing the affected component, in order from most to least likely (based on rough statistics and amount of movement). 
- Dwarf-Splitter cable (fatigued cable).
- Dwarf board (wobbly port).
- Sandwich/Splitter board (wobbly port).
- Take this chance to upgrade the Dwarf connector cover design (not crucial but can reduce shaking).

**TORUBLESHOOTING TIPS**
- You can move cables between positions to check if the problem follows one or a few specific cables.
- If not, you can move tool-heads between positions to check if the problem follows one or a few specific Dwarf baords - or narrow the problem down to the specific port on the Sandwich/Splitter board.
- In all csaes, _you are strongly encoraged to contact our customer support for help before proceeding with replacements_.

Michele Moramarco
Prusa Research