prusa3d/Prusa-Firmware-Buddy #4580: [FEATURE REQUEST] Extruder Thermal Runaway 17204 - offer option to pause print to diagnose and fix instead of hard reset only

teeny23, 2025-04-04:
### Printer model

XL, XL Multitool all models

### Describe the feature

Less Wasted Time
Less Wasted Filament

ERROR Extruder Thermal Runaway 17204, gives no option to continue, and should not be a "FATAL" error

### Expected functionality

ERROR Extruder Thermal Runaway 17204 occurs, give option
RESET PRINTER
PAUSE PRINT JOB & RETRY

CZDanol, 2025-04-10:
Hello, thermal runaway is a quite serious problem that if not handled with caution could for example result in things catching on fire.

Adding any exceptions or softening up the conditions to this feature to something more sophisticated than "turn off everything, stop everything, panic" would increase chances of bugs being introduced and the safety measure not triggering properly in a situation where it should have been.

Thus, we do not intend adding options like this. We want to lean on the side of caution have this as straightforward as possible.

teeny23, 2025-04-10:
> Hello, thermal runaway is a quite serious problem that if not handled with caution could for example result in things catching on fire.
>
> Adding any exceptions or softening up the conditions to this [...]

Fair enough.  What led me to this request is that I got the error and maybe it’s a bug as was triggered improperly. Turns out my nozzle through many cycles of heating and contraction, had worked its way slightly loose from the heating block. The heating block would swivel around the nozzle about 5 degrees. From “tight”. 

There was no actual runaway issue, but a mild cyclical heat transfer issue to the nozzle as it was looser and tighter as the print head was moving and printing.  A simple snugging of the nozzle and resetting the nozzle set screw was all it took to eliminate the error 

Basically it triggered outside of the value “can’t maintain temp for xxx seconds”and I believe it triggered on some mild fluctuations