prusa3d/Prusa-Firmware-Buddy #4581: [BUG] Prusa XL goes offline when left in the menus

ChrisHeerschap, 2025-04-04:
### Printer model

XL

### Firmware version

6.2.2+8853

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

If I do something with the printer at the LCD - ex: changing filaments - and forget to return to the home screen, the printer goes offline in Prusa Connect and I can't start prints until I return to the printer and exit the menus.

"Menu Timeout" is enabled in Settings... User Interface, so it doesn't appear to be working either.

### How to reproduce

Go into the change filament menu. Do things. Leave it there. Note that the printer has gone offline in Connect.

### Expected behavior

Don't go offline when in the menu, alternately, automatically return to the "home" screen after a period of inactivity.

Gik-0101, 2025-04-06:
I can confirm this bug.

The easiest way to trigger the bug is to manually park a toolhead. The XL doesn't automatically return to the main screen from this submenu. Therefore, the XL goes offline in Prusa Connect and remains offline until you manually return to the main screen via the touchscreen.

CZDanol, 2025-04-23:
Internal ticket: BFW-7098

ChrisHeerschap, 2025-04-23:
If you send a print through PrusaConnect while the printer is offline in the menus, and then realize and exit the menus so it goes online, it shows the "Downloading..." screen but never finishes the download. I usually have to reset the printer, delete the transfer and remove the print from the queue, then send everything all over again once the printer has re-established the connection.

w531t4, 2025-05-01:
+1 i see this issue as well with a Prusa XL 2T - v6.2.5.

ChrisHeerschap, 2025-07-01:
This still happens to me routinely and is a serious annoyance. Do I have to open a call with support to get some attention on this?

Honestly Prusa quality has really gone in the tank, and my Core One is likely the last Prusa I'll be buying.

eqdeng, 2025-09-25:
> +1 i see this issue as well with a Prusa XL 2T - v6.2.5.

I just experienced this on XL 5T v6.2.6.

Ro3Deee, 2026-01-06:
seems to be resolved in fw v6.4.0

Gik-0101, 2026-09-01:
fixed in firmware 6.10.1.

Thanks!

closed 2026-11-10 by github-actions[bot]