prusa3d/Prusa-Firmware-Buddy #4582: [ENHANCEMENT] Can this headbanging be stopped please

rinkek, 2025-04-04:
### Printer model

Core One

### Describe the enhancement

When homing the head bangs agains the corner like a crazy person. 
Please stop doing this, it drives me nuts. 
Sometimes it takes several minutes.

### Expected functionality

Hit the corner once or twice and then stop.

me25542, 2025-04-04:
Is this every time it homes?

rinkek, 2025-04-05:
> Is this every time it homes?

Not every time.

murk-sy, 2025-04-05:
Please provide at least a video so the devs have something to work with. With the information given there's nowhere to really even start investigating or trying to reproduce the issue. If it happened on every machine with some obvious circumstances I imagine they would've known about it.

rinkek, 2025-04-05:
> Please provide at least a video so the devs have something to work with. With the information given there's nowhere to really even start investigating or trying to reproduce the issue. If it happened on every machine with some obvious circumstances I imagine they would've known about it.

Sorry, will post a video as soon as a i can. I just thought everyone would know what i meant, it's very obvious behavior of this printer when it starts a print.

rinkek, 2025-04-05:
> Have you checked the tension of the belts? I had this problem with my first CoreOne, it sometimes went on for several minutes. After checking the belt tension (with the Prusa app or via https://help.prusa3d.com/article/adjusting-belt-tension-core-one_845048) I realised that they were far too loose and out of synch. After I adjusted them correctly, the problem was gone and homing only takes a few seconds. I don't have this problem with my second CoreOne, the belt tension there was ok from the start.

Thanks for your tip.
But yes i did, it helped a bit. But why does it needs to bang so much, i know it's for something like sensorless homing, but other printers with this feature don't do this.

e-basteln, 2025-04-08:
Another vote on this. Even with the belt tension nicely adjusted and the X gantry squared, the Core One forcefully bumps into the front right end stops multiple times -- in the best case. If it is somehow unhappy with the StallGuard sensing results, it easily repeats a dozen times.

Can this detection *please* be made less noisy (slower attack speed) and more reliable (less repeated attempts)? 

If that should be technically impossible, can "precise homing" be made optional? The only reason I see for it is to enable resuming at precisely the correct XY position after a power outage. That functionality is irrelevant for me since power outages are very rare here -- but the headbanging annoys me before every print. A less accurate homing just to confirm the printable area would be good enough for me, and I hope at least that can be achieved with less noise and without the many repeats.

Artimus70, 2025-04-16:
Same problem here, newly received CoreOne since the first print started headbanging... a lot... a lot means up to 4 minutes strait.. I checked the belt and they seems ok. Also the HW test reports is all good... I really hope Prusa will fix this soon.

uktricky, 2025-05-09:
Assembled Core One Batch 11 - Same issue, sometimes the banging goes on for minutes, the next time it could be half-dozen knocks....Why it has to do it when unloading filament also no idea!

ademars, 2025-05-14:
This forum thread documents a large number of users experiencing this issue:

https://forum.prusa3d.com/forum/prusa-core-one-hardware-firmware-and-software-help/constantly-recalibrating/

miklosbecsei, 2025-05-21:
I'm experiencing the same as well. Few seconds of loud banging before every print. MK4S to Core One conversion kit. Belt tensions are good and in sync.

rinkek, 2025-06-04:
> Its all about gantry allignment. Belt tension SHOULD je the same, but the allignment of the gantry is the key... It is annoying , yes, but after some time invest the gantry is fully alligned and homing is done in seconds.

I have 4 printers, 3 corexy, they all do alignment. The Core One is the only one with this head banging.

But nice they're looking into this. It feels cheap.

limburg11, 2025-06-05:
> Its all about gantry allignment. Belt tension SHOULD je the same, but the allignment of the gantry is the key... It is annoying , yes, but after some time invest the gantry is fully alligned and homing is done in seconds.

Indeed there was a small gap between the end of the gantry and the front right y stop, while the gantry end at the left y stop was flush. I've adjusted the belts asymmetrically (loosened left side a bit and tightened right side) until both gantry ends were flush at the front y stops. Now the left belt has a resonance frequency of 80 Hz and the right one has 90 Hz. Y homing now takes about 10 s (still hitting the front right corner several times), but much better than 1 or more minutes of headbanging! 🤘😎

klinwer, 2025-06-08:
Same issue with my core one conversion kit. The gantry is alligned, the belts are set to 85Hz. Headbanging all the time. Small success with different frequencies. The left belt with 40Hz, the right one with 30Hz. This works a little bit better. Far away from a reliable homing process!

maxkueng, 2025-06-08:
I have an assembled Core One kit from batch 4. Both my belts are tuned to exactly 85Hz. 

The headbanging when homing is different every time. Sometimes it works as expected, sometimes it it headbangs for a minute, sometimes it detects a crash while headbanging and starts over. 

Today it headbanged for 8 minutes until I pressed the reset button. Just stopping the print didn't work because it wanted to finish the homing sequence first. After rebooting it homed as expected with just a few bangs on the frame.

Are there any logs that we can post?

drsunnyk, 2025-06-09:
Assembled Core One. Firmware 6.3.3 installed today. Received printer 2 months ago. Prints great. Takes an eternity to actually get it to print with minutes of bang the extruder head against the right front corner. Banging the extruder could be 20-30 seconds to 3-4 minutes. Especially annoying when print has finished a few hours ago (cold printer) and I’m just trying to get my PETG filament unloaded and into a dryer at the end of the day.  Not sure why it needs to calibrate any axis to unload filament. 

Would post a log if you tell me how.

JeffWDH, 2025-06-09:
I have a Prusa assembled Core One and have had issues with this since the beginning. I *think* I have resolved this but am unsure as this behaviour has been intermittent since the start. 

I had no luck using the Prusa belt tuner in the app, or on the web, on multiple devices. I thought I had success with Spectroid but the head banging came back after a bit. I also resquared the gantry as per https://www.reddit.com/r/prusa3d/comments/1ku9a7e/tuning_belts_on_core_one/mu0qgsd/ as mine was off by a few mm and thought that fixed it but it also came back. 

Today I printed this (https://www.printables.com/model/1166964-prusa-mkxlminicore-6mm-belt-tensioner-6mm-merka-re) in Overture PETG (0.15 structural profile, 100% infill) and tightened the belts so they were equally within the min/max markers. After a long initial auto home I have had a consistent 15 quick auto homes, including after multiple power cycles.

I don't know if this is solved 100% but it seems to be working a lot better. 

For reference, here is the serial log of the auto home that seems to have made things right:
```
start
echo: External Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2019-10-18 | Author: (none, default config)
echo:Compiled: May 22 2025
echo: Free Memory: -268543173  PlannerBufferBytes: 1792
echo:  G21    ; Units in mm (mm)

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 T2 D1.75
echo:  M200 T3 D1.75
echo:  M200 T4 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X100.00 Y100.00 Z400.00 E380.00
echo:Maximum feedrates (units/s):
echo:  M203 X350.00 Y350.00 Z30.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X10000.00 Y10000.00 Z1000.00 E1500.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1250.00 R1250.00 T1500.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_>
echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z2.00 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S0 Z0.00

Unified Bed Leveling System v1.01 inactive
echo:PID settings:
echo:  M301 P14.00 I1.00 D100.00
echo:  M304 P126.13 I4.30 D924.76
echo:Z-Probe Offset (mm):
echo:  M851 X0.00 Y0.00 Z0.00
echo:Stepper driver current:
echo:  M906 X550 Y550 Z600
echo:  M906 T0 E450

echo:StallGuard threshold:
echo:  M914 X-2 Y-2 Z4
echo:Driver stepping mode:
echo:Filament load/unload lengths:
echo:  M603 T0 L20.00 U105.00
echo:  M603 T1 L20.00 U105.00
echo:  M603 T2 L20.00 U105.00
echo:  M603 T3 L20.00 U105.00
echo:  M603 T4 L20.00 U105.00
echo:Tool-changing:
echo: Z2.00
echo:Phase stepping:
echo:  M970 X1 Y1
echo:enqueueing "G28 P"
echo:endstops hit:  X:12.23
echo:Backoff ipos:980
echo:busy: processing
echo:endstops hit:  X:250.00 Y:-29.54
echo:Backoff ipos:3684
[... 575 lines cut ...]
echo:endstops hit:  Z:60.00
echo:busy: processing
echo:endstops hit:  Z:60.00
X:240.00 Y:10.00 Z:2.00 E:0.00 Count A:25000 B:23000 Z:800
echo:enqueueing "G28 P"
echo:endstops hit:  Y:10.00
echo:Backoff ipos:980                                                          
echo:busy: processing                                                          
echo:Backoff ipos:3900                                                         
echo:endstops hit:  X:251.92                                                   
echo:endstops hit:  X:242.36                                                   
echo:busy: processing
echo:endstops hit:  X:242.36
echo:endstops hit:  X:242.68
echo:endstops hit:  X:242.68
calibrated home cycle A:30 B:0
echo:busy: processing
echo:endstops hit:  Z:60.00
echo:busy: processing
echo:busy: processing
echo:endstops hit:  Z:60.00
echo:busy: processing
echo:endstops hit:  Z:60.00
X:240.00 Y:10.00 Z:2.00 E:0.00 Count A:25000 B:23000 Z:800
```

jonnieZG, 2025-06-17:
It is especially annoying when need to print something late in the evenings - it wakes up everyone. 

The printer should need to display more information about the **reason why the calibration is triggered**, it might help troubleshoot problems if the belts are loose or the gantry is skew, or there is something else.

It also needs a **Calibration precision / sensitivity** setting, depending on how likely is a power-outage to occur. Lower setting would have more generous thresholds for triggering calibration and would require less head-banging than the higher setting.

Virge, 2025-06-17:
I have this issue on my kit core one. I was able to find a work around. Just reset the printer between prints and the issue goes away. If I reset the printer, it only corner humps 4 times. If I don't reset the printer, it will corner hump for 10 mins and then gives up and starts the print.

jonnieZG, 2025-06-18:
> Just reset the printer between prints and the issue goes away. If I reset the printer, it only corner humps 4 times. If I don't reset the printer, it will corner hump for 10 mins and then gives up and starts the print.
> 
> I’m curious if this _solution_ works for others experiencing the same issue.

This could be referred to as a temporary workaround rather than a "solution".

Virge, 2025-06-19:
Update/Clarification on Previous Comment

After spending a few more hours testing, I realized that some of my original assumptions were off and wanted to provide a more accurate update for anyone else troubleshooting this behavior.

The repeated corner probing ("corner humping") was mostly resolved after I re-tensioned and properly routed the belts in the XY gantry. I initially threaded the belts through the extruder backing plate at a slight downward angle, which caused misalignment and added lateral tension. This seems to have made the belts pull slightly during homing.

After adjusting the belts to match the routing shown in the official build guide, the issue improved significantly. Now, when starting a print, it typically only bumps each corner 4 times during homing and proceeds normally, rather than cycling for minutes or restarting.

I know this issue tracker may not be the best place for build troubleshooting, and I apologize if this crosses that line. I just wanted to clarify my earlier post and share what ended up making the difference in case it helps others.

Virge, 2025-06-20:
> In any case, a message on the LED suggesting a belts tune-up if the printer notices unreliable calibration might be in order.

I completely agree. During the repeated corner probing, the screen remains mostly blank and doesn’t provide any feedback about what’s going wrong. It would be incredibly helpful if the printer displayed some basic diagnostic information during this process, such as which axis is failing to calibrate or if excessive retries are being triggered.

Even just showing the X/Y coordinates during homing (as it does during initial calibration) would give users a clue about what's happening. Right now, it’s just BANG BANG BANG with no context, which makes troubleshooting a guessing game.

Some kind axis-specific hint would go a long way in helping users catch and fix the issues sooner.

Maybe something like this. "Homing failed on [X/Y]-axis. Do you wish to continue?"

BootsSiR, 2025-06-22:
I updated my MK4S to the Core One.  Had my belts tuned to 85Hz .. everything worked great for about two weeks.. lots of prints, no headbanging.... it started doing the headbanging thing last night.. I checked the belts... 75Hz ... odd.. tightened them back up to 85Hz (according to the app), hit auto home, bam, found home in about 5 seconds.  Started a print... headbanging constantly... restarted it, new print, went ok.  Changed filaments... headbanging... new print...headbanging... getting old real fast, especially where it was working awesome for two weeks.

MartinEmrich, 2025-06-28:
After retensioning, realigning the X-Gantry etc. 6 times now (I almost can tune the 85Hz by ear now, feels like tuning a Floyd Rose-equipped guitar) it randomly works or not. 

I use "Auto Home" in the menu and it bumps a few times and is happy, then I immediately try to print and it bumps for 5 minutes before hard-resetting itself and presenting the red "tension your belts" error. 
On the next attempt without touching the printer, it bumps for 1 minute and starts printing, and the print is impeccable perfect.

With all tolerances of the design, a perfect X-Gantry square fit (no gap both right or left) won't stay for long. And I think a tiny bit of tolerance there (<1mm) should be acceptable, the gantry brackets are from springy flexible material (steel sheet) after all.

So I am now of the opinion that the Pass/Fail decision tolerances in that homing procedure in the firmware are just way too strict...

tlf30, 2025-07-15:
Having the same issue. Band new core one. The banging is random. Sometimes it will get stuck and go for minutes before hit the reset button then try again only for it to work after just a couple bangs. Definitely seems like a firmware issue and not a bug you would expect from a printer of this cost.

MartinEmrich, 2025-07-15:
The most important part: The banging seems not to care about equal belt tension, it tests whether the left and right part stop equally at the front end stops. And seems to be overly picky about it.

After adjusting the belts so I have absolutely no play on either side, It bangs only two or three times before printing. My belt tension is no longer equal, but I have to ignore that for now. (Though much play at equal tension, or high tension difference if aligned correctly points to a crooked X gantry and/or un-equal belt length/stiffness, which i beyond scope for this github issue, better discussed on the Prusa forums or r/Prusa3D)

outerSpaceToast, 2025-07-20:
I second this request. A fraction of a millimeter difference between left and right endstop ist enough to cause the printhead go nuts. That is not acceptable in the long term.

M1chlCZ, 2025-07-20:
I also have this issue, what helps is that if you move head by hand further from the corner, then it just bangs twice and its ok. So my suggestion is to move first move head towards the back and then bang. 

I have pre-assembled unit.

uktricky, 2025-07-20:
I had a discussion with Prusa support today on this as mine wouldn't stop at all even after a good few minutes - eventually we used the Prusa app to tune the belts afterwards couple of bangs each axis and printer working well.

zapta, 2025-07-21:
Had the same issue on a new kit and it's a known issue that people keep reporting, e.g. here https://forum.prusa3d.com/forum/prusa-core-one-assembly-and-first-prints-troubleshooting/printing-head-calibration-going-wild/paged/2/#post-758279

A couple of questions for the developers:

1. What is the expected of 'head bangs' during an X/Y homing on a properly configured CORE One printer?

2. Why not just one bang on X and one on Y as many printers do, is it for accurate re-homing for resume-on-power-failure ?

MartinEmrich, 2025-07-24:
I like it for the sole cause that giving users options is always a good thing, while meticulously keeping backwards compatibility.

But I can already see (and relate to) the unavoidable counter argument: Prusa builds a "feature" that "makes the printer worse" (less precise homing) to mask a design flaw". Nevertheless +1 from me.

The precise banging does only 4-10 bumps for me as I put quite a lot of work in alinging the left and right rails, including cold-setting the head rail mounting brackets to be exactly perpendicular under the never-perfectly-even belt length/tension (Which should have been made from a precisely machined/printed instead of flexible metal sheet material in the first place).

If it takes 10s of seconds or even minutes (>20 bumps), this indicates such a mechanical issue, the printer should abort immediately and present a link to the "How to ignore belt Hz values and correctly align the gantry" help page, instead of torturing itself and the user with minutes before hard resetting and presenting a red error.

zapta, 2025-07-25:
@MartinEmrich  said:
> The precise banging does only 4-10 bumps

You can count the exact number of head bumps by invoking the `G28` command from a serial terminal such as CoolTerm (CORE One creates a serial port when connected to a computer via the lower USB connector).  For example, in the log below there are 15 head bumps.  

```
G28
echo:endstops hit:  Y:0.00
echo:Backoff ipos:471
echo:busy: processing
echo:endstops hit:  X:250.00
echo:endstops hit:  X:250.00 Y:-10.08
echo:Backoff ipos:2169
echo:endstops hit:  X:242.32
echo:busy: processing
echo:endstops hit:  X:242.28
echo:endstops hit:  X:252.16
echo:endstops hit:  X:242.60
echo:endstops hit:  X:252.08
echo:endstops hit:  X:242.60
echo:endstops hit:  X:252.16
echo:busy: processing
echo:endstops hit:  X:242.60
echo:endstops hit:  X:252.08
echo:endstops hit:  X:252.16
echo:endstops hit:  X:242.60
echo:endstops hit:  X:252.16
calibrated home cycle A:31 B:-1
echo:busy: processing
echo:endstops hit:  Z:60.00
echo:busy: processing
echo:busy: processing
echo:endstops hit:  Z:60.00
echo:busy: processing
echo:endstops hit:  Z:60.00
echo:Starting probe at 1
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:Probe classified as clean and OK
X:245.00 Y:0.00 Z:2.00 E:0.00 Count A:24500 B:24500 Z:800
ok
Done printing file
```

mkyral, 2025-07-26:
> If it takes 10s of seconds or even minutes (>20 bumps), this indicates such a mechanical issue, the printer should abort immediately and present a link to the "How to ignore belt Hz values and [...]

This don't answer a question why it is ok with first print, but bumps like crazy when I want continue with another print. And why reset usually fix it.

zapta, 2025-07-27:
@CZDanol, how about disabling the homing refinement for the CORE One until this is fully resolved?

The current implementation of the home refinement makes the CORE One a worst product so rolling it back will give us a relief until a verified improvement will be available.

CZDanol, 2025-07-28:
> The current implementation of the home refinement makes the CORE One a worst product so rolling it back will give us a relief until a verified improvement will be available.

I will consult this with our product department (BFW-7537). Please take into consideration that the precise homing is required for the power panic to work correctly. Without precise homing, the printer would not be able to home to the same position on power recovery, resulting in layer shifts. There might be some more reasons for precise positioning requirements, but I can't remember more right now for the C1. On the XL, docking/undocking during toolchanges would not be possible for example.

zapta, 2025-07-28:
Thanks @CZDanol!

The resume-after-power-failure is a nice to have feature but most of us don't care about it. We have a stable grid, we don't print 24/7, and if power fails during a print once in 5 years, restarting it is not a big deal.

Making this small change will reduce drastically the user complaints in the Prusa forums and will increase the overall user satisfaction.

Racemade, 2025-07-28:
Wirst thing eher is, that even your gantry allignment and also the hz frequency are on point, but After second or third precise homing its still headbanging. Sometimes its quick and sometimes there are minutes of banging...

zapta, 2025-07-28:
I wonder why this issue is classified as an **enhancement** and not as a **bug**. This is clearly a malfunction rather than a request for an additional printer feature.

@rinkek, any thoughts on reclassification?

3d-gussner, 2025-07-29:
@zapta 
>The resume-after-power-failure is a nice to have feature but most of us don't care about it. We have a stable grid, we don't print 24/7, and if power fails during a print once in 5 years, restarting it is not a big deal.

There are other areas in the world where short power-outages are "common" and the function may not be useful to you but for these users it saves prints.

mix579, 2025-07-29:
Make "precise homing" an option that can be turned on or off in settings. Being able to recover from a power loss is obviously a useful feature. But as others have pointed out, many (most?) users don't care, and for those of us the headbanging is just a nuisance.

IftiBashir, 2025-07-29:
Another vote here - would prefer the option to switch this off and I would happily sacrifice the precision when it comes to resuming after power outage. The head banging is seriously frustrating.

zapta, 2025-07-29:
> The head would bang for 4 minutes straight until I did those two things. Now it's perfect every time.

Hi @JeffWDH, how many head bangs do you get with the 'perfect' operation? I couldn't get it below 12 head bangs for a single homing operation. 2 on X, 2 on Y, and then 4 X,Y double bangs.

zapta, 2025-07-29:
6 bangs is 3 times more than my HyperCube Evo sensorless homing. ;-)

Here is the experience of the Canuck Creator printing on a preassembled CORE One   https://www.youtube.com/live/byPlvkcSuAw?t=2811s

e-basteln, 2025-07-29:
> I am down to 6 sometime 8 much better experience

Are you sure? I have never seen and heard anything less than 12 bangs: Right, right, front, front -- right, front, right, front, right, front, right, front. Neither in my own Core One, nor in any video. And 12 is already pretty embarrassing and annoying in a "quality" printer for > $1000, in my opinion.

Racemade, 2025-08-01:
Ok now it fu**** up.
Belt tension and allignment was perfect and also there as no headbanging.
Printing ASA, and it banged all day Long. I think the heat changed the allignment.

Its like the XL. The First year you can play bullshit bingo with the machine. Hope it get fixed soon.

lvillani, 2025-08-07:
Just wanted to chime in and say that I also experience homing issues with my self-assembled Core One. I followed the updated belt tensioning guide and ensured the gantry is correctly squared. The first homing is usually fine when the printer is cold. If I let the printer cool between prints, it homes fine. However, if the printer is warm, it recalibrates for minutes, and resetting doesn't always help. It seems to be a temperature-related issue in my case. (Firmware 6.3.4)

zapta, 2025-08-08:
> Its like the XL. The First year you can play bullshit bingo with the machine. Hope it get fixed soon.

The firmware fix is very simple, disabling the non standard X/Y homing refinement algorithm and sticking. Unfortunately we can't do it ourselves since the firmware needs to be signed by Prusa.

Racemade, 2025-08-15:
Fucked up. Headbanging after every Single Print (especially after asa prints)

BootsSiR, 2025-08-15:
This has become a non issue for me after tuning to the new belt frequencies.  Homes perfectly every time.

tlf30, 2025-08-16:
Can we please get a quick fix to allow us to print until there is a better fix? At this point I cannot print PETG, it just bangs all day long. PLA is no issue for some reason. Something to do with the temp difference causing alignment issues? IDK, but basically my $1000+ printer is unusable right now

tlf30, 2025-08-24:
I think I have finally solved this, at least how to not bang starting a print. 
I have found that if I center the print head (both X and Y) axis prior to starting the print, it will only bang the minimum for the calibration. If the print head starts on the side of the printer (which is where it parks at the end of a print) it will bang endlessly until I power cycle. 

I've tested this 3 times now, centering the print head on the gantry, and so far it is working well. The height of the print bed does not seem to matter as I tested at three different heights. 

Perhaps the easy fix would be to have the firmware auto center the print head prior to starting the calibration as the calibration seems sensitive to the starting position of the print head.

marvinoffers, 2025-08-28:
Also having this issue. Bangs for ~2:50 mins, then a collision (that is not happening), then another banging for ~10-30 seconds and then it starts printing without issues.

Prusa-Support, 2025-08-28:
_I would like to highlight that this is ultimately caused by assembly inconsistencies that need to be corrected._

I recognize that belt fine-tuning and X-gantry perfect alignment can be tricky sometimes, but I encourage you to reach out to our Customer Support for help. If anything, this may serve to rule out potentially additional, more or less severe structural or assembly issues.
.

When it comes to continuous head-banging, this is actually related to Precise Homing.
We **don't** plan to make Precise Homing optional for now (not advisable), but we plan to improve this feature and reduce scenarios where Precise Homing is enforced.

Unfortunately, we lack official information at this time, and the current plans may change in several ways. Nonetheless, under normal circumstances, _Precise Homing movements shouldn't be repeated "too many times"_.
.

If Precise Homing takes too long, please double-check belts and X-gantry alignment together with our Customer Support - we can't exclude additional hardware problems.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

ademars, 2025-08-28:
Surely you jest. Many people appear to have the problem after carefully fine tuning the belts. Some people say it goes away after tuning and then returns. Most people say a simple reset while it’s occurring makes it go away. One user says it goes away if he manually places the head in the center before each print. This problem is a huge embarrassment to Prusa. And you’re basically telling us to try harder, the real solution is hard, you’ll address it someday, we likely aren’t tuning it properly even when the problem occurs on the first print from a factory built unit. I donated my perfectly working MK4 to the local robotics club after “upgrading” to the Core One. Your reputation as a solid reputable 3D printer supplier has gone from a 10 to a 3 in my opinion. Do better, like you used to.

e-basteln, 2025-08-28:
> We **don't** plan to make Precise Homing optional for now (not advisable)

Could you elaborate on the "not advisable" bit please? What reasons are speaking against making it optional? I do *not* need power panic recovery, since we don't get frequent power outages in my part of the world. Are there any other benefits Precise Homing provides?

My Core One is well-aligned, by the way. I reproducibly get 12 bangs for homing, which is the optimum to my knowledge: Right, right, front, front -- right, front, right, front, right, front, right, front. I still consider that a nuisance and think that four bangs (as required for the low-precision homing) are more than enough.

tlf30, 2025-08-28:
> _I would like to highlight that this is ultimately caused by assembly inconsistencies that need to be corrected._
>
> I recognize that belt fine-tuning and X-gantry perfect alignment can be tricky [...]

I'm highly disappointed by this response.
First, if there is an assembly issue on factory assembled printers, then a product advisory needs to be issued and support needs to assist with a correction.

Second, as mentioned in previous posts in many cases a simple power cycle of the printer or manually moving the print head will resolve the issue. This to me looks like a software issue.

wbphelps, 2025-08-28:
I'll add my two cents - my C1 prints beautifully, no VFA, but every so often it goes into the "head banging", nothing changed, just randomly, often right after a sucessful print.  It bangs and bangs, no stopping it.  Do a quick power cycle, it restarts automatically, does the homing, 4 bangs, and on it goes to print.  THAT is NOT a sign of mistuning or any misadjustment.  How can it be when after restarting it does the homing flawlessly?  Been a software developer for over 60 years now, and this is a classic example of a bad value in memory which is reset by the power cycle.  NO OTHER ADJUSTMENTS, CHANGES, ETC, just a power cycle and off it goes.  Sorry Prusa, intermittent bugs are the hardest, but own up to it, this is NOT a symptom of misalignment...

bionicbone, 2025-08-29:
@tlf30 funny, before reading this I just did exactly the same thing, reset, centered the head to the middle of printer and started the print again. It homed as normal.

Not a fix because I may not have been sitting next to the printer.

Also note the next print it banged again, after 6 minutes it finished some sort of recalibration and continued. Significantly, I noticed the bed LED turned yellow just before it finished and carried on.

I spoke with Prusa CS and we noticed the left side carriage was not fully touching when fully forward, so we adjusted, note adjusting the left side changed the right side so a bit of playing was needed to get them perfect. Then we got both belts as close to 95Hz as we could (94 & 97), by loosing both side by equal amounts

It still bangs about 6 to 8 times so far but I think that could be acceptable, when new out the box it did 4 bangs consistently from memory.

Time will tell if it is resolved.

jkufner, 2025-08-29:
> I spoke with Prusa CS and we noticed the left side carriage was not fully touching when fully forward, so we adjusted, note adjusting the left side changed the right side so a bit of playing was needed to get them perfect.

@bionicbone How did you adjust for that? I have like 1mm gap on the left side when the right side touches.

jkufner, 2025-08-29:
@bionicbone  Thanks. I was calibrating the belt tension before, but getting the gantry square required the tension to be too different, and the printer was making a bit unhealthy sound when moving.

So, after a bit more reading, I took more direct aproach. I loosened the belts completely and the screws near the bearings on the rods (just to not stress the bearings). Then, I pushed the gantry to make it square, slightly bending the L brackets on the sides of the gantry. It took some force, but after a while I got both sides to touch the end stops at the same time. Finally, I tightened everything and tuned the belts again. It seems better now.

When I was investigating, I also loosened all the screws on the L brackets of the gantry, and I noticed that right-side bracket was not 90° (by about 1-2 mm gap at the end of the bracket, towards the center of the gantry). So, this was what needed to be strightened.

bionicbone, 2025-08-29:
@jkufner 

> When I was investigating, I also loosened all the screws on the L brackets of the gantry, and I noticed that right-side bracket was not 90° (by about 1-2 mm gap at the end of the bracket, towards the center of the gantry). So, this was what needed to be strightened.

That is interesting, CS moved onto checking this next but I ran out of time. I will check mine too just in case.

[13 more replies]