prusa3d/Prusa-Firmware-Buddy #4583: [BUG] Blue screen of death out of the blue

jimokeane, 2025-04-05:
### Printer model

MK4S

### Firmware version

6.2.2

### Upgrades and modifications

None (stock MK4S built from kit)

### Printing from...

USB Drive and/or online

### Describe the bug

I found this blue screen of death on the printer this morning. I had printed something yesterday via PrusaConnect, but that had finished hour before. I dumped the error log file and have emailed it.

### How to reproduce

No idea

### Expected behavior

Not crash

CZDanol, 2025-04-05:
Hello, please provide at least the crash dump, otherwise we have nothing to investigate.

jimokeane, 2025-04-05:
As per bug report directions, crash dump was emailed to Prusa directly,
since it contains sensitive information.