prusa3d/Prusa-Firmware-Buddy #4584: [BUG] [CoreOne] [6.3.1] Printer restarts when door is opened after nozzle cleaning has failed

1Rolf, 2025-04-05:
### Printer model

CoreOne

### Firmware version

6.3.1

### Upgrades and modifications

Directly out of the Box

### Printing from...

USB Drive

### Describe the bug

I took the printer out of the box, copied the firmware update to the USB drive, plugged the printer in, completed the initial setup and started the supplied KeyChain print (single colour).
The nozzle cleaning failed, I opened the door to check, the printer shut down completely restarted after a short time.

The procedure was repeated several times, it restarted printing each time, nozzle cleaning failed, door open, crash. 

If I wait longer until the printer stops moving, then the door can be opened without it crashing

[Video](https://imgur.com/a/AzusNJt)

### How to reproduce

Start the Print with the provided KeyChain file, wait until nozzle cleaning fails dialog appers, open the door

### Expected behavior

No shutdown / restart when opening the door

me25542, 2025-04-05:
Is this related to [#4502](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4502)?

PrintFandango, 2025-04-20:
yes I have the same issue when nozzle cleaning fails.

CZDanol, 2025-04-22:
Hello, could you please send us a log from the nozzle cleaning procedure?

1. Turn on logging to a file in Settings → System
2. Run the nozzle cleaning (possibly as part of the print)
3. When it fails, abort the print
4. Turn off the logging
5. Share the log.txt file

Thanks :)

CZDanol, 2025-04-22:
Relevant ticket: #4523

mkyral, 2025-05-08:
This happen to me as well today. The nozzle cleaning procedure failed. There was a quite big piece of petg on nozzle, so I opened door during repeating cleaning procedure, thought it will stop and I remove the plastic using tweezers. But instead, printer shut down itself and start again after short pause. And then, when I closed door, cleaning procedure continued.

tfvlrue, 2025-05-20:
I'm able to reliably reproduce this before a nozzle cleaning failure. Basically I'm just:

- Starting a print job
- Waiting for the bed to start moving up
- Opening the door

At that point, the printer goes dark and restarts. For reference, the printer is on firmware 6.3.2.

I tried enabling logging per @CZDanol 's suggestion above, however the log.txt I got back was completely empty. I'm not sure if I'm doing something wrong (I enabled logging to file from the settings and then reproduced the issue). I'm happy to provide any other info that would help get this fixed. Thanks!

CZDanol, 2025-06-25:
Logging is heavily buffered and part of it gets saved only when disabling it. Power panic is an emergency procedure that doesn't wait for the logging to flush, so not having any logs is expected in this case.