prusa3d/Prusa-Firmware-Buddy #4585: [BUG] MK4S FW6.2.3+8862 Persistent Y Axis Layer Shift

caprazzo, 2025-04-06:
### Printer model

MK4S

### Firmware version

6.2.3+8862

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

I am consistently getting Y-axis layer shift at the same spot for a specific model, but a different test model prints without issue. 

The printer has performed well for a long time, but I did suffer from layer shift in older firmware that had the issue.

Images show the same model printed 3 times and failing with layer shift at the same height, all in the Y direction. All three have been sliced differently, while the 3rd has been re-sliced at lower speed and layer height.

The model mesh has a high number of triangles, and the printer has not been restarted for a few days at least.

The fourth image is a tall cylinder printed after the failures, showing that the printer does not seem to have an hardware issue.

[image]
[image]
[image]
[image]

[image]

[image]

I can share codes and STLs, but the upload of the zip file keeps failing. 

Thanks

CZDanol, 2025-04-06:
Hello, please share with us the stl and the exact gcode you used for printing. If github upload is not working, feel free to use any file sharing service. Or you can send it via e-mail to info@prusa3d.com

caprazzo, 2025-04-06:
Hi, I sent the files via email.

Prusa-Support, 2025-04-08:
@caprazzo I'm unable to find any result in relation to the GitHub tile or your username.
Please send the email again _to my attention_ at [info@prusa3d.com](mailto:info@prusa3d.com) so eventually it would be redirected to me. My name is in the signature.

_Ideally, please include the name and URL of this GitHub issue!_

Alternatively, send it to **reports@prusa3d.com** as if it was a BSOD dump file.
This way you would automatically get back a _PR- code to to simply share for reference here on GitHub_.
https://help.prusa3d.com/article/blue-screen-of-death-bsod_689421

Michele Moramarco
Prusa Research

caprazzo, 2025-04-08:
Thanks Michele. I sent Gcode and STLs info@prusa3d.com to your attention

CZDanol, 2025-04-09:
Hello, we've received the gcode. There are 4 sliced bgcodes with vastly different print times - which one should we use for testing?

caprazzo, 2025-04-11:
Hi, all 3 gcodes are the sema STL sliced at different speed and layer height. All ended up failing with layer shifting, so you can start from the one that makes your life easier for testing. Thanks!

CZDanol, 2025-04-11:
Okay, thank you for the information. Internal task: BFW-7045

caprazzo, 2025-04-23:
Hi, this happened once more with a completely different model. Should I share the source via email?

CZDanol, 2025-04-23:
I guess we have enough model for testing at this moment, thanks :) We will first try going through what we have.

caprazzo, 2025-04-23:
Posting the latest failure for reference.

There no STL as this os ASCII Gcode generated by a proprietary slicer.

[image]

caprazzo, 2025-04-23:
Thanks for your reply.

This is not actually gyroid infill. It's gyroid lamp shade designed in Grasshopper and sliced with a custom slicer. Of course it is a Gyroid non the less. I see that the latest release is 6.2.4. Do I need to wait for 6.4 for a fix?

Looking at it in detail it is really similar to the previous failures I reported (the latest layer is shifted on the Y) which were sliced with Prusa Slicer and had no infill.  This same model printed without issues just days ago with slightly different slicing and using PLA instead of PVB.

I have had number many similar issues in the past 6 months and I  tried everything I could think of on modeling, slicing and printer configuration, but it keeps happening seemingly at random. I'm now at a the point where I'm scared to trying again a print like the above and waste half a roll and further delay on a committed deliveries. 

Another user reported an issue that sounds very similar to mine: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4532

Any help to alleviate the situation would be really welcome.

caprazzo, 2025-04-28:
Hi, I had another failure similar failure, twice in a row, this time with a **brand new, pre-assembled Core One**. 

This happened right after two successful prints with the same printer. 

I caught on camera - will share stl, gcode and video at the same email as before.

caprazzo, 2025-04-28:
I'll also raise a new issue for Core One specifically

ivous, 2025-05-22:
Hello @caprazzo, I am trying to reproduce the issue with provided UOL_T10025_VA model on MK4S. I have tried to print it 3x with nozzle 0.6mm and all prints are fine. 
However when I print with nozzle 0.4mm, then I can see the layer shift at similar layer as on your side. Could you please let me know which nozzle do you use?

ivous, 2025-05-27:
Hi @caprazzo, thanks for sharing. For your information, we have already excluded the nozzle diameter influence, the failure does not depend on it. 
Testing prints on Corre One shows that this particular model UOL_T10025_VA is printed correctly, so it is impacting MK4S (maybe older as well).
Investigation of the root cause is still ongoing.

Prusa-Support, 2025-09-13:
There is a new final release available for Prusa Slicer ([2.9.3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.3)), and a new firmware pre-release ([6.4.0-alpha](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha)).
Please check out and possibly test our stock profiles - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4573#issuecomment-3288285067.

While this issue is quite specific to the printer model MK4S, please note that the release notes of firmware 6.4.0 briefly mention bugfixes that are expected to reduce uncontrolled layershifting scenarios.

Michele Moramarco
Prusa Research

closed 2025-12-29 by github-actions[bot]