prusa3d/Prusa-Firmware-Buddy #4592: [BUG] MMU bowden length register does write and read, but no effect.

Voxel3DPrinting, 2025-04-09:
### Printer model

CORE one

### Firmware version

6.3.1

### Upgrades and modifications

COREBOXX

### Printing from...

prusa connect, serial coms via Putty

### Describe the bug

Hi,

I am playing around with the MMU on CORE one, which is supported by the 6.3.1 firmware, but I am running into some issues regarding changing the Bowden feed length

**My goal;** 
Having the MMU load 800mm quickly, then 100mm slowly into the extruder.

**Done to get the goal;**
- Increase the Bowden length via `M708 A0x22 X380`
- Verified correct register programming via `M707 A0x22`, which reads the newly programmed value correctly

**Gotten result;**
- Filament still only loads the first 300mm-ish quickly, then proceed to load the remaining 600mm-ish slowly
- Tried with "load to nozzle", "Loading test", and print

### How to reproduce

- Install MMU3 on CORE one, with the latest firmware on both
- Increase the registered bowden length via `M708 A0x22 X380`
- Verify correct register programming via `M707 A0x22`, which reads the newly programmed value correctly
- Try to load to nozzle

### Expected behavior

-  Filament loads the majority of the filament quickly, followed by the last section slowly. This is verified working on MK4S with same bowden length

DRracer, 2025-04-09:
`M708 A0x22 X800` is the correct setup IMHO

Voxel3DPrinting, 2025-04-09:
yeahh I also asked on the forum. 

The value should be in decimal, not hexadecimal. Confusing information on the forums, but it works now! Thanks!

closed 2025-04-09 by Voxel3DPrinting