prusa3d/Prusa-Firmware-Buddy #4593: [BUG] CORE One, Filtration Fan Not Spinning Error

christopherv5, 2025-04-10:
### Printer model

CORE One

### Firmware version

6.3.1

### Upgrades and modifications

Advanced Filtration System

### Printing from...

USB Drive

### Describe the bug

Given that the MK4 Enclosure blower is the same model (confirmed by support), I designed a bracket that will attach to the CORE One similar to that of the official filtration system.

When enabled in the firmware via the settings page, you can visibly and audibly hear the fan spinning, although the printer states that the fan isn't spinning and to check it. 

Seems like the printers isn't reading the blower correctly. 

### How to reproduce

Connect the blower fan > enable advanced filtration system > let run for 30 seconds before seeing an error message.

### Expected behavior

Blower fan should be read correctly if there isn't anything obstructing.

CZDanol, 2025-04-10:
Hello, there might possible be an issue with your fan (not sending the RPM readouts correctly). Could you check what the fan RPM is showing up in the Sensor Info screen?

yawor, 2025-04-17:
I'm the one that reported that issue to @christopherv5 on Printables. For now I've disabled fan check so the error doesn't show up, but during printing I've opened the Tune menu and then into Info screen. The filtration fan power and RPM is being reported there. It showed 71% and the RPM is oscillating around 2645 RPM +- 10 RPM more or less. But from time to time it shows either over 50000 RPM or **stuck** for a brief moment and then goes back to normal value. It looks like some pulses from the fan may not be counted properly.

CZDanol, 2025-04-17:
Internal ticket: BFW-7090

CZDanol, 2025-04-25:
Hello, so while the support guy was correct - it is indeed the same fan model - the MK4 variant has a longer cable than the XL one (which we are also using for the official C1 filtration system). This extra length seems to be introducing enough noise to the tacho wire of the fan so that the processor is sometimes picking up garbage.

My quick test seems to indicate that this might be solved by simply enabling a filter on the processor, so I will try to get that into our next firmware for you to test it.

If you don't want to wait, shortening the wire should do the trick for you as well.

CZDanol, 2025-04-25:
Btw, it is also possible to disable the fan checks using a toggle in the Settings screen. This will however disable all checks, so it's somewhat dangerous.

yawor, 2025-04-25:
@CZDanol thank you for that info. That actually explains a lot, especially that all that extra wire is tucked above the board compartment where there is a lot of other signal cables and they may interfere with each other.

If I were to shorten the wire, can you tell us what is the part number for the original fan plug that plugs into the extension board? I'd rather like to crimp a new plug on the shortened cable than to cut in the middle and splice it together.

CZDanol, 2025-04-25:
MOLEX 502578-0400

yawor, 2025-04-25:
Thank you very much

rainer1146, 2025-05-05:
I also have this problem. If it could be solved via the firmware, it would be very helpful for many who want to continue using their MK4 fan.
Thank you very much.

christopherv5, 2025-05-06:
If we cut the cable, what would the proper length be?

yawor, 2025-05-06:
About 25-30 cm (measuring from the edge of the fan) seems to be enough.

I haven't done mine yet, as I wanted to crimp the new connector instead of cutting the cable in the middle and splicing it together, but Molex ClikMate connectors are not easy to work with. I've hoped that my crimp tool would be able to crimp them, but the pins are way too small for it.

In the end I've decided that I don't want to shorten the original cable. Instead I've just ordered pre-crimped black 24AWG (0.25 mm²) wires, 30 cm in length and I'm waiting for them to arrive. Also I've bought the BLHP2432-H24 blower fan directly from Alveo3D (Prusa's is BLHP2432-H20) and I'll just solder the pre-crimped wires to fan's PCB (under the sticker). I'll just sell the original fan together with MK4 and the enclosure.

rainer1146, 2025-05-06:
The original has 0.5 A, the BLHP2432-H24 has 1 A as far as I know.
I would be concerned about the board.
I trust that there will be a simple solution via the firmware.

yawor, 2025-05-06:
@rainer1146 that's true, but according to xBuddy Extension Board's schematics, the 24V power supply for each fan switched on by BUK6D120-60PX MOSFET, which is rated to up to 8 A, directly from the main 24V rail. Unless the traces are really thin I don't think drawing up to 1 A instead of 0.5 is going to break anything. This doesn't affect the PWM signal either, as there's not much current draw on that line. The fan has more power, so I can always just set the minimal speed to lower value than with the BLHP2432-H20.

https://www.prusa3d.com/downloads/Electronics_drawings/MK4-xBuddyExtension-06.pdf

yawor, 2025-05-07:
I've received the pre-crimped wires today. I've soldered them to BLHP2432-H24 and installed the fan on the printer. The cable length is exactly 30 cm from the fan's hub to the clikmate plug.

A big WARNING: if anyone like to use a non-original fan like the Alveo3D one for example, please make sure you check the order of the wires on the motor hub if you decide to solder new wires to it. Depending on the actual OEM manufacturer the order may differ even if the sticker and fan model seems to be the same!

I've enabled the filtration fan in the Control > Temperature menu and run it for some time at different speeds while monitoring reported fan speed in Info -> Sensor Info menu. I didn't notice any erroneous readouts, the speed is stable. I haven't tried printing yet to check if I get any filtration fan errors again, as it's already late here today. Unfortunately I'm going away for a week tomorrow, so it has to wait, but I think it should be alright.

I've also checked the board with a thermal camera after the fan has been running for some time. The fan is definitely more powerful, but fortunately I haven't noticed hot spots at all near the FAN3 socket, where all the fan control logic sits, especially no issue with the MOSFET which provides 24 V for the fan.

As I've mentioned, the fan is more powerful. On Prusa's one I've got about 2600 RPM at 70% power. This one got 3600 RPM at 30% power and spins up to 5130 RPM at 100% power (it seems it doesn't scale linearly under load when pulling air through the filter). Of course it is more noisy because of that, but on lower power when the RPM is similar, the noise level is also similar between both fans. But it's nice to have an option to spin it faster and pull much more air from the print chamber.

Prusa-Support, 2025-05-30:
This issue could be closed as solved by FW 6.3.3, which mitigates chances of potential false fan errors.

> Added signal filtering to reduce false fan errors on DIY filtration setups using MK4 fans with longer cables. This should resolve tachometer signal issues.

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.3.3

However, final feedback would be appreciated.

Michele Moramarco
Prusa Research

daafgo, 2025-06-18:
I still have this problem in my core one, the printer says that one of the fan it's not  spinning but it working fine