prusa3d/Prusa-Firmware-Buddy #4596: [BUG] CORE ONE not able to home Z when nozzle heater is turned on

Voxel3DPrinting, 2025-04-12:
### Printer model

CORE one

### Firmware version

6.3.1

### Upgrades and modifications

MMU

### Printing from...

USB or Connect

### Describe the bug

When starting a print, my CORE one is not able to home the Z axis reliably, it has many false-positive triggers, and keeps moving up and down randomly until it throws a collision error, which it of course cannot resolve.

Video;
https://photos.app.goo.gl/1KD3DmwBA4T5oRCz5

However, the machine passes all tests and calibrations and can home perfectly fine when homing via the LCD, issuing solely a G28 via serial, or writing GCode.

### How to reproduce

I verified and tested loads, and found the following;

When the nozzle heater is turned on, the Z axis cannot home.
 Therefore, an easy way to test is to send this gcode to the machine;

```
M109 S200
G28
```

This will cause the issue seen in the video above

When the nozzle heater is off, it homes fine in whatever way I can find and test

### Expected behavior

Homing just fine with the Heater set to whatever temperature

### Files

I do not know if this is an electrical issue with Noise inside of the printhead main cable or the loveboard, or if it is solely a firmware bug regarding sensitivity of the loadcell/ PWM output.

other things I ruled out;
- I hard reset the machine
- redid all calibrations
- Lubed all threaded rods of Z
- aligned nuts, spindles and motors perfectly
- Bed heater does not affect the issue
- E-motor/ disable all steppers does not affect the issue
- Checked for pinches in loveboard cable
- Loosening and tightening the 3 bolts that hold the heatsink to the carrier

Voxel3DPrinting, 2025-04-13:
Similar issues happen on other nextruder machines;
CORE one;
- https://forum.prusa3d.com/forum/prusa-core-one-assembly-and-first-prints-troubleshooting/z-axis-trouble-but-only-when-printing/

MK4((S)
- https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/mk4-probing-for-plate-in-mid-air-issue-homing-z-axis-when-starting-a-print/
- https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/loadcell-issue-mk4/
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3392

Voxel3DPrinting, 2025-04-13:
Following my posts above, I have proceeded to try and fix my  issue (since I really want to use the machine for continued development of my own projects)

**Left-over possible root causes of the issue, based on concluded findings above;**
- 1) Thermal expansion of nozzle section could result in changed strain
- 2) Position of the printhead during probing could result on different strain
- 3) Electrical Noise in the loadcell signal

**I proceeded testing all of the above;**
1) 
- Tried removing the heatsink fan to scale expansion, no effect.
- Tried to let the heatsink soak and stabilise for 40minutes, no effect
- Excecuted custom code that set the heatsink temperature target to 80C, to prevent fan from interfering, no effect
2)
- Probed via handwritten gcode in 9 different spots, no effect ((makes sense since the HX717 detects peaks instead of threshold
3) (was more plausible for me, since I have been involved in 3D printer PCB design before, where electrical noise exceeded 70V peaks at maximum, when unfiltered.)
- Tested by referring to the J13 connector pinout of the LOVEBOARD [found here](https://www.prusa3d.com/downloads/Electronics_drawings/[FDM-MK4-LoveBoard](https://www.prusa3d.com/downloads/Electronics_drawings/FDM-MK4-LoveBoard-38.pdf)-38.pdf).
I learned that the data (and E-motor wires are twisted white/black, heater wires are black only, I separated the twisted from the untwisted wires and found that it did not do much, although it did affect it a bit, but irregular.

Now my assumption is that there is some noise that interferes inside of the Loveboard or Xbuddy

Voxel3DPrinting, 2025-04-13:
> Might be [#4571](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4571)

oh my god, look at my reply I just posted, was writing the last sentence when I saw your comment!

So then yes, it indeed looks like it is related to noise caused by the heater PWM. Unfortunately, tho, I cannot compile code (hardware guy, not a huge firmware fanatic).

Voxel3DPrinting, 2025-04-13:
> > oh my god, look at my reply I just posted, was writing the last sentence when I saw your comment!
> > So then yes, it indeed looks like it is related to noise caused by the heater PWM. Unfortunately, tho, I cannot compile code (hardware guy, not a huge firmware fanatic).
> 
> You could try taking out pin 7 (analog ground) from the nextruder wire connector (connected to the xBuddy board) and wire it to a better ground somewhere on the xBuddy board ([schematic](https://www.prusa3d.com/downloads/Electronics_drawings/FDM-xBUDDY-44.pdf)). You could fry something if you don't know what you're doing, so be warned.

ohkee good idea, Pin7 at Xbuddy side I then assume? But where inside of the machine could I find a better ground? also, wouldn't it be easier/ less risky to upload a less sensitive firmware?

Kachidoki2807, 2025-04-13:
Keeping an eye on the loadcell output from the sensor info menu gaves me a lot of interesting data.
Especially the value variation on a finger press is around 250. But when moving the head around the bed it can goes way above this threshold. When heating the nozzle too.

My idea is that the cable bundle plus the pneufit and PTFE are hardly constraining the loadcell because they are attached to the top of the aluminium heatsink. You can easily see yourself that slightly shaking the cables leads to very big variations on the loadcell output.

See my related ticket #4555.

CZDanol, 2025-04-14:
Internal ticket: BFW-7075

Shikadi297, 2025-04-23:
I have the same issue, a workaround is to go to the tune menu and set the nozzle temperature to 0 when it's trying to home. It will succeed quickly and set the temperature back to another value on the next step

swaggner, 2025-05-04:
> Could you guys please test the following?
> 
> * Have the "log to txt" in Settings → System enabled during the failing bed leveling, send the log file here
> * Check that all connectors to the toolhead are properly clicked in, and that none of the wires are pinched
> * Check that the heatsink is attached to the 3D printed gantry piece with a plastic spacers, not metal ones

All connections and wires look good.  I’m not sure about the spacers you’re referencing, but I got an assembled version of the printer so I’m assuming PRUSA used the correct ones. I’ve attached the log.  Thanks for your help. I’m anxious to get an official solution. 

[image]

[log.txt](https://github.com/user-attachments/files/20029746/log.txt)

XSIS-Energy, 2025-05-06:
I have exactly the same problem, any solutions for this bug?

klibardi, 2025-05-06:
I can confirm exactly the same behavior on my machine. **It must be interference from the pwm** as solely setting nozzle temp to zero solves Z-homing and nozzle cleaning issues (very frequent failures due to detected collision)

a-pushkin, 2025-05-10:
Exactly same issue with firmware 6.3.2: severe Z-axis problems (problems whenhoming, failure to clean nozzle, retries when doing bed leveling) when the nozzle heater is on.

Turning off  the nozzle heater solves all these problems even though nozzle is still hot.

Here's my forum thread about the issue - https://forum.prusa3d.com/forum/prusa-core-one-assembly-and-first-prints-troubleshooting/z-axis-trouble-but-only-when-printing/#post-749398 It seems that I'm not the only one having it.

Prusa-Support, 2025-05-10:
> Could you guys please test the following?
>
> * Have the "log to txt" in Settings → System enabled during the failing bed leveling, send the log file here
> * Check that all connectors to the toolhead [...]

Said spacers (heat-sink — x-gantry) are supposedly the ones depicted in the assembled step 6.23.
https://help.prusa3d.com/guide/6-nextruder-assembly_849034#849995

Thanks for your cooperation.

Michele Moramarco
Prusa Research

XSIS-Energy, 2025-05-10:
[log.txt](https://github.com/user-attachments/files/20140442/log.txt)

i checkt the complete nextruder but anything is fine, I post the txt file

beatwinthewave, 2025-05-18:
On the forum, two users report having replaced the main cable and having resolved the problem
https://forum.prusa3d.com/forum/prusa-core-one-assembly-and-first-prints-troubleshooting/z-axis-trouble-but-only-when-printing/paged/2/

drg407, 2025-05-18:
> On the forum, two users report having replaced the main cable and having resolved the problem https://forum.prusa3d.com/forum/prusa-core-one-assembly-and-first-prints-troubleshooting/z-axis-trouble-but-only-when-printing/paged/2/

That was me, support had me replace the heatsink/loadcell with no luck. Then they had me try m104 s0 before mbl but that didn't seem to work. Then I tried it after every m109 command and that allowed me to complete homing/nozzle cleaning without errors. Support then determined that the main harness could be faulty/have interference and sent me a new one. I replaced the harness, and tightened the plastic posts for the heatsink with a wrench just in case hand tightening was not sufficient. Besides that I didn't deviate from the assembly instructions. It's been about a week since I've had the new harness in with zero collision errors on z homing and nozzle cleaning.

aclii, 2025-07-23:
I haven't assembled my Core One yet, but it was interesting to read about this bug so I decided to check Loveboard grounds and the main cable.
It looks like there is no electrical connection between heater ground and analog ground on the Loveboard. Looking at the main cable you can see some twisted black/white wires and straight black wires:
 - extruder wires (4) are twisted in pairs for each phase to reduce EMI on other wires in the cable.
 - RS-485 serial communication (4) are twisted in pairs to improve noise rejection on differential lines
 - For some reason heater wires are just straight though even though we have quite powerful low frequency PWM running on them along 1+ meter cable.

I would try to twist heater wires in 2 pairs (+24 and GND) and then twist pairs to minimize possible interference on other wires in the cable.

[image]

RPospisil, 2025-07-27:
> Hi, I've just rebuild to Core One and exactly the same issue. Tried all hints mentioned here and in referred links, but only what works is to turn off heating during homing.

Update: I've received a replacement cable from Prusa and the issue disappeared.

bkerler, 2025-08-19:
I just tested on my machine and running 
```
M109 S200
G28
```
works fine as expected on my Core One + MMU. Thus this seems to be no SW-related bug as it seems but HW issue that should be resolved using Prusa Support (most probably cable harness issue as already stated).
So I assume that this fw issue can be closed as it is hw related ?

GWdd, 2025-09-07:
Is there a repository of HW bugs requiring action, as in a 'recall' where users can know by serial number if their cable harness is bad?  I have a new Core One and I'm experiencing nozzle cleaning errors. It'd be nice to know if I have one of the earlier 'bad' cables.

GWdd, 2025-09-10:
Factory assembled printer. No reason to doubt the assembly or blame a "bad user" of things.  And since Prusa has acknowledged changing the cable to a better and more reliable design, I am simply asking do I have the ealier questionable design or the new design And I am having problem that match the symptoms listed above. And since I bought an assembled printer, I am not going to do work that should be covered under warranty.

bkerler, 2025-09-10:
@GWdd sure, totally understandable. Contact Prusa Support, I'm sure they can help you ;)

Prusa-Support, 2025-09-26:
> Prusa Support Chat no longer works, so I sent emails but haven't received a reply; plus, since I purchased from Printed Solid, Prusa CZ isn't my support channel. Printed Solid doesn't claim to know about this issue. I'm sort of in a black hole here.

This is probably off topic, and I beg you all pardon, but I need to ensure that direct support (<ins>not on GitHub</ins>) is not denied.

- The chat service has never been interrupted. If chat does not show up at the bottom left, on our site ...
  - Make sure to accept 'functional' cookies in 'cookies settings' (bottom of the page).
  - Perform Hard Refresh of the web page via `CTRL + Shift + R`.
  - Try alternative browsers.
  - In case of need, use this direct link to open a chat - https://direct.lc.cat/9644675 - which is usually available at the bottom of our emails.
- Emailing works too. The response time can't compare to live chats but it is actually recommended for advanced cases and we must sort out all the emails that we receive.
- Finding solutions for virtually all cases is a mandatory policy for Printed Solid too. Warranty application modes may slightly differ but Prusa Support remains available for direct help anyway. 

Michele Moramarco
Prusa Research

3d-gussner, 2025-10-16:
@tgeneste Thanks for sharing this detailed information and I am happy you could solve the issue.
The hardware, electronics and development teams are aware of this approach of @aclii and look it to that.

andro12368, 2025-10-18:
I have the exact same issue on my Prusa core one kit. I contacted the support and they are sending me a new main cable, I just  hope that they already did that twisting thing to reduce interference because of the PWM noise, or couldn't they just turn off in the firmware the nozzle heater during that 20 seconds of homing?

che2cbs, 2025-11-17:
I just built my Core One kit and immediately had intermittent problems with the nozzle cleaning, probing the bed for level, and occasionally Z-homing.  When it did work it would produce great prints but it failed more often.

After trying a few troubleshooting steps in the Prusa doc I found this thread.  Instead of twisting the cables like @aclii and @SoftwareRenderer, I used two small barrel-type ferrite chokes to isolate each pair of heater wires near the xBuddy board.  They were super easy to install and appear to have fixed all of my issues.   I did have to disconnect the main cable from the xBuddy board to attach the chokes.  So, there is a chance reseating the main cable was the fix.

Gman2606, 2025-11-30:
@aclii big thanks to you, and all the community here!
It's really disappointing to find out this issue after completing the assembly...
Doing the harness trick is annoying but I'm happy it solved the issue🙏🏻

petrkurka, 2025-12-05:
I’m experiencing the exact same issue. With the nozzle heating on, the Core One struggles to home the Z-axis, but with heating off, Z-homing works fine. The higher the nozzle temperature, the worse the homing. However, if you first heat the nozzle, then turn off heating and immediately start homing, it works = so it’s not caused by thermal expansion of the Loadcell. (Firmware 6.4.0)

If the Core One is pre-homed before printing, it finishes print Z-homing before the watchdog timer overflows and the Z-homing error is not triggered (because the Z distance is shorter). However, the issue then appears as frequent “nozzle cleaning failed” errors and excessive bed-level probing touches, etc... Checking the raw Loadcell values confirms that, with the nozzle heating on, they jump all over the place = the printer is constantly detecting false touches, which causes all sorts of problems.

Prusa support responded very kindly and sent me a replacement cable free of charge, but that did not fix the problem. What finally solved it was twisting the nozzle heating cables together to reduce electromagnetic interference. I really want to thank the community for finding this solution. @Prusa-Support, please take this problem into account, because aside from this, the printer now works really great.

The worst part is that the issue likely varies from unit to unit due to production randomness: if the nozzle-heater and Loadcell cables run very close together along the full length of the print head cable, you get frequent full Z-homing errors; if they’re only occasionally close, it mostly shows up as nozzle-cleaning errors; and if you’re lucky and they’re well separated, the printer works normally.

SOLUTION:
A quick and dirty software fix would be to disable nozzle heating during Z-axis homing. A better solution is to prevent electromagnetic interference from the nozzle-heater cables on nearby sensor cables.

ojkoenig, 2025-12-26:
Twisting the cables as described above by @SoftwareRenderer solved the problems for me, thanks a lot!

Gaugeforever, 2025-12-29:
> Here's a pic of the ferrite choke I used:

Is there any chance you could be more specific about which wires to place the ferrite around?  This would be very helpful to me as I have this issue intermittently and do not like the idea of removing the main cable to twist pairs heh.  Thank you for your contribution either way.

Gaugeforever, 2025-12-30:
After looking at photos, if you look at the cable connector head on with the connection lever on top going left to right, pin 1 and 3 as well as pin 2 and 4 are the wires that people are twisting or putting ferrite on.  Please correct me if this is incorrect information.

Surgikill, 2026-01-01:
> If you don’t want to use ferrite or twist pairs, which is completely understandable, you can add custom gcode to your slicer to disable temperature during homing and probing. I think most people here are looking for a hardware solution.
> 
> You can find more details on the gcode [here](https://forum.prusa3d.com/forum/postid/751199/).

It would be great if heating was only disabled for the probe move, but that's a firmware level change.

Surgikill, 2026-01-07:
I'm back to having nozzle cleaning and bed probing fail after twisting the wires together. Something to note: I serviced the gearbox for the nextruder. I'm thinking that the torque of the screws on the gearbox cover is critical for the load cell. I'm getting really frustrated with this machine.

petrkurka, 2026-01-08:
I see I was a bit misleading. Sorry. Please do not use kitchen aluminum foil. I meant proper aluminum foil tape. waffles881, thanks for pointing that out.

waffles881, 2026-01-08:
I tried moving the chokes and placed 1 choke on 1 & 3, and 1 choke on 2 & 4, and it does not seem to have improved my false triggering of the loadcell during homing when the heater is on.  I am going to try twisting and shielding the wires next.

L-a-n-g-o-l-i-e-r-s, 2026-01-11:
@Surgikill Hi there, just curious, but are your twists evenly distributed? I used a sharpie to help me distribute them more evenly. I haven't yet put it back together, but, I put 80 twists into it on accident and then counter twisted with 40 on the pairs, I may have reduced the length of the cable too much. I guess I'll found out soon enough. (I think people said they put half as many twists in both steps.)

I had thought about using my drill's chuck but I think it would crush the crimps. I did wonder about maybe putting something tough over that wouldn't squish them and then using tape to sort of get the wires to not move inside of the circular object. If I have to undo and redo it all I might just go that route as it should be quite uniform in twisting too... that said, I wouldn't recommend doing it that way as I haven't tried it.

jsmith59, 2026-01-11:
I tried using the two chucks on the two heater wire pairs (1 & 3, 2 & 4) but that didn't seem to do anything to alleviate the problem. I then decided to remove the pins from the housing and try the twisted pairs and then twist those together as has been mentioned previously. Since I had a chuck from the previous attempt, I put that on the twisted bundle of wires. Everything is working perfectly so far. Don't know if the chuck did anything but leaving it on for now.

josy1024, 2026-01-18:
i had to **re-adjust** X/Y **belt tension** and now it seems to work as expected. 
(i used manual resonance method becaus i cann see/feel the vibrations)

seems something happed (filament jammend on spool), or there is a change after der first 200 hours "running-in" printing time ... 

Firmware 6.4.0, Kit 2025/12

question: is there a service/check intervall for the core one?

colorblindead, 2026-01-21:
@L-a-n-g-o-l-i-e-r-s i put about 50-60 twists in each pair and then about 40 between the 2 pairs. Ferrite tape on loveboard end and ferrite chokes on xbuddy end. However this was done on the replacement cable support sent me. My sheer laziness compelled me to do all the fixes in one go of if I was going to swap the main cable anyways. So far so good.

cjkeithley, 2026-08-31:
I’m also getting a collision during z-axis homing.  A check of the loadcell values appears to show dramatically different values (-6400 vs. 1.4) depending on whether the nozzle is warm.  I’ve also received noisy loadcell errors a few times when doing the loadcell test.

Sadly, I only received a single reply when I submitted a support request.  That asked for a video showing the z-axis collision detected error, which I provided. Unfortunately It’s been more than week and they still haven’t responded. I’m at a loss as to how to proceed.

savely-krasovsky, 2026-09-02:
Upgreaded mine to Gen 2 yesterday and suddenly have this exact problem. Tried to proceed the print anyway and got collision with the print in the process... Never had any problem previously.

cjkeithley, 2026-09-02:
I’ve heard back from Prusa and they’re investigating the problem. No solution as yet.

The earlier comments for this issue made me wonder if the nozzle temp was somehow affecting the loadcell, which prevents z axis homing and causes false collision errors. So I tried turning on the nozzle heater (~90C) and checked the loadcell readings; there is a significant difference in the loadcell reading when the nozzle heat was off/cool and when it was 90C.  I then went to the control page and did an ‘auto home’… after successfully homing the x & y axis the printer struggled to home the z axis, when it gave up I got a red screen with ‘home z error’ error and a message saying to check the loadcell!

This behavior seems to match the earlier comments that the nozzle heater power (pwm?) affects the loadcell reading.

I’m keeping my fingers crossed that Prusa will ultimately have a solution! They been really responsive after the initial delay in replying to my first video email.

Soveu, 2026-09-02:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4596#issuecomment-5014573537
So, after replacing the cable and the loveboard, the issue was solved. Did not have any issues in the last weeks.

savely-krasovsky, 2026-09-02:
Did they changed something with LoveBoard design or they send it "just in case"?

Leeooww88, 2026-09-06:
Thanks to @aclii i can actually use my printer! Since prusa basically told me to fuck off and wound't help with the printer not being able to home on any temp beyond 80. 
I twisted the cable pairs and also cleaned up in the electronics bay since the moron putting my printer together routed the bed power through the cable harness...i can only guess this also put some emi into the mix.

Now i just have to solve the useless filament sensor and replace the useless ptfe couplings...never ever ever buying anything from prusa again.

mish358, 2026-09-11:
I had the same problem described here.

I had already noticed that the manual homing worked while homing during the pre-print routine failed when the nozzle was heating. Finding this issue helped me confirm that I was looking in the right direction and made the troubleshooting much faster.

A simple workaround that works reliable for me is to switch the nozzle heater off directly before G28 and switch it back on afterwards. This can be added directly to the start G-code in PrusaSlicer, so there is no need to manually edit every print file:

M104 S0
G28
M109 R{first_layer_temperature[0]}

The placeholder automatically uses the correct first layer nozzle temperature from the selected filament profile, so it also works when changing materials. 

The nozzle does not need to cool down, the important part seems to be that the heater is electrically off during homing.
Since I made this change, the printer has been working normally for me.

To be clear: this is only a workaround, not a root-cause fix. It avoids the problem during homing, but the underlying electrical interference between the heater and the loadcell is still there.

Thanks to everyone in this thread for the information. It definitely saved me a lot of time during troubleshooting.

cjkeithley, 2026-09-18:
I also found it necessary to add a “M104 S0” right before the bed leveling, otherwise bed leveling fails. Basically, turning off power to the nozzle heater before the printer needs to use the load cell (during homing & leveling) made all the difference; my prints now turn out perfectly.

I wonder if the INDX upgrade has the same “providing power to the nozzle heater affects the load cell signal quality” problem.