prusa3d/Prusa-Firmware-Buddy #4603: [FEATURE REQUEST] Support Nylon for coldpull since PLA breaks too much when doing so

5jvm0u4, 2025-04-14:
### Printer model

MK4S

### Describe the feature

Support Nylon for coldpull since PLA breaks too much when doing so, PLA is too brittle for this job, even Joel from 3D printing nerd suggest using Nylon(PA) for this job.
Please, the nozzle clogs too much that I need to do this weekly, I'm using Polymaker PolyLite, this should not be this bad.
https://youtu.be/MGt8cnVIKW8?si=H4gi4Fu6dJD4XMcI

### Expected functionality

Support Nylon for coldpull

u89djt, 2025-04-14:
(From a fellow user) It definitely makes sense to be able to do cold pulls with nylon, but for your specific situation, there's a likely remedy while you wait for that to happen.
PLA is brittle when it's not dry. If it snaps like spaghetti, it's wet. If you try to snap PLA in good dry condition, it will just bend. If you're having to do a lot of cold pulls, that suggests your filament is generally wet and brittle, and therefore fragmenting in the grip of the extruder's gears, because needing to do frequent cold pulls isn't normal. Wet filament is a normal problem when the air it's stored in is humid.
A new roll of PLA might not be dry. It's crossed an ocean with a single packet of dessicant that might have filled up. I've had to dry some rolls fresh from the packaging. They're always better after drying. I use sunlu dryers because they were cheap one day. Anything that holds a steady temperature, ideally with a fan, is fine.
An experiment for you: cut about a foot of PLA and hold it at 50 degrees C or local equivalent for an hour or two, then see how that behaves.

5jvm0u4, 2025-04-14:
Got it, looks like I falsely assume a spool of "Polymaker" will be dry when sealed, I dry my other filaments but not with PLA, I'll try it sometime.

But still, Nylon please, it makes more sense then only having PLA.

u89djt, 2025-04-14:
I wonder if cooling from nylon temperatures has to be handled differently because of increased risk of softening the structure of the tool components.

5jvm0u4, 2025-04-14:
What do you mean by softening the structure of the tool component? You mean PETG printed parts? If so it shouldn't be a problem, PA6 prints at 260C

u89djt, 2025-04-14:
Yeah, PETG glass transition temperature is about 85 degrees C. It'll writhe if it gets much higher than that if it hasn't been annealed, I think. No problem when the cooling fans are running in room temperature air.

5jvm0u4, 2025-06-16:
an update

Ro3Deee, 2025-12-30:
Yes, support for cold pulls for all common materials should be added (PLA, PETG, TPU and PA). This is a valid request