prusa3d/Prusa-Firmware-Buddy #4605: [BUG]

Chips83, 2025-04-14:
### Printer model

Core One

### Firmware version

6.3.2

### Upgrades and modifications

none 

### Printing from...

None

### Describe the bug

When you choose custom under preheat menu then the printer reboots automatically and boots into a BSOD. Then you have to restart and the printer works again. 

Issue is reproducible. 

### How to reproduce

Preheat-Custom and then crash

danopernis, 2025-04-15:
@Chips83 are you by any chance using language other than English? Perhaps the one with really long words? We are aware of similar issue when running firmware in German language and we have a fix ready for next release.

Chips83, 2025-04-15:
Yes, I use the german firmware. The word is "Benutzerdefiniert". 

How long will it take for the next update?

danopernis, 2025-04-15:
I learned to not give any estimates, because the decision is not only up to developers. I personally hope it will be a few weeks maximum, but that's as much as I can say 😐

LukaszMoskala, 2025-04-24:
Same thing happens when GUI language is set to polish. However I can still set temperature using the "controls" menu instead, so it's not a big problem for me personally.