prusa3d/Prusa-Firmware-Buddy #4607: [ENHANCEMENT] Add compatibility with .15 and .1mm diameter nozzles

nesdnuma, 2025-04-15:
### Printer model

XL MultiTool

### Describe the enhancement

Hello,

With the V6 Nextruder adapter, it has become possible to install a .15 or a .1mm nozzle on the XL. However the firmware won't go below .2mm when setting the nozzle diameter for each tool.

Prusa-Support, 2025-08-09:
Hey @bkerler, I'm not a fan of incredibly small nozzle ⌀ (I would rather go for [SL1S](https://www.prusa3d.com/it/prodotto/original-prusa-sl1s-speed/)), but I'm not amused by huge nozzle ⌀  either. However, you have recently worked on a similar feature (higher max nozzle diameter https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/4565) and I wonder if this feature request may be of interest to you.

Michele Moramarco
Prusa Research

bkerler, 2025-08-09:
Me neither. Sure, I can add support for it, the question is, would it work with such a small diameter ? I'm thinking about clogging issues mainly

3d-gussner, 2025-08-14:
I created an internal ticket.

Please understand that these nozzle sizes:
- Have a higher risk of clogging
- May cause layer to layer adhesion issues on large prints (XL) where the previous "thick" layers cools down more than usual, because of a "long" print time for the fine details.
- There will be **NO** 0.10 or 0.15 Nozzle diameter profiles from Prusa in PrusaSlicer. Prusa sells 0.25, 0.4, 0.5, 0.6 and 0.8 https://www.prusa3d.com/product/nozzle-v6-0-4-mm/ nozzles and only these are officially supported.

Prusa-Support, 2025-10-03:
The [new firmware](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases) 6.4.0 supports smaller nozzle diameters in the _toolhead hardware settings_.
This firmware release is currently available as a Release Candidate for CO, MK4/S, MK3.9/S, and Alpha version for XL and other supported printer models.

Please consider nozzle diameters that are not listed among Prusa Slicer pre-sets as experimental: please use at your risk.

Michele Moramarco
Prusa Research

closed 2025-10-03 by Prusa-Support