prusa3d/Prusa-Firmware-Buddy #4610: [FEATURE REQUEST] Need local NTP Server Setting & Proxy

hman2, 2025-04-17:
### Describe the feature

This is a reopening of a pre-existing record.

We still need the ability to defined a local (!) NTP server in the printer FW. The fact that the FW now tries to access via DNS is no improvement, because Prusa's server is outside of every company's LAN (except for Prusa). If you have strict company rules, like every company should have (must have) then you can't "phone home" talking binary (meaning: not possible to be scanned for security risks).

For getting current date und time, there is a local company NTP server, but I cannot add it's IP address to the printer. Yes, theoretically it would be possible to alter DNS resolution to point to the local server, but that would require a complex request to the LAN department and it would be VERY hard to explain to the security deparment. Almost impossible to obtain their okay. Standard reply from them is "Buy a better printer that can cope with the reality of a company LAN with rules...".

I would like to expand the original request that a setting to use a Proxy is necessary. What applies to NTP applies to Pruxa Connect as well. You simply cannot "phone home". If that would be really needed, there is a Proxy server for that, but current firmware (6.2.4) cannot handle that as well...

Thanks
hman2

hman2, 2025-04-17:
For us, this is for a MK4S with MMU3.

CZDanol, 2025-04-17:
Hello, this seems to be a duplicate of #3261

closed 2025-04-17 by CZDanol

hman2, 2025-04-22:
created, #4621.