prusa3d/Prusa-Firmware-Buddy #4613: [BUG] Print bed moves too far during print start and pushes the nozzle upwards - Crash!

ShakieVan, 2025-04-17:
### Printer model

XL

### Firmware version

6.2.1 - 6.2.4

### Printing from...

PrusaLink

### Describe the bug

When starting a print with my Prusa XL, print head and print bed starting homing after each other. During homing of the print bed it moves up from below as expected. However, instead of stopping immediately upon contact with the nozzle, it continues to rise slightly, pushing the entire print head upwards by a few millimeters before stopping. This behavior is concerning and could potentially damage the printer.

### How to reproduce

- switch on printer
- load filament in nozzle 1
- start print with PrusaSlicer

-> Print head starts and finishes homing
-> Print bed starts homing and crashes into nozzle

I have recorded a video showing this behavior:
https://www.youtube.com/watch?v=0uNAfCaLquc

Notice that the bed does not stop upon contact but pushes the nozzle (and print head) upward. You can hear a cracking sound.

It seems to me that the printer needs to stay turned off for a while ... one night or so ... to reproduce the behaviour.
If I repeat the process immediately it does not work.

### Expected behavior

The bed should stop immediately when it reaches the nozzle and not apply upward pressure to the print head.

### Files

It does not depend on the print file.

CZDanol, 2025-04-17:
Hello, please contact our support for troubleshooting, this is likely something specific to your printer.

MyOpenLab, 2025-04-17:
I have updated my MK4 firmware and now have print ventilator fail test, X, Z and Y axis wrong positioning and crash when homing, I will back to an old firmware

ShakieVan, 2025-04-18:
> Hello, please contact our support for troubleshooting, this is likely something specific to your printer.

Hi CZDanol,

when I contact the support I want to give them an explanation. I want to avoid to be a ping pong ball.
Could you explain your thoughts why this is specific to my printer, please?

CZDanol, 2025-04-22:
I'm not sure what you mean. I think this is specific to your printer, because it generally does not happen on other printers, to my knowledge.

senergy5, 2025-04-23:
Hello ShakieVan,

I also have this issue with my Prusa XL. It looks near identical with the way you have discribed your machine, though on my printer it happens randomly, but more and more often, the more prints that I do. I cant make it do a crash reliably by the way you have discribed.

I contacted Prusa Tech for support for help. I was instucted to update the firmware. This didnt fix the problem. Ive also scoured the web for anyone else that may be experiancing this, but have not found a single hit, until i seen your post just today. So I think this is indeed a rare problem.

Im currently in communication with Prusa, we are attemting to find a solution, though they are pretty busy ATM, so its taking some time between emails. Im getting tect support for this problem and also extreme nozzle grind on previous layers, which is also progressivly getting worse the more prints I make, I have a hunch that these two issues maybe are related.

Here is a short vid of my printer crashing. It cant be doing the head assembly much good.
https://youtube.com/shorts/Lq-qsA_hHtE?feature=share

If I have some info come in that may also help you with this issue, Ill send you a post from this sight. Good luck ;)

CZDanol, 2025-04-23:
Some ideas to check out:

1. Does the same happen when you're homing with other toolheads?
2. You could try disconnecting the other toolheads, to rule out the MODBUS being overloaded by some other dwarf
3. You can try disabling the network and Connect and such to rule out the CPU being overloaded by some other task
4. You can try tapping the nozzle and watching the loadcell value in the sensor info screen
5. You can try turning "Log to file" in Settings → System for the homing sequence. I don't expect there to be anything interesting, but just in case...

ShakieVan, 2025-04-23:
I have tried to narrow down the issue.
The incident did not occur during this.

While doing this, I noticed that the printer behaves differently.

When the problem occurs (Case a):
1a: When loading the filament, the print head homes and the loading process completes.
2a: Then, at the start of the print, the print head homes again (X-Y).
3a: It then moves to a position for Z calibration.
4a: Then the print bed moves up and collides.

When the problem does not occur, as is currently the case (Case b), the following happens:
1b: When loading the filament, the print head homes and the loading process completes.
2b: Then, at the start of the print, the print head homes again (X-Y).
3b: Then the print bed moves up and stops 5mm before the upper endstop.
4b: Then the print head moves to a position for Z calibration.
5b: Then the print bed continues to move up and touches the nozzle.

Step 4B is noteworthy.

What puzzles me in the OK case (Case b):
Regardless of where I had the print bed previously, in step 4B it stops exactly 5mm (or so) before the endstop.
Question 1: How does the print bed know its position, even though I had previously turned off the printer? I don't see a sensor for this.
Question 2: Why is step 4b skipped in Case a?

ShakieVan, 2025-04-23:
@CZDanol 
By the way your point 2 is not possible. The printer starts with a fatal error while booting.

I tried it with toolhead 2 and switched both toolheads and tried it again with both of them. The case sadly did not orrur. Neither with the standard toolhead nor with the other one.

For me it seems to happen after a long off time of the printer and when the print bed is about half the way down.

Maybe... there comes another thought in my mind. What if it happens with large printfiles? Could it be that the printer is somewhat busy with the loaded data while starting the print? But today I have no time left.

senergy5, 2025-04-24:
ShakieVan. Nine out of ten times I get Case A, homing without issues. In Case B, there is no sequence I can do to cause a crash to happen. And 3b & 4b does not occur in my homing cylce. The roughly 10% of homings that crash have no set of pre-set events, its completly random. So we are having the same symptom with the collition to nozzle, some things are a little different.

CZDanol. I have just the single head unit. This is most definitly the proverbial needle in a haystack! Though there is a root cause, and we will inevitably find out what it is.

ShakieVan and CZDanol. Have you contacted Prusa tech support? They will be happy to assist you, Ive received very good support from them. Ive given Andres from Prusa support this post address, it may help that we pool our infomation and resources to find a solution to this frustrating problem.

Dragonos2000, 2025-04-30:
Maybe related to this issue below?
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4395

My XL also causes deep bumps in the PEI sheet. However, I did not notice it pushind the head/gantry upwards or having crack-sounds...

ThomasHalwax, 2025-05-26:
I'm afraid the Core ONE has a similar problem. Pushing the nozzle into the print sheet may not be as hard as in your case but it has already left small marks on the print sheet and my damage the nozzle as well.

https://github.com/user-attachments/assets/ae0b645d-1a58-4a73-97a0-a39edce12f27

Schleichmichl, 2025-07-28:
I had an similar issue with my XL.
I had an uncontrolled lifting of the bed at print start. I was printing the same job on and on, everthing was fine. The nozzle of tool 2 scratched into the bed before purging, leaving two marks on the way to the dock. The contact was so hard that the heatbreak broke, the bed is deeply scratch (~0.3mm on the power textured sheet) The screw for the heat break was still tight. A loose hotend could not be the cause. The scratch to the dock is one line which is interrupted. I could only guess that the bed was lifted twice, otherwise these markings don't make any sense to me.

[image]
[image]
[image]

nicke, 2025-09-18:
I have the same issue. it's the second broken nuzzle.

[image]