prusa3d/Prusa-Firmware-Buddy #4616: [FEATURE REQUEST] Ventilation grille prompt

mik-at, 2025-04-20:
### Printer model

Core One

### Describe the feature

Hi!

Printing a lot with filaments with chamber above 45°C and it really is a pain to always have to walk to the printer and press continue after it was powered off between prints.

So it would be cool to either have one of the two options:

* Have a setting to turn off the Ventilation grille prompt
or
* Have the prompt after changing to certain filament types that are typically need higher or lower chamber temperature

Thank you very much!

noxhirsch, 2025-04-21:
Yes, this is quite frustrating, especially since I turn off my printer after most prints. This prompt should definitely appear during filament changes instead of after starting a print, and it should **additionally** be possible to disable it if desired (since there will be people who motorize and automate the whole thing).

RPDengel, 2025-04-24:
Hi, i also agree that the ventilation grill prompt is a little bit annoying. 
You should been able to disable the prompt by menu, or place the prompt at the filament change as noxhirsch mentioned before.
THX!

CZDanol, 2025-04-25:
Internal ticket: BFW-6853

pstimpel, 2025-05-17:
‘Little bit annoying’ is the wrong term. It blocks the screen, and Prusalink keeps saying something needs attention. This whole warning and the way it was implemented is just completely user-unfriendly. Same goes for the dialogue in Prusalink that expects me to confirm that the print bed is clear when starting the next print.

barneym, 2025-05-31:
+1  Just ran into this while trying to print remotely with EasyPrint.  Seems like non-sensical baby-sitting, or at least tell us what the prompt is over PrusaConnect and let us override.  Should be an option for ALL/Most prompts that don't represent an imminent danger.

pstimpel, 2025-06-12:
@michalrudolf : Great news! Do you guys check more than this grilles thing, or do I need to open another issue for the config of the "clean your bed before printing" babysitter dialogue?

pstimpel, 2025-06-13:
> [@pstimpel](https://github.com/pstimpel) Sounds like a non-related issue, so please open another one.

Your wish ...

rfilgas, 2025-06-14:
> Hello, From 6.3.4 onwards, there will be an option to disable the ventilation grilles check in the Hardware menu and in the dialog itself.

Where is this fixed? I'm new to this codebase so I'm trying to familiarize myself. I was thinking about attempting to fix it myself and flash custom firmware if it wasn't done already.

rfilgas, 2025-06-14:
I see there's an internal branch and release system which is why I can't see it. Is there an ETA on when 6.3.4 will be released?

3d-gussner, 2025-08-14:
Assigned myself to prevent Stale-bot closing it. Reopen until major firmware is released.