prusa3d/Prusa-Firmware-Buddy #4618: [ENHANCEMENT] CORE One - Allow manual axis movement with door open

TheNexusAvenger, 2025-04-20:
### Printer model

CORE One

### Describe the enhancement

When attempting to move an axis manually in `Control` -> `Move Axis` with the door open, the CORE One prompts to close the door, even though this isn't needed. During maintenance or recording videos for support, this becomes an annoyance of either closing and re-opening the door for just the movement or holding down the door switch during the movements.

### Expected functionality

I expect to be able to do manual axis movement with the door open or closed (not just closed).

CZDanol, 2025-04-22:
Hello, unfortunately we cannot allow this due to legislative reasons. You can however disable the door sensor from the settings at your own risk.

closed 2025-04-22 by CZDanol

TheNexusAvenger, 2025-04-22:
Oh, not the answer I was expecting.

Are you able to cite what legislation prevents this?

CZDanol, 2025-04-23:
The requirement comes from EN ISO 12100. I cannot cite from the norms, because the law prohibits it :D