prusa3d/Prusa-Firmware-Buddy #4619: [BUG] M92 being ignored.

kenour, 2025-04-22:
### Printer model

MK3.5S

### Firmware version

6.2.4+8909

### Printing from...

Prusa Connect

### Describe the bug

I use M92 to calibrate my machine for dimensionally accurate prints. It has been working flawlessly for years.
I've been having general issues since 6.2.3 I think it was, and I just checked the size of my prints, and it's undersized.
I then ran my calibration again, and even with the below line in my 'Start G-code' section;
M92 X100.29 Y100.53 Z401.06 ; calibration
My calibration shape printed as if it was ignored. I then considered it may have drifted over the years, so changed it to;
M92 X100.64 Y101.05 Z402.08 ; calibration
But it printed exactly the same size, which leads me to think the M92 command it being ignored. I confirmed it was in the produced gcode by inspecting it with the gcode viewer, and it was present as expected.

All my models are now printing undersized since M92 has started to be ignored.

### How to reproduce

Print a test shape, use M92 to calibrate it's size, print it again to confirm no change to size.

CZDanol, 2025-04-22:
Internal ticket: BFW-7094

kenour, 2025-06-23:
> Thank you for your contribution to our project. This issue has not received any updates for 60 days and may be considered "stale." If this issue is still important to you, please add an update within the next 7 days to keep it open. Administrators can manually reopen the issue if necessary.

Hello Mr Bot. Yes it's still important, and from what I understand it's now been moved to an internal ticket. Is there anyway to follow this internal ticket to see the outcome/updates?

kenour, 2025-06-23:
Have requested #4697 to have this calibration added to the advanced menu after the issue has been fixed.

kenour, 2025-08-25:
> Internal ticket: BFW-7094

Any chance of finding out what stage the fix is up to?

bkerler, 2025-08-26:
I just checked and I don't seem to see anything off in the code, except that the settings aren't persistent ... as you wrote M92 was used as start gcode, the written values should be used in the planner (variable planner.settings.axis_setps_per_mm per axis). Did your gcode maybe overwrite the values ?
Will try to emulate this behaviour to see if anything interfers with the planner ...

kenour, 2025-08-27:
> I just checked and I don't seem to see anything off in the code

Sorry do you mean in the current prod, or RC?

3d-gussner, 2025-08-27:
@kenour I don't have the MK3.5 but tried the `M92` on my MK4 and it is working.

My steps:
- Connect via USB to printer and use [Putty](https://help.prusa3d.com/article/crash-dump_364959) or similar serial terminal.
  - Connect with putty just after the `Bootloader 2.x.x` changes to `Firmware 6.x.x .... 
- `M503` to get used values
  - `echo: M92 X100.00 Y100.00 Z400.00 E380.00` 
- `G28` to home the printer
- `M114` to get positions
  - response `X:-1.00 Y:-4.00 Z:2.00 E:0.00 X:-100 Y:-400 Z:800`
- `G1 Y100` to move the bed
- Placed a lineal inline of the nozzle on X axis
- `G1 X100`
  - On lineal I can see that it is at 100mm
  - `M114` shows that the printer is at X100 Y100
- `M92 X50` 
- `M503`to verify if `M92 X` changed from `100` to `50`
- `G1 X0
  -  On lineal I can see that it is at 50mm
  - `M114` shows that the printer is at X0 Y100
- `G1 X200`
   -  On lineal I can see that it is at 150mm
  - `M114` shows that the printer is at X200 Y100

So it moved the X axis after changing `M92 X` from 100 to 50 by 1/2 on the lineal.

One difference between MK3.5 and MK4 is that the MK3.5 uses the MK3/S/+ 200 step motors while the MK4 has 400 steps motors, so your values may differ.

As long the printer is powered on the newly set `M92 X` value stays active, but as @bkerler pointed out it is not stored in permanent storage and is reset to default values after a reset or power off-on.

kenour, 2025-09-07:
> > [@kenour](https://github.com/kenour)
> > As long the printer is powered on the newly set `M92 X` value stays active, but as [@bkerler](https://github.com/bkerler) pointed out it is not stored in permanent storage and is reset to default values after a reset or power off-on.
> 
> Yeah I'm aware it's not persistent, that's why I have the M92 calibration values in the start of my gcode.
> 
> I can confirm, this has not been working since the specified firmware update.

Here's the **SAME GCODE** printed on the **6.1.3+7898 firmware**, the last firmware to properly honour M92.

[image]

Same GCODE, only difference is the firmware version, I don't know how to make it any clearer... Somewhere between 6.1.3+7898 and the next firmware release, y'all borked the way the printer handles M92 (as in, it just completely ignores it). I've shown you the calibration code I added in the start of my GCODE to compensate for printer inaccuracies (I used M92 X125 Y125 Z400 ; calibration in this example to exaggerate the affect), this brings my prints perfectly on size. This isn't a material shrinkage problem (as it's non linear), this is an easily fixable issue.

[image]

Now, I've requested #4697 as a way to allow calibration in the menu. I've also suggested a ghetto way to compete with the Bambu Lab Vision Encoder Calibration Tool, which is essentially following my workflow of printing a calibration pattern, scanning it using the screen printing on the bed as a known distance, then working out the scaling factors to input in the calibration menu.

For printers with a load cell, I'm sure you could make a stepped model with 50/100/150/200mm features and probe it after printing for Z or something similar (for people without accurate callipers). I dunno, I think this solution is on par with the upcoming strobe timing for belt tensioning, throwing grey matter at a calibration issue 🤷‍♂️ Either way, I still feel like...

[image]

bkerler, 2025-09-09:
@kenour can you retry your gcode, but please make sure you run `M92 X125 Y125 Z400 ; calibration` AFTER this line `G28 ; home all` ?

It could be the cause is coming from the G28 gcode, that's why.

kenour, 2025-09-17:
> [@kenour](https://github.com/kenour) can you retry your gcode, but please make sure you run `M92 X125 Y125 Z400 ; calibration` AFTER this line `G28 ; home all` ?
> 
> It could be the cause is coming from the G28 gcode, that's why.

RIGHT! Well that worked! As I said, I've used the same calibration code in that spot for years, so something after 6.1.3+7898 caused it to be ignored if before G28. Has the G28 code been updated to clear M92 or something?

mike-bee, 2025-09-25:
In my case (XL 5TH) PrusaSlicer emits two G28 commands. "G28 XY; homeXY" and "G28 Z; home Z (with MBL tool) short after. I guess I have to place M92 calibrations behind the latter, right?

Are you guys looking into G28 clearing calibrations?

mike-bee, 2025-09-25:
> > Are you guys looking into G28 clearing calibrations?
> 
> Or better yet, get everyone to put their weight behind [#4697](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4697) 🙏

I already did. But first they need to get the basics right ...

kenour, 2025-09-26:
> > > Are you guys looking into G28 clearing calibrations?
> > 
> > 
> > Or better yet, get everyone to put their weight behind [#4697](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4697) 🙏
> 
> I already did. But first they need to get the basics right ...

I mean if they implemented it they would need to ensure it worked. Maybe...

kenour, 2025-11-15:
> [@kenour](https://github.com/kenour): What do you know that I don't? The issue still exists with my XL!

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4619#issuecomment-3271415598

Oh this worked for me, did you try it? The issue I reported is solved for me so I closed it.