prusa3d/Prusa-Firmware-Buddy #4620: [FEATURE REQUEST] Side filament sensor calibration test

sirmc, 2025-04-22:
### Printer model

Core One

### Describe the feature

I installed my Core One, and all the calibration tests passed successfully. However, it turns out that the side filament sensor (IR sensor) did not work correctly. The sensor was detected as connected, but the value was always NINS/ 1, even when trying to manually block the IR sensor.

As a result, the printer will ask for filament changes non-stop during prints, when it detects that there is filament in the Nextruder, but the side sensor reports NINS. This behavior can be fixed by disabling the side filament sensor, but from the behavior it is not immediately obvious what is going on.

During setup, there is no check to see if the side filament sensor actually works. All the filament sensor tests pass if there is filament inserted in the Nextruder.

It would be a good idea to add an explicit check for calibration of the side filament sensor, allowing for detection of a faulty IR sensor or poorly assembled sensor lever mechanism.

mkyral, 2025-05-15:
This works also in opposite way - today I wanted use all remaining fillament on the spool and use another one. But Core One did not detect fillament run out with side sensor, so I had to remove top cover and remove the remaining fillament from the Nextruder side.

Fortunately print continued and finished successfully.

mkyral, 2025-07-15:
Still valid for me.

mkyral, 2026-05-18:
Any update on „Internal ticket: BFW-7095“ ?