prusa3d/Prusa-Firmware-Buddy #4621: [FEATURE REQUEST] Proxy capability needed

hman2, 2025-04-22:
### Printer model

MK4S/MMU3

### Describe the feature

For using our MK4S (plus MMU3) in our corporate network (with high security rules) it is necessary that the printer can talk to Prusa Link/Connect via proxy server. No equipment inside our LAN is allowed to talk binary to the internet directly (well known ports are exluded, so HTTP(S) etc. do work, but the Prusa printer won't see updates or the cloud. And not internet time - but there is an internal NTP server that can be used insted. To allow usage, NTP functionaliy is needed, but for this I created https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4610.

### Expected functionality

An interface to specify the IP address of the proxy server that out networking department built so communication can go through this server.

hman2, 2025-04-22:
This funcionality (proxy) is needed on both the printer and the slicer as well...

CZDanol, 2025-04-23:
Internal ticket: BFW-6867

CZDanol, 2025-04-23:
For slicer, you'll need to create an issue in the Slicer repository: https://github.com/prusa3d/PrusaSlicer

hman2, 2025-04-23:
Done. https://github.com/prusa3d/PrusaSlicer/issues/14484