prusa3d/Prusa-Firmware-Buddy #4624: [BUG] Printer Becomes Unresponsive to Remote Jobs While in Submenus

metikumi, 2025-04-23:
### Printer model

XL

### Firmware version

6.2.4

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

When the Prusa Buddy XL is operating within a submenu (e.g., **Change Filament**), the printer stops accepting new jobs and appears as **offline** in Prusa Connect. If a user physically approaches the printer and exits the submenu, a transfer screen sometimes appears but remains stuck at 0%, despite the transfer possibly having already completed in the background.

This behavior is both unexpected and frustrating—especially when the printer is located a considerable distance from the controlling computer. Not only is manual intervention at the printer required, but the print job must also be manually restarted after leaving the submenu.

## Proposed Solutions

### Solution 1:
Automatically exit any active submenu when a new print job is received from Prusa Connect.

### Solution 2:
Introduce a timeout in all submenus. If there’s no user activity for a defined period, the printer exits to the home screen.

### Solution 3:
Implement menu virtualization. Allow the actual menu state to be visible and controlled via Prusa Connect. This would not only resolve this issue but also streamline remote troubleshooting and printer management.

### How to reproduce

1. At the printer, start a filament change, but **do not exit** the menu after completing the change.
2. Walk the 200 meters (give or take a marathon) back to your computer.
3. Initiate a new print job via Prusa Connect.
4. Observe as the printer appears **offline**—clearly needing a break.
5. Hike back to the printer, only to find it still chilling in the filament change menu.
6. Upon exiting the menu, a transfer screen pops up but stubbornly refuses to progress beyond 0%.
7. Hit **Quit** to return to the main menu.
8. Return to the computer and try the print job again, feeling more fit, but less amused.

### Expected behavior

1. Complete the filament change and remain in the submenu.
2. Head to the computer and send a new print job.
3. The printer should automatically exit the submenu and start the new print job—no unnecessary steps, and no bonus cardio.

Gik-0101, 2025-04-23:
Same issue/bug like

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4581

CZDanol, 2025-04-23:
Closing as a duplicate.

CZDanol, 2025-04-23:
Btw, I appreciate your humour in the reproduce steps :)

closed 2025-04-23 by CZDanol