prusa3d/Prusa-Firmware-Buddy #4625: [FEATURE REQUEST] Account for Preheating Time When Starting a Print

metikumi, 2025-04-23:
### Printer model

XL

### Describe the feature

Currently, if the print sheet is preheated well in advance of starting a print job, the printer still enforces an additional wait period before beginning to print. This means that even after preheating the print bed (e.g., 10 minutes before printing), users may still have to wait several more minutes, which is both inefficient and counterintuitive.

While I understand that this built-in wait is intended to prevent errors—such as users inserting the print sheet right before the print and skipping proper warm-up—those who proactively preheat their bed are likely aware of such caveats and would benefit from a smarter system.

### Expected functionality

### Proposed Solution

- **Continuous Monitoring:**  
  The printer should record the bed temperature for the last 20 minutes, for example by storing temperature values every 10 seconds in a circular buffer.
- **Temperature Stability Check:**  
  When it’s time to wait for “sheet normalization,” the firmware should check if the bed temperature has been stable (e.g., within ±2ºC of the target) for the required duration.
- **Conditional Waiting:**  
  - If the bed has already been at the target temperature (e.g., 80ºC ±2ºC) for the necessary amount of time, the print should start immediately—no redundant waiting.
  - If there are significant temperature variations (such as those caused by swapping the steel sheet just before printing), the printer should enforce the standard wait period for proper normalization.

CZDanol, 2025-04-23:
Internal ticket: BFW-4779

CZDanol, 2025-08-06:
Reopening as the firmware for the XL is not yet released. We will close this as done after the release :)