prusa3d/Prusa-Firmware-Buddy #4627: [ENHANCEMENT] Remind user to apply coating with the current configuration of steel sheet and filament type

espr14, 2025-04-24:
### Printer model

32b Buddy printers

### Describe the enhancement

Thanks for custom filament option. Still, it can include compatibility with steel sheet type. E.g. PETG should not be used directly on the smooth PEI sheet. However, user can apply a coating before printing. Incompatibility or possibility of a coating could be shown to user just before printing according to selected sheet and filament type.

Combinations for FW hardcoded filaments can be derived from [https://help.prusa3d.com/cs/filament-material-guide](https://help.prusa3d.com/cs/filament-material-guide)

### Expected functionality

Warn user just before printing like proper nozzle size/type, etc.

3d-gussner, 2025-04-25:
@espr14 Great idea and in 8-bit I have an approach see https://github.com/prusa3d/Prusa-Firmware/pull/4832

CZDanol, 2025-04-28:
Good idea. This would though first require tracking what sheet the user has on the printer, just for this purpose, which is currently not done except for Mk3.5 and Mini.