prusa3d/Prusa-Firmware-Buddy #4629: [BUG] NFC does not transmit accented characters

Kachidoki2807, 2025-04-26:
### Printer model

MK4S and CORE One

### Firmware version

6.3.2 and 6.2.4

### Printing from...

the air

### Describe the bug

Using an accented character into the SSID like "é" is not possible, it becomes "?" when transmitted by NFC.
The available local keyboard on the printer does not offer these characters neither.
Probably not related but I also got a red screen once with HAL I2C error while trying to reproduce.

[image]

The printer itself and its WiFi stack can use accented characters because if I use the Scan feature, all is ok (hopefully I was able to fully type the password).
I know that the Buddy3D camera also have issues with special characters, not tested myself, reported by a user on the forum.

### How to reproduce

Just type an SSID with accent: "WiFi lé la", "Déjà vu", "Oh là là!"
Send it to the printer through NFC.
Et voilà. :-)

### Expected behavior

Accented characters and globally any characters should be accepted for the SSID. Even unicode should be usable.
Same apply to the password.

CZDanol, 2025-04-28:
Internal ticket: BFW-7114

Kachidoki2807, 2025-09-01:
I'll be happy to give feedback if a fix is proposed. (This comment purpose is totally to reset the stale counter. 😅)

Kachidoki2807, 2025-09-16:
Ok, this morning I ran some additional tests (on the CORE One) since this ticket is quite old and the Prusa App has received a few updates.

1. When everything is configured and working properly, both the firmware **and** the app can display accented characters correctly. I can see them in their respective menus.
2. In the app, I can enter accents without any issue for the SSID, and in the firmware, the accents are also visible when using the Wi-Fi scan function.
3. Since both sides handle encoding correctly, the problem is probably due to a bad conversion or an incorrect cast between the two.

Additional note: characters like é belong to the extended ASCII table (code 0xE9), but in UTF‑8 they are encoded on two bytes (0xC3 0xA9). Usually, when UTF‑8 is misinterpreted as single-byte encoding (like Latin‑1), you see artifacts such as Ã© instead of é. Here, however, we only see a single ?, which suggests the input is single byte encoded and being replaced during parsing rather than displayed incorrectly.

bkerler, 2025-09-17:
Yes, I tried it now as well and there is support for the accents in the fw (as you already stated for example wifi-scan), so I'm now trying to rule out whether the prusa app is causing this or a process related to the nfc handling. I can reproduce this bug as well for the MK4S and the Core One, so it should be rather easy to hunt down this issue.

bkerler, 2025-09-17:
Small heads-up: It doesn't seem to be a fw issue, but most likely an issue in the prusa app not using utf-8. See here:

[image]

So the data coming from the nfc part is already not encoded in utf-8, thus being displayed incorrectly.

It's now coordinated with the Prusa App Team.