prusa3d/Prusa-Firmware-Buddy #4630: [ENHANCEMENT] Automatic Z alignment procedure

mik-at, 2025-04-26:
### Printer model

Prusa Core One

### Describe the enhancement

Sometimes, when I start a print, following prompt shows on the screen:
"The bed doesn't seem to be aligned properly. Run Z alignment procedure?" "Yes" "No"
I always hit yes, because why should I want to print on a misaligned bed?

The prompt is just delaying the print, because I have to manually hit "Yes" to finally start.
I don't see a reason, why the printer should not start the Z alignment procedure?
The space under the bed should be unoccupied at all times and the procedure is working without me having to watch or interact with it.

### Expected functionality

The printer should, at least the first time this happens, always automatically do the Z alignment procedure. If the bed is still misaligned after trying, it should inform the user.

CZDanol, 2025-04-28:
Hello, fair point. We will discuss this with our product manager.

Racemade, 2025-04-29:
Nice idea! Same for XL for a standard procedure would be nice.

mik-at, 2025-06-30:
still important.

closed 2026-10-05 by github-actions[bot]

mik-at, 2026-10-05:
still important

VinPre, 2026-10-08:
The new 7.0.0 Firmware on the Prusa website says they added "Automatic Z-axis alignment on uneven bed" as a new feature. I think they are referring to this issue. 

https://help.prusa3d.com/downloads/core-one