prusa3d/Prusa-Firmware-Buddy #4631: [BUG] Fix nozzle cooling parameter

espr14, 2025-04-26:
### Printer model

32b Buddy

### Firmware version

6.3.2

### Printing from...

any

### Describe the bug

M109 detects parameter 'F' for fast nozzle cooling but it should be 'C' not to conflict with Marlin's autotemp flag.

Just replace
`(void)thermalManager.wait_for_hotend(target_extruder, no_wait_for_cooling, parser.seen('F'));`
with
`(void)thermalManager.wait_for_hotend(target_extruder, no_wait_for_cooling, parser.seen('C'));`

Introduced in #4014 but stuck in time.

### How to reproduce

```
M107
M104 S100
M109 R100
M104 S50
M109 R50 C
```

### Expected behavior

Fan should turn on (and off after reaching the final temperature).

CZDanol, 2025-04-28:
Hello, none of our printers have AUTOTEMP enabled, so I'd say this is purely cosmetical. Is this flag emitted by the Slicer? Because if it is, when we cannot just change it.

espr14, 2025-04-28:
It's safer to avoid collision. However, Prusa Slicer does not use it so it's a trivial and cosmetical change. Unfortunatelly, it's so simple that nobody cares :-(. But it would be hard once it will be used widely.

closed 2025-07-07 by github-actions[bot]