prusa3d/Prusa-Firmware-Buddy #4636: [BUG][COREONE] Connection to Prusa Connect lost while printing

zuidwijk, 2025-04-27:
### Printer model

COREONE

### Firmware version

6.3.2

### Upgrades and modifications

None, stock

### Printing from...

USB Drive

### Describe the bug

When the printer is idle, it's connected to Prusa Connect. Soon the printer starts a job from the usb drive, it looses connection after short period of time. 

It is connected to the network, I can ping it and connect to Prusa Link. 

But the network status gives TLS error.

As soon as the job is completed, it reconnects and I'll get the notification that the print job is finished.

### How to reproduce

Just start a print job

### Expected behavior

I expect it to stay connected. My other Prusa's (MK4S's, XL and mini+) stays connected while printing.
All printers are connected via ethernet to the same switch (firewall, routing etc is ruled out)

zuidwijk, 2025-04-27:
[image]

aizerin, 2025-05-05:
Same problem, but Ive got just `server error` instead of TLS. But it was working fine at beginning.

[image]

codingcatgirl, 2025-05-05:
@aizerin Nope, Prusa Connect is having issues right now in general, not just on the printer.

ulab, 2025-05-06:
I do have the same issue with prints that use a lot of arc moves (G2/G3). 

I also noticed that the response of the UI on the printer can feel sluggish at times during those prints.

Smooph86, 2025-05-17:
Randomly, I have the Same Connection issues.
Sometimes, my CoreOne loses Connection to Prusa Connect…

dkbay, 2025-05-17:
> Same issues here - sometimes CoreOne Loses Connection to PrusaConnect…

Issue exists on every printer.
CPU gets overloaded and doesn't have enough cycles to do networking as kinematics take priority.
You should also notice that the printer UI becomes noticeably laggy when this happens.

All we can really do at this point is wait for @Prusa-Support to fix it through firmware or make some kind of new board with more CPU power.
I opened a ticket about this about a month and a half ago that is still open.

ulab, 2025-05-24:
This is not an airport. You don't have to announce your departure. Good luck with your new machine.

Groucho83, 2025-05-24:
> This is not an airport. You don't have to announce your departure. Good luck with your new machine.

did you think of that yourself, or is it about a 5 year old joke now? 

I want others who find this thread to realize that prusa has a 60 day refund policy, and i encourage anyone reading this to vote with their wallet. a product in this price class should ship complete. my flashforge 5M base model is more functional than my prusa,  because i can remove the USB stick and prepare the next print while there's a job in progress. i can't even get p link to stay connected on my machine on wifi or wired, despite the fact that i can ping it on my network

dkbay, 2025-05-24:
> did you think of that yourself, or is it about a 5 year old joke now?
>
> I want others who find this thread to realize that prusa has a 60 day refund policy, and i encourage anyone reading this to [...]

I got this reply from Prusa support.
So they are actively working on it and they have acknowledged it.
These things take time to diagnose and fix. I'm also frustrated it has taken a while don't get me wrong.
But I still much prefer my Prusa over the Flashforge I've tried and the quality is generally better than my Bambu printer.

> Thank you for your detailed follow-up and for the time and effort you've dedicated to replicating and analyzing this issue. I completely understand your frustration, and I’m truly sorry that your experience so far has not lived up to expectations, especially given the time you've invested and the premium nature of the product.
> 
> I want to assure you that our development team is actively investigating this matter. We've reviewed the behavior you're reporting, particularly the UI slowdowns, networking dropouts, and overall performance degradation when printing complex models, and we recognize that this issue affects all Core One units under similar conditions. Your GitHub report and community coordination have been extremely helpful and are genuinely appreciated.
> 
> I'm glad to let you know that the root of the issue has been identified, and a fix is planned to be included in the next firmware version. The team is currently finalizing and testing this update to ensure it addresses the overload symptoms without introducing regressions elsewhere.
> 
> While we understand the delay has been frustrating, firmware-level optimizations of this nature require careful implementation and verification. That said, your case has been flagged for internal priority, and we will notify you as soon as the update is available for download.
> 
> Thank you again for your patience and for offering to assist further.

Groucho83, 2025-05-25:
Thanks for adding that. Maybe there's something else wrong with mine because i lose prusa connect even when nothing is printing

dkbay, 2025-05-25:
It might be.
If your printer is connected via wifi that connection could just be
unstable.
I haven't experienced any issues with connect whilst not printing.

ulab, 2025-05-25:
> I want others who find this thread to realize that prusa has a 60 day refund policy, and i encourage anyone reading this to vote with their wallet. 

And I would love for you to realize that this is not a discussion forum or social media. An issue tracker is meant to report and fix problems. You venting, threatening to leave or telling people to buy something else does not help with anything. 

If you want to rant, go to Facebook, Reddit or even the Prusa Forum, but please keep it on topic here.

Groucho83, 2025-05-25:
Sick, i've got the same problem, only i'm having issued with a wired connection and while not printing. There's my report on this forum post that has been open for a month with no assignee but the dev team is working on. I'm happy to report p link does work after another FW reset

koombot, 2025-05-28:
Ive had this a couple of times, usually dies halfway through a code transfer and pauses the print halfway through.

Ive found going to network and swapping from WiFi to ethernet and back again (give it a couple seconds on ethernet) gets it to reconnect.

Groucho83, 2025-06-05:
i just finished speaking with prusa support. i don't think they're working on this

dkbay, 2025-06-05:
If this is a "CPU running out of cycles issue" as we theorize this issue might have the same root cause as this one https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4573#issuecomment-2925280482 given the comments of Prusa support.

zuidwijk, 2025-06-10:
Extra addition, see some of the printers:
[image]

Kwik is printing the cases and is connected. Kwek and Kwak is printing those Skeleton T-rex and are now at 50-ish percent (I can check in Home Assistant:

[image]

And yes, the names Kwik, Kwek & Kwak is Dutch for Huey Dewey and Louie :) 

[image]

daftu, 2025-06-13:
Having same issue on newest Core One firmware. Disconnecting Prusa Connect after 40-50 minutes of printing. On LCD there is TLS Error in Prusa Connect section.
No matter if connected by WiFi or LAN.

This makes Buddy3D Camera unusable as it doesn’t support local streaming or PrusaLink snapshots. Very annoying!

zuidwijk, 2025-06-16:
Indeed! The camera is not accessible via local network or PrusaLink. PrusaLink works as it is local. The integration in Home Assistant works, so there's a workaround. Now I want local access to the camera's.

I've bought 3 camera's (€44 each) and are quite useless as I cannot watch/check the prints remotely :(

Tupson444, 2025-07-01:
I'm having the same problem. I bought the camera to be able to monitor print while I'm away and cancel objects if something detaches from plate. But Core One keeps disconnecting from Prusa Connect shortly after print start, making the camera useless.

Meanwhile, a MK4 stays connected all the time, and it is on the same wifi (standing right next to the Core One), so it's definitely not wifi issue.

Ronin0000, 2025-07-02:
Same problem here when printing a print consisting of only arcs it never stays connected the whole print and the whole UI is really laggy as well.

brpacecar, 2025-07-02:
I am experiencing the same issues described above. While printing the the Core One shows offline in Prusa Connect and the TLS error shows on the printer. The gui is also slow to respond on the printer when printing. Currently running firmware version `6.3.3+10442`

julian-iFactory, 2025-07-10:
Experiencing constant loss of connection between PrusaConnect and Prusa CoreOne located in Brisbane, Australia

Problem persists despite:    

1. Changing WiFi password to remove special characters to allow Prusa + Camera to connect over WiFi - WiFi Connection successful.
2. Firmware upgraded to 6.3.3 (Buddy3D also upgraded).
3. Plugged Ethernet cable into CoreOne for same LAN & internet as PC (PC is not connected via WiFi) .
4. Windows 10 Dell PC, Aussie Broadband Internet with no VPN in use.
5. We are web development company that SSH into many servers each day with no known restrictions or network issues.
6. Printer works very well when connection is active. Over 14 prints mostly using network and PrusaConnect (not USB).

zuidwijk, 2025-07-11:
@julian-iFactory : the issue I'm refering to, is the connection loss mid-printing. When it's not printint, no issues. When it's printing something "not complicated" like a big cube, no connection issue. When you start printing something "complicated" the connection drops. When you look on your display from the Core One, you see that the printer self is still network connected, only the connection to Connect is gone and the error will be something like TLS error.

It's really ridiculous! I had to cancel a print at 58% as one of the items (20 parts) was warping and started to interfer. Normally on Connect you can select the item and cancel that object. Now I could not and therefore had to cancel all ...

It's getting frustrating now!

fftkatarn, 2025-07-12:
Same problem here, when it starts printing, after some minutes prusa connect goes offline with a "TLS Error", when the print ends, it goes back online.
I tried to upgrade the firmware to 6.3.4-RC but the behaviour persists.

zuidwijk, 2025-07-15:
@3d-gussner : I'm slicing now and will perform the tests. Note: I don't have the MK4S anymore, I've converted them to Core Ones. My issues was the TLS error. Never had that on the MK4S, never on the XL, only on the Core One. Simple jobs no issue, oprinting something "not simple": TLS error. No network issue as I can monitor them in Home Assistant via PrusaLink.

1st I'll do the arc tests on one Core One and the XL (other Core One's are printing for business).

zuidwijk, 2025-07-15:
@3d-gussner : 5mm cubes, or 5cm cubes? 

### In case of 5cm
In all fairness, when I slice 16 (4x4) instances with minimal settings (2wall 10% infill) that's almost 500gr of filament...
**Core One**
10 cubes: 300gr
10 instances: 300gr
16 instances: 490gr
**XL**
10 cubes: 300gr
10 instances: 300gr
20 instances: 600gr
A total of 2.5KG filament ( <<-- I hope you meant 5mm cubes ;-) )

3d-gussner, 2025-07-16:
@julian-iFactory Thanks for the info. The PrusaSlicer part is new in this ticket. I saw few reports that PrusaSlicer to PrusaConnect may cause issues on some workstations / PCs and PrusaSlicer needs to be restarted. But that is a different issue related to PrusaSlicer and not the firmware.

Do you have by chance an other printer type you are using? If so, do you have any issues using PrusaSlicer -> PrusaConnect -> Printer ?

Can you please try to slice and download the files and then open PrusaConnect to upload it? I know that is "old way" but just want to see if there is an issue with CoreOne and PrusaConnect (as reported here in this ticket) and  rule out that PrusaSlicer may cause the/another issue.

Avarios, 2025-07-16:
Hi, since two days the printer goes offline with TLS Error even faster. Before that i was able to print more complex parts , now the printer goes offline even on less complex parts.
So it's getting worse

PrintFandango, 2025-07-19:
my printer has been ok since I got about 1-2 months ago, and now, it wont connect to prusa connect anymore, it was running 6.3.2 then 6.3.3 ok, then 6.3.3 stopped connecting or connecting and dropping. I just uploaded to 6.3.4 and now, this is also an issue, it is not connecting to prusa connect anymore. I have not done anychanges to my network.

zuidwijk, 2025-07-24:
> Please try the FW [6.3.5-alpha](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.3.5-alpha) version
> 
> Only change to previous stable version 6.3.4 is the usage of Websockets see [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/compare/v6.3.4...v6.3.5-alpha)

I will try this tomorrow!

ulab, 2025-07-25:
This is not the place to discuss implementations, but objects were never printed from the internet. The printer only prints from it's own memory card.  

This issue is not caused by files being transferred.

zuidwijk, 2025-07-25:
> This is not the place to discuss implementations, but objects were never printed from the internet. The printer only prints from it's own memory card.
> 
> This issue is not caused by files being transferred.

Exactly this (in my case)! The Core One has no issues with Connect. It transfers the files and then starts printing. The issue I've reported is that the connection is lost mid-printing and restored when printer stops printing (either finished or filament runout). Same print on a MK4S (same buddyboard) has NO issues. Can't test that anymore as I've upgraded them all to Core One.

All I've noticed is that the menu/display is sluggish during those prints what made me think of CPU (or code optimalisation). MK4S and XL has none of these issues.

The new alpha fw @3d-gussner has offered has solved for me the connection loss. Well, actually the TLS error, as it was still connected and had still good ping times in the menu.

zuidwijk, 2025-07-25:
> @zuidwijk The MK3.5/3.9/4/S have a "simple" motion system, for each axis one driver, which is doesn't need that much computing time. The CoreOne with the corexy and some other functions uses more CPU as xy movements need to be computed for two drivers/motors.
> 
> MK4S:
> - Move X by 20 and Y by 10 ---> send command to X motor and to Y motor drivers the needed steps
> CoreXY:
> - Move X by 20 and Y by 10 ---> calculate how many steps the left and right motor needs --->send command to Left and Right motor drivers the needed steps.
>  

Yes! I know, that's why I said that the cpu has to work harder. The good news is that it doesn't loose connection due to tls error now (FW 6.3.5-alpha)

3d-gussner, 2025-09-02:
@Tupson444 There have been some updates on PrusaConnect server side, can you please verify if the issue is still there?

N3XT3D, 2025-09-05:
I am currently printing a lithophane image and as described by tupson444 I have a net fail after 13% print progress. The behavior is the same as with the tls error before, but this time with a different error message.

brpacecar, 2025-09-09:
Also getting the "Net fail" error randomly. Running 6.3.5-alpha on a core one.

geco0on, 2025-09-20:
I can confirm that 6.4.0 alpha firmware doesn't disconnect from my prusa connect.
I've been printing 20/24h on it for the past few days, it is OK.
I'm experiencing more Buddy 3D not showing video than before. Not on Prusa Connect nor over RTS stream.
Do I need to open new topic, or here is relevant?

3d-gussner, 2025-09-22:
@geco0on Thanks for the feedback. Please contact PrusaSUpport/Chat about the Buddy camera.

3d-gussner, 2025-10-20:
@Kubis205 Thanks for the update.

zuidwijk, 2025-10-20:
> @3d-gussner ... I contacted Prusa Support, and their recommendation was ... if possible, connect CORE ONE via an Ethernet cable, due to Wi-Fi network overload when multiple devices are connected to the router, which is usually the case... Another problem may be that the Wi-Fi component is enclosed in the metal body of CORE ONE, unlike, for example, MINI+, which has a Wi-Fi component in a plastic cover. Prusa Support also confirmed that 6.4.0 should improve data transfer to Prusa Connect. Unfortunately, some people still report problems even with fw 6.4.0. I connected both printers via an Ethernet cable directly to the router and updated the firmware to 6.4.0 (this was also the conclusion after discussing it with Prusa Support), and I can confirm that everything has been running smoothly for the last 3 days. I am printing on both printers at the same time (CORE ONE / MINI+) and both are CONNECTED to Prusa Connect throughout the entire print, as is the image feed from the Buddy3D camera. ... I hope it stays this way and continues to work. I understand that not everyone can connect CORE ONE via LAN, but there is probably no better solution at the moment...
> 

Interesting... my printers are all connected via ethernet and each buddycam via wifi (no ethernet possible haha). But no issues here. I had it with wifi and ethernet. Maybe your case is strongly network (wifi) related.

closed 2025-12-29 by github-actions[bot]