prusa3d/Prusa-Firmware-Buddy #4637: [BUG] Pre-assembled Core One persistent layer shifting FW6.3.2+10364

caprazzo, 2025-04-28:
### Printer model

Core One (pre assembled)

### Firmware version

6.3.2+10364

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Hi,

I have a print that failed twice in a row with severe layer shifting on a stock, brand new, pre-assembled Core One received just days ago.
* Successfully completed a number of prints, including a large earlier today
* I caught the second failure on video, and shared it with info@prusa3d.com
* I have been encountering the same issue on a MK4S, which is already been tracked in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4585. In fact I bought the new Core One hoping to get printing again.

STL
[UOLT-10023.0_base.stl.zip](https://github.com/user-attachments/files/19941994/UOLT-10023.0_base.stl.zip)

GCODE
[UOL-T10023-AC.slicer_0.4n_0.25mm_PLA_COREONE_3h58m.bgcode.zip](https://github.com/user-attachments/files/19941964/UOL-T10023-AC.slicer_0.4n_0.25mm_PLA_COREONE_3h58m.bgcode.zip)

PIC
[image]

Obviously a printer that does not print is deeply frustrating and I hope you could look at this with great urgency.

Thanks,
Matteo

### How to reproduce

On my printer, printing the attached gcode consistently brings on this failure. I have not trying restarting the printer in between prints in case you need to download any diagnostics.

lrueegg, 2025-04-28:
I had the exact same issue. Do you have enabled 'Avoid crossing perimeters' by any chance? Then it's probably this:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4573

caprazzo, 2025-04-29:
I did have "Avoid Crossing Perimeters" enabled. I disabled it and the print succeeded this time. That said, I have had failures without that setting on my other printer (MK4S), so it may just be luck.

paranojk, 2025-05-07:
I had the same problem, layers were shifted.

Printer: Core One self-assembled, without any modifications
FW: 6.3.2+10364
Printing from Prusa Slicer trough Prusa Connect

"Avoid Crossing Perimeters" enabled for 2 prints, then disabled for one. All failed.

Model from printables:
https://www.printables.com/model/56081-deserts-kiss-diorama-dice-tower/files
Desert's Kiss - Diorama Dice Tower - v1.2.5.stl
I can upload bgcode if needed, but it's 80MB.

Settings:
PLA Prusament
default 0.10mm FAST DETAIL @ COREONE 0.4 profile, with Fill density: 10%, Fill Pattern: grid, Top fill pattern: Archimedean Chords
belt recalibration between prints to 85Hz, just to be sure.

[image]

Snowhoschi, 2025-05-29:
FW 6.3.3
One hour ago, I'd stopped my Print. I think this was the First model, which is / have more or less circles.
Approx. after printing 60 mm in z-direction, I'd have my first Shift. ~3cm to the left in x-direction. But one big shift only.
Then the Print was ok, at this wrong coordinates.
I will Check the settings, if I had the Option activated or not.

theolderbear, 2025-06-26:
FW 6.3.3 I have same issue prints always fails with huge shift and noise. [Video](https://youtube.com/shorts/zYa29_JVerQ)

[image]

[image]

Snowhoschi, 2025-06-27:
Hi,

I'd did a calibration after the FW update now. 
Since now, it seams to work.
Up to now, the layer shift issue I couldn't observe.

theolderbear, 2025-06-28:
@Snowhoschi you mean update to 6.3.3?

sceiler, 2025-07-20:
Still happens with 6.3.4 on CORE One.

Snowhoschi, 2025-07-21:
> [@Snowhoschi](https://github.com/Snowhoschi) you mean update to 6.3.3?

Sorry for this late response. What I mean is, that I did a "Calibration" (the one from printers menu reachable from the display of the printer) for x / y  and z.

Up to now, it seems to be working.

Prusa-Support, 2025-09-06:
> Is this the same issue as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4573 ?

Good question.
If I got this right, the difference between https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4573, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4585 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4637 seems to be that...

- #4573
There is a diffuse concern about the compatibility of specific features of Prusa Slicer, like _Avoid Crossing Perimeters_ and _Arc Fitting_; the issue is not specific to one printer model.
From our side, we are aware of specific print conditions where Gyroid infill may be part of the problem as per [this comment](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4585#issuecomment-2823950808) (potential fix expected on future FW 6.4.x). I personally see a sort of correlation with circular movements (not based on objective data) and specific hardware/ambient conditions (not reproducible on all printers), but hopefully FW 6.4.x will improve the situation.
- #4585 
This issue is clearly based on MK4S printer model and our testers had mixed luck reproducing the issue. So we have an internal ticket and an open investigation.
I personally find it very interesting how the issue arose from an approved filament preset that makes use of very low nozzle temperature, so I'm somewhat more concerned by our own community-approved specific print profile than the firmware itself, but our testers may be onto something. Time will tell.
- #4637 
This issue is clearly based on CO printer model. I don't seem to find other very specific peculiarities on this issue at the moment.
This is our first full-DIY version of a CoreXY design, and our first compact printer with native enclosure - yet another chance to move steps in unexplored territories. So we have done our homework and the good news is that we have de-mystified rather complex troubleshooting steps during our last big research about Vertical Fine Artifacts.
For the sake of consistency, let me quote myself by directing you to [this important piece of troubleshooting](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4582#issuecomment-3247053836) shared on an "apparently rather different" issue, which I hope will become a preliminary check for everyone in this issue too.

What do these three issues have in common?
Oftentimes, the reported individual experiences are based on hardware problems (eg unexpected noises) and substantially customized print settings, or customized printer parameters.
So please consider [1] stepping back to stock parameters and [2] show the problem to our Customer Support for guided hardware checks, and keep us posted about your case resolution if any of these basic practices have helped.

Michele Moramarco
Prusa Research

Prusa-Support, 2025-09-13:
There is a new final release available for Prusa Slicer ([2.9.3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.3)), and a new firmware pre-release ([6.4.0-alpha](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha)).
Please check out and possibly test our stock profiles - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4573#issuecomment-3288285067.

Michele Moramarco
Prusa Research

closed 2026-01-14 by github-actions[bot]

jordanbrown0, 2026-01-14:
I encountered a similar problem with a new Core One.  I printed a few things fine, and then it started getting layer shifting, in jumps of about 1cm  diagonally left and back.

With help from support I tracked it down to a loose set screw on the pulley in the back right.  It apparently loosened up enough that it started jamming that side and restricting it to maybe one turn of  the pulley.  Tightening it up made the problem go away.

Note that there are two set screws on each pulley.  One of them needs to land on the flat side of the shaft.  For me, the loose one was the one landing on the round part of the shaft.

That behavior was easily reproduced "by hand"; with the motors disabled the head was really only willing to go forward-left and back-right; it was only willing to do left-right or forward-back an inch or so.